esp8266/Arduino #5513: [core 2.5.0] Loop Count much lower than with 2.4.2

TD-er, 2018-12-17:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[0fd86a07f0b22aa43f24dc6158b6d8093b607765]
- Development Env:		[Platformio]
- Operating System:		[Windows|Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|other]
- Flash Mode:			[dio]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY_LOW_FLASH]
- Reset Method:		[-]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Loop Count in loops per sec. is much lower than with core 2.4.2. It was also mentioned here: https://github.com/esp8266/Arduino/issues/5476#issuecomment-447115849

I have several test setups running core 2.4.2 and core 2.5.0 (see latest nightly builds of ESPeasy)
For example 2 Sonoff POW r2 nodes running the exact same configuration, the loopcount (per sec) of both differs quite a lot:

Core 2.4.2:
[image]

Core 2.5.0 beta1:
[image]

N.B. Not sure yet why in the node running 2.4.2 less things are being shown.

In general (among my test nodes) the loop-count is:
- 4000 - 11000 when using core 2.4.2 
- 1100 - 2300 when using core 2.5.0 (4200 when doing nothing)

As can be seen from both Sonoff POW r2 nodes, the loop-count is:
- 4200 for core 2.4.2
- 1148 for core 2.5.0

d-a-v, 2018-12-17:
`SensorSendTask` and `sendData` are highlighted in 2.5 but not present in 2.4, but you posted them [before](https://github.com/esp8266/Arduino/issues/5476#issuecomment-447153343).

What's strange to me is
> lwip Variant: [PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY_LOW_FLASH]

because it is Arduino IDE's option "lwIP-v2 (lower memory) (no features)", which has the same/default settings as in release 2.4.2 (although upstream lwIP has changed from 2.0 to 2.1.2 in between).

Do you have the same results with PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY (which is current default Arduino IDE configuration)?

You can also try to use previous lwIP version in core 2.5:
```
cd tools/sdk/lwip2
make download
cd builder
git checkout arduino-2.4.2
cd ..
make clean
make install
```
and replace `liblwip-536` by `liblwip_536` in `tools/platformio-build.py` L164

Next step would be to try the same SDK version as we had in 2.4.2.

TD-er, 2018-12-17:
> SensorSendTask and sendData are highlighted in 2.5 but not present in 2.4, but you posted them before.

I already found why that was in my (new installed) core 2.4.2 node.
I forgot to set the plugin interval timer for that one (ESPeasy config related), after setting that it is now reporting the timers for these function calls too.
[image]

> and replace liblwip-536 by liblwip_536 in tools/platformio-build.py L164

Isn't that the same or do I need some more coffee?
Can I also use a different define to switch LWIP or do I need to build it like you described?

d-a-v, 2018-12-17:
> Isn't that the same or do I need some more coffee?

Get more coffee :)

TD-er, 2018-12-17:
Running with `PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY` is the same:

Loop |   | 244840 | 1187.64 | 0.391 | 0.816 | 512.279
-- | -- | -- | -- | -- | -- | --

I will now test with previous LWIP in core 2.5.0 (just finished my coffee ;) )

TD-er, 2018-12-17:
Hmm, I cannot run `make download` like described.
```
gijs@Imagine:~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2$ make
all               builder/          clean             download          downloadmaster    downloadupstream  install           latestmaster      latestupstream
gijs@Imagine:~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2$ make download
make: Nothing to be done for 'download'.
```

d-a-v, 2018-12-17:
Then is it already downloaded in `builder/`
Is this directory empty ?

TD-er, 2018-12-17:
No it is not empty
Seems I don't have the complete build toolchain?

```
gijs@Imagine:~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2$ cd builder/
gijs@Imagine:~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2/builder$ git checkout arduino-2.4.2
M       lwip2-src
Previous HEAD position was c434c6f less verbose make (use standard V=1 to make it verbose)
HEAD is now at 163bb82 fix setting down interface on the esp side (ip must not be set to 0, and properly re/set FLAG_UP)
gijs@Imagine:~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2/builder$ cd ..
gijs@Imagine:~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2$ make clean
make -C builder -f Makefile.arduino clean
make[1]: Entering directory '/home/gijs/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2/builder'
make -f makefiles/Makefile.build-lwip2 \
                target=arduino \
                SDK=../.. \
                LWIP_ESP=../../lwip/include \
                LWIP_LIB=liblwip2.a \
                LWIP_LIB_RELEASE=../../lib/liblwip2.a \
                LWIP_INCLUDES_RELEASE=../include \
                TOOLS=../../../xtensa-lx106-elf/bin/xtensa-lx106-elf- \
                TCP_MSS=536 BUILD=build-536 \
                clean
make[2]: Entering directory '/home/gijs/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2/builder'
rm -rf build-536 liblwip2.a glue-lwip/lwip-err-t.h glue-lwip/lwip-git-hash.h
make[2]: Leaving directory '/home/gijs/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2/builder'
make -f makefiles/Makefile.build-lwip2 \
                target=arduino \
                SDK=../.. \
                LWIP_ESP=../../lwip/include \
                LWIP_LIB=liblwip2_1460.a \
                LWIP_LIB_RELEASE=../../lib/liblwip2_1460.a \
                LWIP_INCLUDES_RELEASE=../include \
                TOOLS=../../../xtensa-lx106-elf/bin/xtensa-lx106-elf- \
                TCP_MSS=1460 BUILD=build-1460 \
                clean
make[2]: Entering directory '/home/gijs/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2/builder'
rm -rf build-1460 liblwip2_1460.a glue-lwip/lwip-err-t.h glue-lwip/lwip-git-hash.h
make[2]: Leaving directory '/home/gijs/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2/builder'
make[1]: Leaving directory '/home/gijs/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2/builder'
gijs@Imagine:~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2$ make install
make -C builder -f Makefile.arduino install
make[1]: Entering directory '/home/gijs/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2/builder'
make -f makefiles/Makefile.build-lwip2 \
                target=arduino \
                SDK=../.. \
                LWIP_ESP=../../lwip/include \
                LWIP_LIB=liblwip2.a \
                LWIP_LIB_RELEASE=../../lib/liblwip2.a \
                LWIP_INCLUDES_RELEASE=../include \
                TOOLS=../../../xtensa-lx106-elf/bin/xtensa-lx106-elf- \
                TCP_MSS=536 BUILD=build-536 \
                install
make[2]: Entering directory '/home/gijs/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2/builder'
( \
echo "// script-generated, extracted from espressif SDK's lwIP arch/cc.h"; \
echo "#define LWIP_NO_STDINT_H 1"; \
grep -e LWIP_ERR_T -e ^typedef ../../lwip/include/arch/cc.h \
) > glue-lwip/lwip-err-t.h
makefiles/make-lwip-hash
make -f makefiles/Makefile.glue-esp BUILD=build-536 TCP_MSS=536
make[3]: Entering directory '/home/gijs/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2/builder'
../../../xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -std=gnu99 -Wall -Wextra -c -Os -g -Wpointer-arith -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -ffunction-sections -fdata-sections -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -DLWIP_OPEN_SRC -DLWIP_BUILD -DUSE_OPTIMIZE_PRINTF -DTARGET=arduino -DTCP_MSS=536 -Ibuild-536 -I../../include -I../../lwip/include -Iglue glue-esp/lwip-esp.c -o build-536/glue-esp/lwip-esp.o
make[3]: ../../../xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: Command not found
makefiles/Makefile.rules:7: recipe for target 'build-536/glue-esp/lwip-esp.o' failed
make[3]: *** [build-536/glue-esp/lwip-esp.o] Error 127
make[3]: Leaving directory '/home/gijs/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2/builder'
makefiles/Makefile.build-lwip2:47: recipe for target 'liblwip2.a' failed
make[2]: *** [liblwip2.a] Error 2
make[2]: Leaving directory '/home/gijs/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2/builder'
Makefile.arduino:6: recipe for target 'install' failed
make[1]: *** [install] Error 2
make[1]: Leaving directory '/home/gijs/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lwip2/builder'
Makefile:3: recipe for target 'install' failed
make: *** [install] Error 2
```

d-a-v, 2018-12-17:
Ah this is platformio, `builder/Makefile.arduino` would need an update of this variable
`TOOLS=../../../xtensa-lx106-elf/bin/xtensa-lx106-elf-` pointing to the right gcc directory.

You can also take already compiled binaries from [2.4.2 release](https://github.com/esp8266/Arduino/tree/2.4.2/tools/sdk/lib) (liblwip2.a - sorry in fact there is no dash/underscore, I'll get some coffee too)

In any case you need to add this line in `tools/sdk/ld/eagle.app.v6.common.ld.h`
```
    *liblwip2.a:(.literal .text .literal.* .text.*)                <- to add
    *liblwip2-536.a:(.literal .text .literal.* .text.*)            <- already present
```

TD-er, 2018-12-17:
I have downloaded the binary, added it to the ld.h file and also changed this line
> and replace liblwip-536 by liblwip_536 in tools/platformio-build.py L164
, only then into 'liblwip2
Also reset the `PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY_LOW_FLASH` parameter.

But the resulting binary does not boot.
Any other tips?

d-a-v, 2018-12-17:
> Any other tips?

I tried myself, and indeed it is not as simple as it seems. I will find some time to provide a lwIP-2.0.3 binary compatible with core 2.5.0 if the other leads are not conclusive.

Can you point to the source code of these `sendData` and `SensorSendTask` to check which parts of the core are involved ?

TD-er, 2018-12-17:
> 
> Can you point to the source code of these `sendData` and `SensorSendTask` to check which parts of the core are involved ?

These are my own functions and they take about the same time on both cores.
- [sendData](https://github.com/letscontrolit/ESPEasy/blob/ebe11246a1ba1f9120fe1f7eef73ae8bbba18f59/src/Controller.ino#L4)
- [SensorSendTask](https://github.com/letscontrolit/ESPEasy/blob/ebe11246a1ba1f9120fe1f7eef73ae8bbba18f59/src/ESPEasy.ino#L832)

`SensorSendTask `does call `sendData`. So if `sendData `takes quite some time, the other one will too.

In short:

`SensorSendTask `does collect the data from a plugin and performs some string replacements + calculates some formulas if needed.
This collecting data can also involve some reading from the sensor. Not all plugins in ESPeasy are yet converted to do the reading in the background, so some may do quite a lot while reading.

`sendData `does send the sensor data to one or more selected controllers and triggers some rules.
The rules are our own script language. At least it is about reading from SPIFFS and parsing rules (could benefit from some optimizations).
The rules parsing may also do a lot using String objects.

A controller is something which sends data to a remote computer, formatted for a specific use case (e.g. Domoticz MQTT, OpenHAB, Http, UDP, ESPeasy-p2p, etc.)
For almost all controllers, I have added some kind of buffer system to allow the controllers to send data (and retry) at some predefined pace specific for that controller. (e.g. Thingspeak only allows a message per N seconds for the free accounts and you don't want to wait for that)
So for most controllers, this sendData is mainly about storing it into a buffer.
The stored messages will later be processed by our own scheduler, so the actual network traffic is not being counted in the sendData call (except for the few plugins not yet using a buffer)

TD-er, 2018-12-17:
About the impact of those functions:
[image]
These timings are covering about 7900 seconds (2h19m)
There were 132 calls to these functions, which is about once a minute.
That's about 100 msec per minute. The rest of the time is spent in other areas of the code.

In a loop() it is mainly a call for the scheduler.
This scheduler does use STL containers. You had moved some stuff of that to flash? Could that affect performance?

TD-er, 2018-12-17:
As a test, I have changed the flash frequency to 80 MHz

Frequency | Description | #calls | call/sec | min (ms) | Avg (ms) | max (ms)
-- | -- | -- | -- | -- | -- | --
40 MHz | Loop | 10806208 | 1367.80 | 0.516 | 0.668 | 896.423
80 MHz | Loop | 441145 | 2554.65 | 0.303 | 0.374 | 258.775

Same node running same config @ core 2.5.0

earlephilhower, 2018-12-18:
>  This scheduler does use STL containers. You had moved some stuff of that to flash? Could that affect performance?

No, `std::*` was always in flash until I made it appear in IRAM it when I added the -nox library.  That wouldn't explain a difference.

d-a-v, 2018-12-19:
Can you instrument separately in `sendData` sub-bocks like (at least) this one `CPlugin_ptr[event->ProtocolIndex](CPLUGIN_PROTOCOL_SEND, event, dummyString);` and these two ones `LoadTaskSettings(event->TaskIndex);` (they are potentially taking time).

TD-er, 2018-12-19:
Adding timers you mean, so we can compare those parts?

d-a-v, 2018-12-19:
Yes

TD-er, 2018-12-19:
I added a bit more (commit will be pushed in a minute)
[image]

[image]

As can be seen, the handle_schedule function is taking quite a bit longer.

TD-er, 2018-12-19:
Running for about 10 minutes:

Core 2.5.0:

Description | Function | #calls | call/sec | min (ms) | Avg (ms) | max (ms)
-- | -- | -- | -- | -- | -- | --
Load File |   | 11 | 0.02 | 19.153 | 19.551 | 21.246
Loop |   | 842552 | 1309.42 | 0.479 | 0.737 | 391.926
Plugin call 50 p/s |   | 32103 | 49.89 | 2.291 | 2.385 | 2.727
Plugin call 10 p/s |   | 6429 | 9.99 | 2.572 | 2.633 | 2.924
Plugin call 10 p/s U |   | 6429 | 9.99 | 0.276 | 0.299 | 0.340
Plugin call 1 p/s |   | 644 | 1.00 | 2.928 | 3.301 | 22.692
SensorSendTask() |   | 11 | 0.02 | 98.892 | 102.294 | 107.163
sendData() |   | 11 | 0.02 | 96.084 | 99.625 | 104.560
Compute formula |   | 11 | 0.02 | 0.014 | 0.014 | 0.015
setNewTimerAt() |   | 42468 | 66.00 | 0.175 | 0.197 | 0.269
Delay queue MQTT |   | 11 | 0.02 | 1.847 | 1.868 | 1.885
WiFi.isConnected() |   | 2488246 | 3867.00 | 0.015 | 0.024 | 0.113
rulesProcessing() |   | 55 | 0.09 | 14.259 | 15.177 | 17.924
backgroundtasks() |   | 1643099 | 2553.55 | 0.138 | 0.189 | 391.306
handle_schedule() idle |   | 800107 | 1243.45 | 0.243 | 0.317 | 391.463
handle_schedule() task |   | 42446 | 65.97 | 0.383 | 2.767 | 107.564

Core 2.4.2:

Description | Function | #calls | call/sec | min (ms) | Avg (ms) | max (ms)
-- | -- | -- | -- | -- | -- | --
Load File |   | 10 | 0.02 | 18.154 | 18.473 | 19.223
Loop |   | 2454654 | 4049.56 | 0.151 | 0.237 | 259.272
Plugin call 50 p/s |   | 30267 | 49.93 | 1.323 | 1.409 | 1.856
Plugin call 10 p/s |   | 6060 | 10.00 | 1.456 | 1.550 | 2.033
Plugin call 10 p/s U |   | 6060 | 10.00 | 0.155 | 0.210 | 0.262
Plugin call 1 p/s |   | 606 | 1.00 | 2.017 | 2.295 | 15.314
SensorSendTask() |   | 10 | 0.02 | 67.307 | 72.162 | 78.346
sendData() |   | 10 | 0.02 | 65.414 | 70.261 | 76.438
Compute formula |   | 10 | 0.02 | 0.010 | 0.010 | 0.011
setNewTimerAt() |   | 40034 | 66.05 | 0.146 | 0.177 | 0.248
Delay queue MQTT |   | 10 | 0.02 | 1.567 | 1.615 | 1.644
WiFi.isConnected() |   | 7326792 | 12087.36 | 0.004 | 0.004 | 0.111
rulesProcessing() |   | 50 | 0.08 | 7.650 | 9.592 | 13.117
backgroundtasks() |   | 4869693 | 8033.77 | 0.031 | 0.059 | 258.065
handle_schedule() idle |   | 2414639 | 3983.55 | 0.065 | 0.098 | 3.018
handle_schedule() task |   | 40014 | 66.01 | 0.264 | 1.707 | 78.615

TD-er, 2018-12-19:
And the same nodes, running for 24'447 sec:

[image]

[image]

Some counters probably have had some overflow, since the average value displayed is lower than the minimum.
But the trend is very well visible.
Running code on the core 2.5.0 is slower than on core 2.4.2

earlephilhower, 2018-12-19:
Just a note, the graphic image posts are much harder to use than the text ones you started with.  Can't dump it into a .ODS or .XLS and compare automatically.

That said, since you know your code better than we do, do you have one specific section that's egregious and can be isolated?  I know your app is reporting fewer loops/s, but doing perf analysis at that level from the outside is not a fast process.

If there's some small, isolated thing (i.e. this routine X takes 100 numbers, averages them, and sends to a MQTT endpoint) where it's more approachable we'd stand a better chance.  Especially if it's something you can hack out into a separate sketch (even if the input and out data its sending is garbage), that means it can be repro'd and instrumented at the low-level needed to identify what's up and if it's in the core, a lib, of the SDK.  The smaller the code bit and fewer libs needed, the better, for this kind of thing.

TD-er, 2018-12-19:
I understand.
One of the striking things in these timing results to me is the delay in the rules handling.
The active rules are not communicating to the outside world, just reacting on an event and setting a GPIO pin.
There are 2 main things at work then:
- Reading from SPIFFS
- Parsing Strings (and lots of String operations)

Those String operations can do a lot of memory allocations too, which could also explain the difference when sending data to a controller, since it is then placed in some buffer.

So my 2 main suspects right now are:
- SPIFFS handling
- Memory allocation

I think for both a fairly small test could be written.
Did the SDK change or the core 2.5.0 update also include some changes in moving code to flash?
If more code is now executed from flash, that could also explain these slow-downs.

earlephilhower, 2018-12-19:
To make it clear, code was always in flash.  I introduced a bug in the -noexceptions option where instead of placing those std:: functions in flash it ate up IRAM.  IRAM is 16KB total, ~10KB taken by the hidden SDK functions, and the rest is used for interrupt-safe routines, certain library routines that may be called from the WiFi stack, etc.  Today the code is all landing in flash unless you say "place in IRAM" which is the correct behavior, and matches prior releases.

Memory management lives in `cores/esp8266/umm_malloc/umm_malloc.c` and is pretty much untouched in the past 2+ years AFAICT.  Couple bug fixes in the log, but not seeing any red flags.

SPIFFS can be very variable and slow in my experience.  I didn't dig into it, but it seems to have very high variability in response time even on the same hardware (so it's not flash speed related).  The git log's (`cores/esp8266/spiffs_api.c`) is also pretty much untouched for 3 years now.

Would love to get a test that can be used for isolated A/B comparisons, because I think it's elsewhere...

devyte, 2018-12-20:
@earlephilhower could something be hitting the SDK exception handler for reading unaligned addresses?

igrr, 2018-12-20:
Might be worth disabling unaligned access handler, at least in debug builds?

earlephilhower, 2018-12-20:
@devyte, if @TD-er's running the same code on both builds then the code would crash hard on the older version, so I think it's probably not too likely.

@igrr's point about disabling in debug builds might be good, but @TD-er's app most likely can't even build in debug mode (umm_malloc grows a lot w/heap checking and there are many IRAM/IRQ user routines in his build).

TD-er, 2018-12-20:
> if @TD-er's running the same code on both builds then the code would crash hard on the older version, so I think it's probably not too likely.

The only difference in both builds is the used set of build flags as defined [here](https://github.com/letscontrolit/ESPEasy/blob/a7f1d6153c640465e85e3d15fec8a3e3ecba34ae/platformio.ini#L72-L88)

> @igrr's point about disabling in debug builds might be good, but @TD-er's app most likely can't even build in debug mode (umm_malloc grows a lot w/heap checking and there are many IRAM/IRQ user routines in his build).

Maybe the "minimal" set could be built for a 4M flash environment. (using `-DPLUGIN_BUILD_MINIMAL_OTA` as build flag)
It uses the minimal amount of plugins.
Alternative is to use these flags:

- CONTROLLER_SET_NONE
- PLUGIN_SET_NONE
- NOTIFIER_SET_NONE

Then there is only the web interface, rule engine and almost no external libraries.

Do you have any suggestions on what to make a benchmark test for?

- String allocations/modifications
- Connection test
- Flash I/O
- Arduino calls to lookup pin states (just guessing here)

Then I can create a test which outputs to a page.
And what format do you like? CSV (valus separated by ';' )

devyte, 2018-12-20:
@TD-er I haven't tested anything here, but if I may, I suggest doing something along these lines:
For each entry in the table that shows timing is almost the same bbetween 2.4.2 and 2.5.0 beta2, remove that test from the sketch.
For each entry that shows a runtime that is too short, remove that test as well.
Remove other things that also make sense, like you propose above with plugin none, etc.
From whatever is left, pick one test, e.g. with the highest timing diff, go inside, and measure time wherever it makes sense to do so. Get new tables to compare versions.
Follow the above recursively until it gets narrowed down to some core function call.

earlephilhower, 2018-12-20:
@TD-er, I think @devyte's suggestion is pretty low-effort on your part and would eliminate one potential source, so maybe we can give that a go?

It looks like the unaligned access stuff is installed at runtime, not link, by the OS.  A straightforward if inelegant way to disable this:

Remove user_exceptions.o from libmain.a
````
cd tools/sdk/lib
cp libmain.a libmain.bak
ar d libmain.a user_exceptions.o
````

Then in your main .ino add
````
extern "C" {
extern void __wrap_system_restart_local();
void load_non_32_wide_handler()
{
    __wrap_system_restart_local();
}
};
````

It'll then do a stack dump on misaligned accesses and pinpoint the offending routine if PMEM is being used as byte-wide memory.

TD-er, 2018-12-20:
I can do that later this evening.
But then I also must make sure I can actually read the stackdumps.
Last time I checked, I was not able to run the Arduino-IDE plugin to decode stack dumps. (Windows 10)

Would be nice to have that option though, since now I am "debugging-in-the-dark", which takes a lot of time.

earlephilhower, 2018-12-20:
We've got you covered! :)  One of the fixes in 2.5.0 (and part of the reason for the new toolchain build) was that GDB.EXE wasn't included in the espressif toolchain bundle.  GDB is required to decode address->line of code in the EspExceptionDecoder.

The new toolset has GDB for all platforms, so this should "just work" (assuming you're using ESPExceptionDecoder).  If not, please let us know in a new bug.

TD-er, 2018-12-24:
Just a small update from my side.
My mind was not really into debugging this the last few days, so that's why there's no update yet.
I hope to get to it later this week.

earlephilhower, 2019-01-03:
@TD-er, hope you had a good break.  Do you have any more info to share?

d-a-v, 2019-01-17:
@TD-er 
Can you comment/disable the update of `ummHeapInfo.freeSize2` in `cores/esp8266/umm_malloc/umm_malloc.c` and try your perf test again ?

TD-er, 2019-01-17:
@d-a-v 
Changed it to:
```c++
    if( UMM_NBLOCK(blockNo) & UMM_FREELIST_MASK ) {
      ++ummHeapInfo.freeEntries;
      ummHeapInfo.freeBlocks += curBlocks;
/*
      ummHeapInfo.freeSize2 += (unsigned int)curBlocks
                             * (unsigned int)sizeof(umm_block)
                             * (unsigned int)curBlocks
                             * (unsigned int)sizeof(umm_block);
*/
```

It is now building.

TD-er, 2019-01-17:
It is running now on 2 nodes.
"Breadboard2" with:
- BME280
- OLED Framed plugin

"CO2_tracker1" with:
- GPS plugin
- BME280
- Senseair S8 CO2

On the "Breadboard2" 
- before:  51.20% (LC=1171)
- now: 	49.10% (LC=963)

On "CO2_tracker1":
- before:  32.10% (LC=1757)
- now: 	31.10% (LC=1844)

N.B. the CPU load is based on the number of loops not needed to run a set task. The amount of idle time is summed and used to compute the CPU load.

See the comparison in this [Excel sheet](https://www.dropbox.com/s/uki77a1w2sp8iya/Benchmark_core2_5_0_20190117.xlsx?dl=0)

d-a-v, 2019-02-17:
Will you be able to try with #5763 ?

TD-er, 2019-02-17:
So I just build with a define for `NONOSDK221` ?

TD-er, 2019-02-17:
I tried building it several times (PlatformIO staged, so it should use your master branch), but I keep getting this as a build: `ESP82xx Core 2.6.0-dev, NONOS SDK 3.0.0-dev(c0f7b44), LWIP: 2.1.2 PUYA support`
N.B. The "Core 2.5.0-dev" is something I add in the build.
I added the `NONOSDK221` as a define in the build.
I also tried to add `board_build.sdk           = NONOSDK221` to the platform.ini section for this build, but it keeps reporting the NONOS SDK 3.0.0-dev.

So can you give me some hints on what I need to do to make it building a NONOS SDK 2.2.1?

d-a-v, 2019-02-17:
include and library paths must be updated
check for patches to  platform.txt and tools/pio.py files

TD-er, 2019-02-17:
Not sure why, but the files I get when using PlatformIO feature/stage is not having the latest changes you have in that PR.

Sometimes I just feel a bit lost in all abstractions made for PlatformIO to make it all "easier" to work with.
I still don't know what is configured where, what files are defined where and collected from where.
It is a bit too obfuscated to see through at once and I feel like I need to make some drawing or something to get where all parts are coming from.
These abstractions are great for normal workflows, but if you want to use a specific part from the core libs, it isn't that clear where to set things to get it right.

TD-er, 2019-02-17:
Ah, now I get it, it wasn't merged yet.
Will fetch this PR manually and try to build then

TD-er, 2019-02-17:
OK, I did a checkout of your PR and built it.
But still the "NONOS SDK 3.0.0-dev(c0f7b44), LWIP: 2.1.2 PUYA support"
So what do I need to change, and where is it located?
This is the top of my `git log` : 
```
commit d0e49a65500c0528bc4063f841971bfd1b203b55 (HEAD)
Merge: 14f1b1d4 3dd54fd3
Author: david gauchard <gauchard@laas.fr>
Date:   Sun Feb 17 04:29:06 2019 +0000

    Merge 3dd54fd3689636d95fa3eb3328d39f2f536979b9 into 14f1b1d4a74a2e65207f23aedabcc7038ffae652
```

d-a-v, 2019-02-17:
With platformio,
you need to change in `tools/platformio-build.py` the two occurences of `NONOSDK3V0` to `NONOSDK221`.
I don't know how it could be made better, not being a pio user.

TD-er, 2019-02-17:
That worked :)

`ESP82xx Core 2.6.0-dev, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support`

I will let it run for a few minutes to see what the loop-count is and get you an Excel sheet with before- and after timings of that node.

TD-er, 2019-02-17:
And here the comparison of the 3 versions I already mentioned in the previous post.
[Compare_core250_SDK3-0-0_vs_SDK2-1-1.zip](https://github.com/esp8266/Arduino/files/2873262/Compare_core250_SDK3-0-0_vs_SDK2-1-1.zip)

Swiftnesses, 2019-02-17:
Do you also get slow ping responses on SDK 3.0.0?

TD-er, 2019-02-17:
@Swiftnesses 
Both nodes run the same hardware & config, main difference is the core lib version. 
These are both Sonoff POW r2 nodes.

```
C:\Users\gijs>ping -n 10 192.168.1.112

Pinging 192.168.1.112 with 32 bytes of data:
Reply from 192.168.1.112: bytes=32 time=2ms TTL=255
Reply from 192.168.1.112: bytes=32 time=2ms TTL=255
Reply from 192.168.1.112: bytes=32 time=2ms TTL=255
Reply from 192.168.1.112: bytes=32 time=3ms TTL=255
Reply from 192.168.1.112: bytes=32 time=4ms TTL=255
Reply from 192.168.1.112: bytes=32 time=5ms TTL=255
Reply from 192.168.1.112: bytes=32 time=2ms TTL=255
Reply from 192.168.1.112: bytes=32 time=18ms TTL=255
Reply from 192.168.1.112: bytes=32 time=3ms TTL=255
Reply from 192.168.1.112: bytes=32 time=2ms TTL=255

Ping statistics for 192.168.1.112:
    Packets: Sent = 10, Received = 10, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 2ms, Maximum = 18ms, Average = 4ms

C:\Users\gijs>ping -n 10 192.168.1.98

Pinging 192.168.1.98 with 32 bytes of data:
Reply from 192.168.1.98: bytes=32 time=4ms TTL=255
Reply from 192.168.1.98: bytes=32 time=1ms TTL=255
Reply from 192.168.1.98: bytes=32 time=1ms TTL=255
Reply from 192.168.1.98: bytes=32 time=3ms TTL=255
Reply from 192.168.1.98: bytes=32 time=1ms TTL=255
Reply from 192.168.1.98: bytes=32 time=3ms TTL=255
Reply from 192.168.1.98: bytes=32 time=2ms TTL=255
Reply from 192.168.1.98: bytes=32 time=1ms TTL=255
Reply from 192.168.1.98: bytes=32 time=3ms TTL=255
Reply from 192.168.1.98: bytes=32 time=2ms TTL=255

Ping statistics for 192.168.1.98:
    Packets: Sent = 10, Received = 10, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 1ms, Maximum = 4ms, Average = 2ms
```

- .98 => ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
- .112 => ESP82xx Core 2.5.0-beta2, NONOS SDK 3.0.0-dev(c0f7b44), LWIP: 2.1.2 PUYA support

d-a-v, 2019-02-17:
```
First impression:
    * core 2.4.2: 12.90% (LC=6134)
    * SDK 3.0.0: 19.20% (LC=2471)
    * SDK 2.2.1: 12.40% (LC=6902)
```
Per #5736, this thread, my personal experience with esp8266Audio (with which performance difference can be heard), I think we should revert.

However current sdk-v3 brings some wifi fixes especially noticed by tasmota project @ascillato .

If we had to revert, it could be with #5763 (default: sdk2.2.1, allowing to use sdk-pre-3.0.0 when needed).

TD-er, 2019-02-17:
Is it possible to create some 'board' definition which will force it to use the SDK 2.2.1?
Then we can build both and just switch between the SDK versions.
This way the building of both will be tested for a longer period and users will be able to test both.
Or some flag in PlatformIO definition to make it easy to define a build to use a specific SDK version.

What specific WiFi changes/patches are included in SDK 3.x, which are so useful?
I know @arendst mentioned deep sleep mode a few times, are these fixes related to that?

d-a-v, 2019-02-18:
> Is it possible to create some 'board' definition which will force it to use the SDK 2.2.1?

A menu only for generic esp8266 board could be added. And sdk back to 2.2.1 for all other boards.

> Or some flag in PlatformIO definition to make it easy to define a build to use a specific SDK version.

A flag can be added in pio.

d-a-v, 2019-02-19:
#5763 is ready to test.

> some flag in PlatformIO definition to make it easy to define a build to use a specific SDK version.

Included (search for SDK in platformio-build.py). I am not able to test with PIO.

TD-er, 2019-02-19:
Juist building based on PR #5763 results in a build with this SDK:
`ESP82xx Core 2.6.0-dev, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support`

And by just adding this single define in the platformio.ini `PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK3` I get a build with this SDK:
`ESP82xx Core 2.6.0-dev, NONOS SDK 3.0.0-dev(c0f7b44), LWIP: 2.1.2 PUYA support`

Just like this in out platformio.ini file:
`build_flags               = ${esp82xx_2_5_0.build_flags} -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK3`

So it seems to be working and quite easy to switch.
As soon as this has been merged, I will add the SDK3.0 to our nightly builds (marked as 2.6.0 alpha)

Will this merge lead to a "2.5.1" label?

d-a-v, 2019-02-19:
Thanks for testing

> Will this merge lead to a "2.5.1" label?

Yes I think there will be a fix release, though I can't exactly say when

dm5xx, 2019-04-06:
Hi guys. I tried to change to 2.5.0 today and realized the same thing. 2.4.1 all was fast, switching to 2.5.0, wemos d1 mini was slooooow and with high latency. mike.

devyte, 2019-11-10:
@TD-er this issue doesn't apply to our current builds due to our using SDK 2.2.x . It would apply if we were to switch to SDK 3.x, but I rather cross that bridge when we get to it.
I'm closing this for now. Let's revisit when we look at SDK 3.x again.

closed 2019-11-10 by devyte