esp8266/Arduino #5514: Mdns problem

mustafayuce33, 2018-12-17:
I can see my device on avahi browser like programs with previous version of SDK. When I tried with the version https://github.com/esp8266/Arduino/tree/92373a98370618dea09718010b30d311a97f3f25
then there is no problem. When I tried with the latest version then i cant see my device on avahi browser.

d-a-v, 2018-12-17:
92373a9 is 2.5.0-beta1. mDNS has been fixed in 2.5.0-beta2, please try it,
and if the problem persists, please **fill the issue template** in a new issue.

closed 2018-12-17 by d-a-v

Shayko94, 2019-04-02:
Same problem with mDNS and dns-sd in windows bonjour... last version which works properly is 2.4.2 . 
Newer versions works for linux but not for windows (bonjour).