esp8266/Arduino #5521: Which SDK version is 2.5-beta2 based on?

marcelstoer, 2018-12-18:
Simple question triggered by this comment https://github.com/espressif/ESP8266_NONOS_SDK/issues/8#issuecomment-448153689

Which SDK version is 2.5-beta2 based on?

The [version file](https://github.com/esp8266/Arduino/blob/2.5.0-beta2/tools/sdk/version#L1) says 2.2.0 something while `system_get_sdk_version()` appears to be saying 3.0.0 something. I tend to believe the latter 😉

d-a-v, 2018-12-18:
It's not real 3.0.0
git hash is in [`tools/sdk/version`](https://github.com/esp8266/Arduino/blob/master/tools/sdk/version).
We cherry-picked the "last" v2 before v3 taste,
```
commit 89920dcc40aee30da11899c832f311e381312b38 (HEAD -> pre3)
Author: Espressif Systems <wujiangang@espressif.com>
Date:   Tue Jun 26 17:43:01 2018 +0800
```
right before the final serie of commits for v3 and major internal SDK / API changes.

All following commits have another author,
```
commit cb46633acae2577c68dd7c8d97f34566cfc95b96
Merge: 1498a86 e8d53c1
Author: Xu Chun Guang <xuchunguang@espressif.com>
Date:   Thu Jul 12 21:21:38 2018 +0800
```
The signature of the new era is very visible with `gitk`. Tube/subway map starts right after that.
I bet on this version for a try, and we finally kept it, until we are ready for v3.
I don't remember whether that was before or after they added this `pre3` tag.

closed 2018-12-18 by devyte

marcelstoer, 2018-12-19:
@d-a-v thanks for the explanations. Pity the SDK version is embedded in a SDK binary (tools/sdk/lib/libmain.a?) and can't be changed by you. It's not good that we have to live with this inconsistency.

d-a-v, 2018-12-19:
To keep track on dependencies and API changes, there is a new constant, visual, comparable with `#if` or evaluated at compile time with `if()`, increasing with every commits:
`constexpr int coreVersionNumeric()` in [`core_esp8266_version.h`](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_version.h).
```
// Examples:
// case 2.4.2 (fresh version/tag)   20402000
// case 2.4.2-91-gcb05b86d:         20402091
// case 2.5.0-beta3-1-gcb05b86d:    20499903
// case 2.5.0:                      20500000
```