esp8266/Arduino #5524: esp8266.a: No such file or directory

domenic-d, 2018-12-19:
#### Platform
- Hardware:				ESP-12E
- Development Env:		Arduino IDE (1.88)
- Operating System:		MacOS

### Settings in IDE
- Module:  			NodeMCU ESP-12E
- Flash Size: 		[4MB/1MB]
- lwip Variant:   Higher Bandwidth
- Reset Method:		nodemcu
- Flash Frequency:	[40Mhz]
- CPU Frequency:		160MHz
- Upload Using:		OTA
- Upload Speed:   115200

#### Extra Info:
This is a fresh install of Arduino IDE and libraries.
Library installation steps were followed from online documentation.
This error has been reproduced on two separate computers.

### Problem Description
I am receiving the following error during sketch compilation

xtensa-lx106-elf-gcc: error: /var/folders/8q/zh68fx3s74xc_4pv3tfp7jcw0000gn/T/arduino_build_382396/libraries/esp8266/esp8266.a: No such file or directory

The sketch I have been attempting to compile is wifimanager.
https://github.com/tzapu/WiFiManager

Any suggestions on possible cause are welcomed.

domenic-d, 2018-12-19:
I have managed to resolve this error by reverting to ESP8266 Community version 2.4.1.
Any version above that introduces missing file errors.

klrock, 2019-01-20:
I have exactly the same issue.  I tried the suggested solution of using ESP8266 Community version 2.4.1, but it did not change anything.  I have searched the web and found nothing other than this issue.  I have painstakingly searched all the places on my PC where ESP8266 libraries are found and did not locate esp8266.a.  I am now 1.5 days into this effort and have run out of ideas.  

Any suggestions?

domenic-d, 2019-01-20:
It's a really frustrating problem. I too spent a couple of days trying to resolve this without success.
I honestly do not know what to suggest for you. The problem resolved itself for me when I reverted back to a version of libraries I was using over a year ago.

Perhaps one of the developers will read this and provide some direction on what is causing this and how to resolve?

d-a-v, 2019-01-20:
Did you try to reinstall everything from scratch ?

klrock, 2019-01-20:
I found this in the compile comments

Using library esp8266 at version 1.0 in folder: C:\Other\Arduino\hardware\esp8266com\esp8266\libraries\esp8266

I checked the Arduino Library Manager and found that ESP8266 Built-in version 1.0.0 was installed.  But there was no option to uninstall it or to update to a newer version.  Should I just replace the folders manually with the newer version or is there a better way to do it?

d-a-v, 2019-01-21:
I suggest you delete/rename your installation and follow the instructions
https://arduino-esp8266.readthedocs.io/en/latest/
Don't forget to remove/rename also the `arduino15` directory
https://forum.arduino.cc/index.php?topic=362507.msg2498993#msg2498993

klrock, 2019-01-21:
d-a-v was right delete and reinstall fixed the issue.
Thanks d-a-v.

merchination, 2019-01-26:
My pc isn’t recognizing them at all however any other device it does so I think it’s the devices themselves are fried.

earlephilhower, 2019-01-31:
Looking through this it seems like an installation issue that's cleared up per several comments.  Closing.

closed 2019-01-31 by earlephilhower

merchination, 2019-01-31:
it actually wasnt but ok...

 
 
Closed #5524.

acoulson2000, 2019-09-21:
I'm pretty sure this is still and issue. I consistently get the problem with the latest ESP8266 tooling.

For example:
xtensa-lx106-elf-gcc: error: C:\Users\acoulson\AppData\Local\Temp\arduino_cache_63660\core\core_Heltec-esp8266_esp8266_generic_xtal_80,vt_flash,exception_disabled,ResetMethod_nodemcu,CrystalFreq_26,FlashFreq_40,FlashMode_dout,eesz_4M,ip_lm2f,dbg_Disabled,lvl_None____,wipe_none,baud_115200_fb446bf806028fa10ee7813456ed1bea.a: No such file or directory

It's pretty clear that build process is creating these longs path names based on the board options and exceeding the Windows 260 character limitation. The example above, for example, is 295 characters long. I can't even delete the file via Windows explorer or command line.

timmost, 2019-10-19:
This problem is still there; works with "esp8266 by ESP8266 Community Version 2.4.2 " but not with Version 2.5.2 is available. same error "esp8266/esp8266.a: No such file or directory"
Best Timmo

acoulson2000, 2019-10-19:
I'm pretty sure it's showing up in newer releases because so many build options have been added and they all get reflected in what winds up being a very long file name. I'm not sure why they build the filename this way instead of just using a hash or something.

timmost, 2019-10-19:
well, the build command itself contains the file name - so it looks to me that this specific file is missing. I also installed python, (unrelated), it asked me if I want to extend filename contraints of Win10, but that did not change anything.

is there a possibility to reopen this issue?

`"C:\Users\Timmo Sturm\AppData\Local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-gcc" -fno-exceptions -Wl,-Map "-Wl,C:\Users\Timmo Sturm\AppData\Local\Temp\VMBuilds\Sketch1\esp8266_d1\Debug/Sketch1.ino.map" -g -w -Os -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static "-LC:\Users\Timmo Sturm\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lib" "-LC:\Users\Timmo Sturm\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lib/NONOSDK221" "-LC:\Users\Timmo Sturm\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/ld" "-LC:\Users\Timmo Sturm\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/libc/xtensa-lx106-elf/lib" "-Teagle.flash.4m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read -o "C:\Users\Timmo Sturm\AppData\Local\Temp\VMBuilds\Sketch1\esp8266_d1\Debug/Sketch1.ino.elf" -Wl,--start-group "Sketch1.cpp.o" "esp8266\esp8266.a" "VM_DBG\VM_DBG.cpp.o" "VM_DBG\VM_mem_check_sam.cpp.o" "C:\Users\Timmo Sturm\AppData\Local\Temp\VMBuilds\Sketch1\esp8266_d1\Debug\core.a" -lhal -lphy -lpp -lnet80211 -llwip2-536-feat -lwpa -lcrypto -lmain -lwps -lbearssl -laxtls -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group "-LC:\Users\Timmo Sturm\AppData\Local\Temp\VMBuilds\Sketch1\esp8266_d1\Debug"
 
xtensa-lx106-elf-gcc*: error: esp8266\esp8266.a: No such file or directory`

timmost, 2019-10-23:
Opened a new issue #6663 to restart discussion.