esp8266/Arduino #5532: Code Question

Mondeo20, 2018-12-20:
Hi,

I compile the source and it give me a warning:

...arduino/sonoff/support_wifi.ino:252:67: warning: statement has no effect [-Wunused-value]
     if (best_network_db < -WIFI_RSSI_THRESHOLD) { best_network_db +WIFI_RSSI_THRESHOLD; }

Which is true, makes no sense. Can somebody saying if this line is needed and if yes what is the correct line.

Mondeo20, 2018-12-20:
Here the next code which makes no sense:

xdrv_19_ps16dz_dimmer.ino:156:82:

if (ps16dz_byte_counter || (!ps16dz_byte_counter && serial_in_byte == 'A'));
      ps16dz_rx_buffer[ps16dz_byte_counter++] = serial_in_byte;

d-a-v, 2018-12-21:
@Mondeo20 You should address your questions or propose pull-requests to tasmota project.
This does not concern the esp8266 arduino core.

closed 2018-12-21 by d-a-v

Mondeo20, 2018-12-21:
Right, my fault.  I thought I was there