esp8266/Arduino #5533: Can't connect to open network with empty string passphrase

frishrash, 2018-12-21:
When empty string is passed as passphrase:
https://github.com/esp8266/Arduino/blob/4c04c63c2a7773ac92b0860735d62e79a66b83bb/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp#L218

to WiFi.begin():
https://github.com/esp8266/Arduino/blob/4c04c63c2a7773ac92b0860735d62e79a66b83bb/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp#L129

My ESP8266 (Wemos D1 mini) can't connect to my open access point (OpenWRT).
_ESP8266WiFiMulti::run()_ returns _WL_DISCONNECTED_ with _WiFiDisconnectReason_ of _WIFI_DISCONNECT_REASON_NO_AP_FOUND_.

When I change the passphrase to _NULL_ everything works.

d-a-v, 2018-12-21:
Can you try again with latest git, #5516 might be related.

devyte, 2018-12-30:
It is. Closing.

closed 2018-12-30 by devyte