esp8266/Arduino #5534: DNSServer does not work with dig unix tool

d-a-v, 2018-12-21:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			latest git
- Development Env:		Arduino IDE
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			qio
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   921600

### Problem Description

DNSServer library has issues.
Related to #5529, more details in [this comment](https://github.com/esp8266/Arduino/pull/5529#issuecomment-449373239).
Also related with the much older #3357.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

DNSServer example (Files>Examples>DNSServer>DNSServer)

### Debug Messages

On unix:
```
$ ping www.example.com
PING www.example.com (192.168.1.1) 56(84) bytes of data.
64 bytes from 192.168.1.1: icmp_seq=1 ttl=255 time=2.07 ms
64 bytes from 192.168.1.1: icmp_seq=2 ttl=255 time=2.37 ms
```
Here, working (not always when following the #5529 comment above)
This warning + no answer must be solved:
```
$ dig @192.168.1.1 www.example.com
;; Warning: Message parser reports malformed message packet.

; <<>> DiG 9.11.3-1ubuntu1.3-Ubuntu <<>> @192.168.1.1 www.example.com
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: SERVFAIL, id: 59041
;; flags: qr rd ad; QUERY: 0, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1
;; WARNING: recursion requested but not available

;; Query time: 4 msec
;; SERVER: 192.168.1.1#53(192.168.1.1)
;; WHEN: Fri Dec 21 13:15:26 CET 2018
;; MSG SIZE  rcvd: 12
```

SimonWilkinson, 2019-01-01:
This issue is down to DNSServer not understanding the extra EDNS information that dig is including in its request, and then getting the error response wrong.

EDNS uses an OPT RR in the additional records section. DNSServer expects there to be 0 additional records with an incoming query and so fails the request. However, there are two separate issues with the error response that it returns:
   a) It should return a FORMERR to make dig retry the request without the EDNS OPT record. Instead, it returns whatever custom error code the user has set
   b) It copies the header from the query when composing the response, but doesn't reset the ARCount, so claims to be returning 1 additional record when it never copies the additional record into the response payload.

These two issues will probably break any EDNS compatible resolver that doesn't work around broken servers.

I started digging into this, and have actually ended up restructuring all of the request handling code in DNSServer to make it more robust. I'll push a pull request for that once I've completed testing it.

devyte, 2019-01-05:
Closed via #5573 .

closed 2019-01-05 by devyte