esp8266/Arduino #5536: Web server in 2.5 Beta not working

rogerp67, 2018-12-21:
Hi,

I installed the latest beta (2.5.. Beta) and suddenly my response to webserver on the Arduino became "400, File not found".
After setting the Arduino esp8266 software back to 2.4.2 it worked again correctly. :)
(FYI: I have a WEMOS D1 Lite by the way and use the webserver library to provide a config web page and an API.)

So I suppose there is something to investigate amd correct

Good luck and thanks!

d-a-v, 2018-12-21:
`"400, File not found".` is spiffs right ?
Maybe you could
- enable logs (Tools>Debug)
- provide more details
- check spiffs
- update the title

earlephilhower, 2019-01-14:
24 days with no feedback or details, so closing.  Reopen if you get the requested info, or better yet follow the template that appears when you open a new bug.

closed 2019-01-14 by earlephilhower