esp8266/Arduino #5544: Soft WDT reset when using hardware serial

Erik84750, 2018-12-23:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- x[ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12-F]
- Core Version:			[2.4.2]
- Development Env:		[Arduino IDE 1.8.2]
- Operating System:		[Windows 7 64bit]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

Soft WDT reset

Hardware serial at 1200 baudrate causes soft WDT reset when more than 12 Serial.print(); lines are sent to the serial monitor unless yield(); is inserted every 10 lines.
Why would hardware serial cause WDT timeouts?

ESP library used: ESP8266WiFi.h

Example of serial output used, with yield(); at regular intervals (yield(); as used here seems to prevent soft WDT reset):

[code]
    yield();
    Serial.print(F(" mov.avg. duty cycle A: "));
    Serial.print(currentDutyCycleA); // battery voltage
    Serial.println(F(" %"));
    Serial.print(F(" mov.avg. temperature A: "));
    Serial.print(currentTemperatureA);
    Serial.println(F(" °C"));
    Serial.print(F(" mov.avg. battery voltage A: "));
    Serial.print(currentBatteryVoltA);
    Serial.println(F(" V"));
    Serial.print(F(" current minute A: "));
    Serial.print(currentMinuteA);
    Serial.println();
    yield();
    Serial.print(F(" mov.avg. duty cycle B: "));
    Serial.print(currentDutyCycleB); // battery voltage
    Serial.println(F(" %"));
    Serial.print(F(" mov.avg. temperature B: "));
    Serial.print(currentTemperatureB);
    Serial.println(F(" °C"));
    Serial.print(F(" mov.avg. battery voltage B: "));
    Serial.print(currentBatteryVoltB);
    Serial.println(F(" V"));
    Serial.print(F(" current minute B: "));
    Serial.println(currentMinuteB);
[/code]

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {
  Serial.begin(1200);
}

Erik84750, 2018-12-23:
Sorry, I tried unsuccesfully to edit out the unnecessary comments on top of above message.

devyte, 2018-12-23:
Because hwserial is synchronous for Tx, I.e. when you send a buffer, it actually waits until the buffer is out to the wire before returning. At low speed, if you send a bunch of times, that takes a long time, and if nobody kicks the watchdog, you get a reset.
So I'd say it's expected behavior.

devyte, 2018-12-23:
BTW, you easily have about 120 chars in all that text before the first yield, which @1200 takes about 1s, so again expected behavior.
Also, the hwserial code doesn't call yield internally because it's meant for use in both execution contexts (you can't call yield in SYS context).
Closing.

closed 2018-12-23 by devyte