esp8266/Arduino #5547: Ajuda com POST especial HTTP Server

laercionit, 2018-12-24:
Hello .. Good night.

I was using release 2.4.2, I upgraded to release 2.5.0-Beta2

Unfortunately I can not give the post below:
http://192.168.1.118/command?print=Func1 (); Funct2 (); Func3 ();

When I command listing the arguments this is excluding the semicolon, this works fine beforehand. Used to send commands, the post is working normally, only it is ignoring the semicolon.

5chufti, 2018-12-24:
where is the filled in template you were offered when opening the issue?
where is the MCVE sketch?
If you are not willing to spend the little time for complying with the minimal requirements to get your problems solved, why should anyone spend much more time trying to reproduce and solve **your** problem?

laercionit, 2018-12-25:
Ola @5chufti , peço desculpas por não ter sido tão especifico e até mesmo ter deixando os requisítos do issue.

- Hardware:			ESP-12E
- Core Version:		2.5.0-Beta2
- Development Env:	Arduino IDE 1.8.6
- Operating System:	Windows

- Module:  	NodeMCU 1.0
- Flash Mode: qio
- Flash Size: 	4MB
- lwip Variant:   v2 Lower Memory
- Reset Method: 	nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency: 80Mhz|
- Upload Using:    SERIAL
- Upload Speed:   115200

### Problem Description

**Ignore semicolon on args**

###Sketch

``
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

const char* ssid = "........";
const char* password = "........";

ESP8266WebServer server(80);

const int led = 13;

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp8266!");
  digitalWrite(led, 0);
}

void handleNotFound() {
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void) {
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
  server.handleClient();
}
```

### Debug Messages
```
File Not Found
URI: /command
Method: POST
Args: 3 
print: Func1 () ****
Funct2 ()**:** 
Func3 ()**:**
```

dragondaud, 2018-12-25:
[RFC 1738](https://www.ietf.org/rfc/rfc1738.txt) states only certain characters are allowed in a URL, and the semicolon is NOT one of them. Encode it (%3B) and it will work. See this [simple explanation](https://www.freeformatter.com/url-encoder.html#url-encoding-explained).

laercionit, 2018-12-25:
Hello @dragondaud and @dragondaud

Thanks for the info, but unfortunately my webserver depends on commands per post with semicolon separation. Changing this now would require a lot of time throughout the web interface and javascripts.

How can I only remove this restriction? It worked fine in version 2.4.2

dragondaud, 2018-12-25:
The semicolon is not valid in a URL, and the core is standards complaint and unlikely to be changed. It would certainly be better, in the long run, for you to fix your non-compliant server instead of expecting others to change core code to accommodate your broken client. But, you are welcome to dig into the code a break the sanitizing in your own private version. Please close ticket, there is no issue in the core.

closed 2018-12-25 by laercionit