esp8266/Arduino #5562: Unable to pass String as an argument to  'BearSSL::WiFiClientSecure::write()

mih380, 2018-12-28:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.4.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Adafruit Feather Huzzah ESP8266]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[160MHz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[|SERIAL]
- Upload Speed:   [115200] 

### Problem Description

I am trying to send JSON data to my AWS IOT instance running with MQTT as a broker. I am able to establish a secure connection and send simple text data. As I move forward I wold like send complete JSON arras to the server using the write method. I have tried below mentioned code and used passed String type variable as an argument. My code does not compile as the library does not have a corresponding matching method for write. Is there a method through which I can send JSON / String stored in a variable directly? 

```cpp

#include <ESP8266WiFi.h>
#include <time.h>
#include <ArduinoJson.h>

const char *ssid = "SSID";
const char *pass = "PASSWORD";

const char *   host = "a2sy7u3nxkdvrh-ats.iot.ap-southeast-1.amazonaws.com";
const uint16_t  port = 8443;
const char *   path = "/topics/mj?qos=1";
StaticJsonBuffer<200> jsonBuffer;

// Set time via NTP, as required for x.509 validation
time_t setClock() {
  configTime(3 * 3600, 0, "pool.ntp.org", "time.nist.gov");

  Serial.print("Waiting for NTP time sync: ");
  time_t now = time(nullptr);
  while (now < 8 * 3600 * 2) {
    delay(500);
    Serial.print(".");
    now = time(nullptr);
  }
  Serial.println("");
  struct tm timeinfo;
  gmtime_r(&now, &timeinfo);
  Serial.print("Current time: ");
  Serial.print(asctime(&timeinfo));
  return now;
}

void fetchURL(BearSSL::WiFiClientSecure *client, const char *host, const uint16_t port, const char *path) {
  if (!path) {
    path = "/";
  }

 
  //root.printTo(Serial)
 
  Serial.printf("Trying: %s:8443...", host);
  client->connect(host, port);
  //Stream Char = {"Hello World"};
  if (!client->connected()) { 
    
    Serial.printf("*** Can't connect. ***\n-------\n");
    Serial.printf("---------- \n"); 
    Serial.printf("client->getLastSSLError() output is : ");
    Serial.println (client->getLastSSLError());  
    Serial.printf("\n ----------");
    
    return;
  }
  Serial.printf("Connected!\n-------\n");

  
  JsonObject& root = jsonBuffer.createObject();
  root["sensor"] = "gps";
  root["time"] = 1351824120;
  String result;
  String conLength;
  root.printTo(result);
  int len1 = result.length();
    
  
  client->write("POST ");
  client->write(path);
  client->write(" HTTP/1.0\r\nHost: ");
  client->write(host);
  client->write("\r\nContent-Type: application/json\r\n");
  client->write("Content-Length: " + len1);
  client->write("\r\n");
  client->write("\r\n");
  client->write(result);
  
  
  uint32_t to = millis() + 5000;
  if (client->connected()) {
    do {
      char tmp[32];
      memset(tmp, 0, 32);
      int rlen = client->read((uint8_t*)tmp, sizeof(tmp) - 1);
      yield();
      if (rlen < 0) {
        break;
      }
      // Only print out first line up to \r, then abort connection
      char *nl = strchr(tmp, '\r');
      if (nl) {
        *nl = 0;
        Serial.print(tmp);
        break;
      }
      Serial.print(tmp);
    } while (millis() < to);
  }
 
  Serial.printf("client->getLastSSLError() output is : ");
  Serial.println (client->getLastSSLError());  
  Serial.printf("\n ----------");
  client->stop();
  Serial.printf("\n-------\n\n");
}

void fetchCertAuthority() {
  static const char ca_cert[] PROGMEM = R"EOF(
-----BEGIN CERTIFICATE-----
MIIDQTCCAimgAwIBAgITBmyfz5m/jAo54vB4ikPmljZbyjANBgkqhkiG9w0BAQsF
.......................................................................................
rqXRfboQnoZsG4q5WTP468SQvvG5
-----END CERTIFICATE-----
)EOF";

static const char client_cert[] PROGMEM = R"EOF(
-----BEGIN CERTIFICATE-----
MIIDWjCCAkKgAwIBAgIVAKAEpMzTvgSj6Cfki29+Bwd2GcxJMA0GCSqGSIb3DQEB
CwUAME0xS..........................
-----END CERTIFICATE-----
)EOF";

static const char client_key[] PROGMEM = R"KEY(
-----BEGIN RSA PRIVATE KEY-----
MIIEpAIBAAKCAQEAuqHZchnoHXQYOvkN0xViIi9ARmp3y9FLkA7thSbu5pQOedzq
fN5zFHUAMC0ye5pXzZ0rtu5+PrpMsxfXcClx6Dcqiwc1Ug1vI5bWzeI1AviRSM+k
.........................................................................
uhdis+by3+AZsbVRujicleG9PaKizWujYvlsOQImqhrX5BcZESadFQ==
-----END RSA PRIVATE KEY-----
)KEY";

    BearSSL::WiFiClientSecure client;
    BearSSLX509List cert(ca_cert);
    client.setTrustAnchors(&cert);
    BearSSLX509List client_crt(client_cert);
    BearSSLPrivateKey key(client_key);
    client.setClientRSACert(&client_crt, &key);
    setClock();
    fetchURL(&client, host, port, path);    
}

void setup(){
  Serial.begin(115200);
  //Serial.setDebugOutput(true);
  Serial.println();
  Serial.println();

  // We start by connecting to a WiFi network
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");

  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  fetchCertAuthority();
  

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages

```
exit status 1
no matching function for call to 'BearSSL::WiFiClientSecure::write(String&)'

```

devyte, 2018-12-29:
`client->write(result.c_str());`

Closing due to user error.

closed 2018-12-29 by devyte

biranchi2018, 2019-08-03:
// Example of the different modes of the X.509 validation options
// in the WiFiClientBearSSL object
//
// Mar 2018 by Earle F. Philhower, III
// Released to the public domain

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <time.h>
#include <ArduinoJson.h>

#ifndef STASSID
#define STASSID "biranchi@unifi"
#define STAPSK  "0340500244"
#endif

const char *ssid = STASSID;
const char *pass = STAPSK;

//const char *   host = "api.github.com";
const uint16_t port = 443; 
//const char *   path = "/";

const char *   host = "17kcb8x98k.execute-api.ap-southeast-1.amazonaws.com";
const char *   path = "/dev/";

// Set time via NTP, as required for x.509 validation
void setClock() {
  configTime(3 * 3600, 0, "pool.ntp.org", "time.nist.gov");

  Serial.print("Waiting for NTP time sync: ");
  time_t now = time(nullptr);
  while (now < 8 * 3600 * 2) {
    delay(500);
    Serial.print(".");
    now = time(nullptr);
  }
  Serial.println("");
  struct tm timeinfo;
  gmtime_r(&now, &timeinfo);
  Serial.print("Current time: ");
  Serial.print(asctime(&timeinfo));
}

// Try and connect using a WiFiClientBearSSL to specified host:port and dump HTTP response
void fetchURL(BearSSL::WiFiClientSecure *client, const char *host, const uint16_t port, const char *path) {
  if (!path) {
    path = "/";
  }

  StaticJsonDocument<300> doc;
  doc["key11"] = "Value11";
  doc["time"] = 1232323;
  
  // serializeJson(doc, Serial);
  
  char buffer[100];
  serializeJson(doc, buffer);
  Serial.println(buffer);

//  String result = "{\"Action\":\"hi\",\"DeviceID\":\"1234567890\"}";
//  int len1 = len(result);
  size_t len = strlen(buffer);

  ESP.resetFreeContStack();
  uint32_t freeStackStart = ESP.getFreeContStack();
  Serial.printf("Trying: %s:443...", host);
  client->connect(host, port);
  if (!client->connected()) {
    Serial.printf("*** Can't connect. ***\n-------\n");
[... 195 lines cut ...]

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");

  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  fetchNoConfig();
  fetchInsecure();
  fetchFingerprint();
  fetchSelfSigned();
  fetchKnownKey();
  fetchCertAuthority();
  fetchFaster();
}

void loop() {
  // Nothing to do here
}

I am able to send POST request, but the post body is null.