esp8266/Arduino #5574: No hostname option in DHCP 2.4.2

cloeffler, 2019-01-02:
#### Platform

- Hardware:					ESP-8266
- Core Version:			latest git hash 
- Development Env:		Platformio
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			[qio|dio|other]
- Flash Size: 		2MB
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   9600

### Problem Description

Upon updating from 1.6 if the espressif platform in PlatformIO. Now using 1.8. The DHCP hostname is no longer int the DHCP discover coming from my 8266. So DHCP will ignore this device and never connect. 
Ive tryed to turn on wifi debugging and use the wifi_station_set_hostname() call, and get the same results. 

here is the packet for the DHCP discover. 

Bootstrap Protocol (Discover)
    Message type: Boot Request (1)
    Hardware type: Ethernet (0x01)
    Hardware address length: 6
    Hops: 0
    Transaction ID: 0x0b00722b
    Seconds elapsed: 0
    Bootp flags: 0x0000 (Unicast)
    Client IP address: 0.0.0.0
    Your (client) IP address: 0.0.0.0
    Next server IP address: 0.0.0.0
    Relay agent IP address: 0.0.0.0
    Client MAC address: BoboClown_80:00:01 (11:22:33:80:00:01)
    Client hardware address padding: 00000000000000000000
    Server host name not given
    Boot file name not given
    Magic cookie: DHCP
    Option: (53) DHCP Message Type (Discover)
    Option: (57) Maximum DHCP Message Size
    Option: (55) Parameter Request List
    Option: (255) End
        Option End: 255
    Padding: 000000000000000000000000000000000000000000000000...

as you can see there is no option 12 (hostname). 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

char ssid[] = "network";
char password[] = "password"
char uid[6] = "123456"; 

void setup() {
    char hostname[11];
    hostname[0] = 'D';
    hostname[1] = 'U';
    hostname[2] = 'D';
    hostname[3] = 'E';
    for (int i = 0; i < 6; i++) {
      hostname[i + 4] = uid[i];
    }
    hostname[10] = 0;
    WiFi.hostname(hostname);

  WiFi.setAutoConnect(false);
  if (!wifi_station_set_auto_connect(false)) {
    Serial.println("disable auto connect failed");
  }

}

void loop() {
  WiFi.begin(ssid, password);
while (WiFi.status() != WL_CONNECTED) {
  Serial.print("waiting..");
}
}

```

### Debug Messages

```
Debug messages go here
```

d-a-v, 2019-01-03:
I have, with `tcpdump -vvni br-lan port 67 or port 68` on the dhcp server:
```
02:41:18.471175 IP (tos 0x0, ttl 255, id 1, offset 0, flags [none], proto UDP (17), length 336)
    0.0.0.0.68 > 255.255.255.255.67: [udp sum ok] BOOTP/DHCP, Request from 60:01:94:1a:8b:cd, length 308, xid 0x37bf4d06, Flags [none] (0x0000)
          Client-Ethernet-Address 60:01:94:1a:8b:cd
          Vendor-rfc1048 Extensions
            Magic Cookie 0x63825363
            DHCP-Message Option 53, length 1: Request
            MSZ Option 57, length 2: 1500
            Requested-IP Option 50, length 4: 10.0.1.30
            Server-ID Option 54, length 4: 10.0.1.254
            Parameter-Request Option 55, length 5: 
              Subnet-Mask, Default-Gateway, BR, Domain-Name-Server
              NTP
            Hostname Option 12, length 10: "DUDE123456"
```

With this fixed sketch
```

#include <ESP8266WiFi.h>
#include <Arduino.h>

char ssid[] = "xxxx";
char password[] = "xxxx";
char uid[] = "123456";

void setup() {
  Serial.begin(115200);
  char hostname[11];
  hostname[0] = 'D';
  hostname[1] = 'U';
  hostname[2] = 'D';
  hostname[3] = 'E';
  for (int i = 0; i < 6; i++) {
    hostname[i + 4] = uid[i];
  }
  hostname[10] = 0;
  WiFi.hostname(hostname);

  WiFi.setAutoConnect(false);
  if (!wifi_station_set_auto_connect(false)) {
    Serial.println("disable auto connect failed");
  }
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.println("waiting..");
    delay(500);
  }

}

void loop() {
}
```

This information does not match the title: `Core Version: latest git hash`

devyte, 2019-01-03:
Closing due to user error.

closed 2019-01-03 by devyte

cloeffler, 2019-01-07:
ok so that is good. So your saying that the hostname is working in the latest git hash?

d-a-v, 2019-01-07:
Yes and in 2.4.2 too with the above sketch.

cloeffler, 2019-01-07:
ah, i just noticed that you saw the hostname in the Request. I am looking
for hostname in the Discover.

d-a-v, 2019-01-08:
I saw nothing in RFCs after a quick search.
From lwIP point of view, this is on purpose.
I invite you to check their code, and ask them why it is so.

http://git.savannah.nongnu.org/cgit/lwip.git/tree/src/core/ipv4/dhcp.c

In this file, search for `dhcp_option_hostname`.
You will notice it is not included in the "dhcp-discover" packet built and sent from the function `dhcp_discover()`.

I have not checked in lwIP-1.4 which is used by previous versions of this core.
http://git.savannah.nongnu.org/cgit/lwip.git/tree/src/core/dhcp.c?h=STABLE-1_4_0&id=3a267586f4d591fae53e08e7c3e01ce40e77dd15
(or check in `tools/sdk/lwip/src/core/dhcp.c`)

d-a-v, 2019-01-28:
Please elaborate

dachshund-digital, 2019-01-29:
My experience is, now, that the DHCP option for specification of host name is ignored by the WiFi client.  Such that when I qualify a host name as part of the DHCP reservation for my ModeMCU device, it is not honored as it would be by almost any other device that supported DHCP assignment of host name from the DHCP server in question.  This issue should be and I suggest must be fixed as soon as possible.

d-a-v, 2019-01-29:
To clarify your comment:

It is known that (correct me if I'm wrong)
- ESP as DHCP client makes a request to the DHCP server with a proposed hostname
- The DHCP server provides an IP address but is allowed to give a different hostname than the one asked by the ESP

I think that
- On the network, DNS name given by the DHCP server will resolve to the ESP IP address given by DHCP, if DHCP server and local DNS server talk to each other (like it is with `dnsmasq`)

Now you say:
- DHCP option for specification of host name **is ignored** by the WiFi client
- ESP should read this hostname

Now my question:
- How does this name should be taken into account in the ESP, is it for display like `Serial.println(WiFi.hostname())`, or what other purpose ?
- What is exactly your use case ?
- What can't you achieve ?

devyte, 2019-01-29:
This discussion has evolved off topic. Please open a new issue, fill out the required info, and provide a MCVE and use case, with details to reproduce.

dachshund-digital, 2019-01-30:
I opened a new issue, but to answer your questions above...

    How does this name should be taken into account in the ESP, is it for display like Serial.println(WiFi.hostname()), or what other purpose ?  

Correct, and for proper network identification, it is a security issue for a device not to have a known and qualified name.  If DHCP provides a host-name, and the host name is NOT statically assigned by the local OS, it should become the actual name for the device on the network, this how Windows, and Linux work.

What is exactly your use case ? 

My use case is that I want the device, i.e. ESP8266 (ModeMCU) to never have a host-name statically assigned in code.  It makes no sense to do this.  Then I would have to have a sketch for each and every device, because of a single static assignment?  IMHO, no, this is not how DHCP and DNS are supposed to work.  Dynamic DNS is not applicable, the device should not push the name, when the DHCP host-name option is published to the device.  Please review DHCP specification for additional information if needed.

What can't you achieve ?

What I want to achieve is consistent behavior across DHCP supporting platforms, Windows and Linux support DHCP host name option, as do every other micro controller I have used until now.  Only the ESP8266 WiFi environment fails to honor the DHCP configuration as published, per my experience thus far.   If ESP8266 device can get its IP from DHCP for example, its DNS server, etc. then it should get the device name from DHCP.  There are many DHCP options that allow for settings that are applicable beyond IP, Gateway, IMHO DNS, NTP, and host name resources should be honored as well.