esp8266/Arduino #5576: Crash in loop_task(ETSEvent*)?

BrandonFanti, 2019-01-03:
### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [ X] If there is a stack dump, I have decoded it.
- [X ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12E]
- Core Version:			[2.5.0-beta2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[512K(no SPIFFS)]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[Serial]
- Upload Speed:   [115200]

### Problem Description
I have configured the ESP8266-12E to act as an access point, and host a simple web server. All I'm doing at this point is some simple logging of IP address and MAC addresses once the client has completed their dhcp request. The logging through serial completes without error, but after several iterations through the main loop it dumps the stack and resets.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```
#include<WiFiClient.h>
#include<WiFiServer.h>
#include<ESP8266WebServer.h>
#include<WiFiManager.h>
#include<avr/pgmspace.h>

const char *SERVER_AP_SSID = "ESPsoftAP_8266";

int AP_CLIENT_COUNTER = 0;
int AP_CLIENT_COUNTER_PREV = 0;
struct station_info *STAT_INFO;

PGM_P rootPage;
IPAddress local_IP(192,168,1,1);
IPAddress gateway(192,168,1,1);
IPAddress subnet(255,255,255,0);

ESP8266WebServer server(80);

//ESP8266 configurations for access point
void stationInit();

void webInit();
void handleWebRoot();
void setupPages();

void setup() {
  delay(10);
  stationInit();
  webInit();
}

void loop() {
  AP_CLIENT_COUNTER = wifi_softap_get_station_num();
  STAT_INFO = wifi_softap_get_station_info();
  
  if(AP_CLIENT_COUNTER != AP_CLIENT_COUNTER_PREV && STAT_INFO != NULL){
    Serial.print("New number of AP clients: ");
    Serial.println(AP_CLIENT_COUNTER);
    
    for(int i = 1; i <= AP_CLIENT_COUNTER && STAT_INFO != NULL; i++){
        printApClientInfo(i, STAT_INFO);
        STAT_INFO = STAILQ_NEXT(STAT_INFO, next);
    }
    AP_CLIENT_COUNTER_PREV = AP_CLIENT_COUNTER;
  }
  
  server.handleClient();
}

void stationInit(){
  //Use ESP's IP as itself, dns services, and gateway.
  WiFi.config(local_IP, local_IP, local_IP, subnet);
  WiFi.mode(WIFI_AP_STA);
  WiFi.softAPConfig(local_IP, local_IP, subnet);
  
  setupSerial();

  Serial.print("Starting access point ");
  Serial.print(SERVER_AP_SSID);
  Serial.print("...");
  Serial.println(WiFi.softAP(SERVER_AP_SSID) ? " Ready." : " Failed!");
  printHostNameAndIP();
  WiFi.begin();
}

void webInit(){
  setupPages();
  server.begin();
  server.on("/", HTTP_GET, handleWebRoot);
}

void setupPages(){
  rootPage = PSTR("Hello");
}

void printApClientInfo(int clientNumber, station_info* STAT_INFO){
  Serial.printf("Client number %d info:\n", clientNumber);
  ip4_addr *stationAddress;
  u32_t address;
  stationAddress = &STAT_INFO->ip;
  address = stationAddress->addr;
  Serial.print("  Client IP address is ");
  Serial.printf("%d.%d.%d.%d", ((unsigned char*)&address)[0], ((unsigned char*)&address)[1], ((unsigned char*)&address)[2], ((unsigned char*)&address)[3]);
  Serial.print(" and MAC address is ");
      
  Serial.print(STAT_INFO->bssid[0],HEX);Serial.print(":");
  Serial.print(STAT_INFO->bssid[1],HEX);Serial.print(":");
  Serial.print(STAT_INFO->bssid[2],HEX);Serial.print(":");
  Serial.print(STAT_INFO->bssid[3],HEX);Serial.print(":");
  Serial.print(STAT_INFO->bssid[4],HEX);Serial.print(":");
  Serial.print(STAT_INFO->bssid[5],HEX);Serial.println("");
}

void handleWebRoot(){
  Serial.print("IP connected: ");
  Serial.println(server.client().remoteIP());
  Serial.print("Sent page of length ");
  Serial.print(sizeof(rootPage));
  Serial.println(".");
  server.send_P(200, PSTR("text/plain"), rootPage, sizeof(rootPage));
}

void setupSerial(){
  Serial.begin(115200);
  Serial.println();
  Serial.println();
}

void printHostNameAndIP(){
  Serial.print("Hostname: ");
  Serial.println(WiFi.hostname());
  Serial.print("IP: ");
  Serial.println(WiFi.localIP());
}

```

### Debug Messages

```
Starting access point ESPsoftAP_8266... Ready.
Hostname: ESP_14460E
IP: 192.168.1.1
New number of AP clients: 1
Client number 1 info:
  Client IP address is 192.168.1.100 and MAC address is AC:37:43:A5:F9:94

Exception (29):
epc1=0x4021e364 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3ffffd60 end: 3fffffb0 offset: 01a0
3fffff00:  4021e8ff 3ffec8c0 3ffec8c0 4021e8aa  
3fffff10:  00000018 00000001 00000001 3ffe9d5a  
3fffff20:  00000000 0000016a 3ffe9d62 3fff0934  
3fffff30:  3ffe9d44 00000000 0000002f 4022e356  
3fffff40:  3ffec8c0 3ffe9d44 3fffdcc0 3ffe8eb0  
3fffff50:  3ffe9d54 3fff1914 00000000 3ffe8eb0  
3fffff60:  00000000 3ffec8c0 00000000 3ffe8510  
3fffff70:  4022dc33 3fffdab0 00000000 40208327  
3fffff80:  3ffe8eb0 40000f49 3fffdab0 40000f49  
3fffff90:  40000e19 40001878 00000002 00000000  
3fffffa0:  3fffff10 aa55aa55 000000ec 40104730  
<<<stack<<<

last failed alloc call: 40100210(20)

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v0fd86a07
~ld

Starting access point ESPsoftAP_8266... Ready.

```

and the decoded stack...
```
Decoding stack results
0x40208327: loop_task(ETSEvent*) at C:\Users\BrANDON\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266\core_esp8266_main.cpp line 133
```

I've excluded irrelevant code so the line number of the decoded stack is different from what is in the above code. Line 133 in my file points to the handleWebRoot() function where the client IP is printed ```Serial.println(server.client().remoteIP());```

But at no point is there an "IP connected: " sent over serial, so I have no reason to think that the function is even being called.

d-a-v, 2019-01-03:
This sketch indeed does not work.
This call `STAT_INFO = wifi_softap_get_station_info();` is honored but makes wdt trigger later (even with "extra4k" disabled).
Running in debug mode, with lwIP-1.4 shows OOM (shows nothing relevant with lwIP-2).
Origin is unknown yet.
Hint: This call gets leases from the dhcp server (included with lwIP, 1.4 or 2).

devyte, 2019-01-03:
I think this is a massive mem leak. You're repeatedly calling wifi_softap_get_station_info() without ever calling wifi_softap_free_station_info(). That agrees with @d-a-v 's OOM finding. To make sure, monitor available heap on each pass through loop().

Whatever else, this isn't the right place to request help for your particular project. This is an issue tracker, meant for tracking issues in the core. For general support, please request help at a community forum like esp8266.com or stackoverflow, and read Espressif's SDK api reference (document 2c).

Closing.

closed 2019-01-03 by devyte

d-a-v, 2019-01-03:
I tried it, it works with `wifi_softap_free_station_info();`.
Thanks @devyte.

BrandonFanti, 2019-01-03:
> I think this is a massive mem leak. You're repeatedly calling wifi_softap_get_station_info() without ever calling wifi_softap_free_station_info(). That agrees with @d-a-v 's OOM finding. To make [...]

That was it. 

Thanks for the help. I had a feeling it was a memory leak with the core which was why I opened the issue, thinking it was internal, but nope it was my own incorrect usage of the api. Thanks again.