esp8266/Arduino #5580: CONVERT 18B20 CODE TO OPERATE WITH DHT11 .

methaqali, 2019-01-03:
I have the following code for 18B20 sensor for convert from analog temp. to analog voltage output but I have DHT11 sensor can any one help me who to get or convert this code to use with DHT11.
#include <OneWire.h>
#include <DallasTemperature.h>

#define ONE_WIRE_BUS 2

OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);

int MIN_TEMP = 0;
int MAX_TEMP = 50;

void setup() {

pinMode(9,OUTPUT);
Serial.begin(9600);
sensors.begin();

}

void loop() {

  float temperature = getTemperature();
  int voltage = temperatureToPWM(temperature);
  analogWrite(9,voltage);
  delay(500);
}

float getTemperature()
{
  float temperature =0;
  sensors.requestTemperatures(); // Send the command to get temperatures
  temperature = sensors.getTempCByIndex(0);
  Serial.println(temperature);
  return temperature;
}

int temperatureToPWM(float temperature)
{
  float temp=0;
  float voltage = 0;
  temp = temperature*10;
  voltage = map(temp,MIN_TEMP*10,MAX_TEMP*10,0,250);
  return voltage;
}

devyte, 2019-01-04:
When you opened this issue you were presented with an issue template, which you completely ignored. The first line of that template says:
> If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

This is a "How do I" type question. There are plenty of examples out there, I'm sure if you investigate you'll find some that suit you.

Closing. See issue POLICY and ISSUE_TEMPLATE docs for more info.

closed 2019-01-04 by devyte

methaqali, 2019-01-04:
Thank you for notification.
Methaq A.Ali       
Electrical Chief Engineer
Testing & Commissioning Engineer
Mobile: +964 7705676403 , +964 7810057451
Email Alternative:methaq.ali2006@yahoo.com

 
 
Closed #5580.