esp8266/Arduino #5582: WebServer lagging

ladafajrak, 2019-01-04:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.4.2 (2.5.0 beta 2 also tested)]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [other] (serial upload only)

### Problem Description

Hello,
I'm trying to transfer data from ESP8266 using JSONfile, but I deal with huge lagging. Normaly, the response time is about 250 ms. But sometimes it rises to tens od seconds. I have also tried to modify example code from library to response with empty .json file for every request (without any additional processes), but the issue is still on.

On the screenshot you can see laging during using javascript for refreshing the overview.json every 1000 ms.

[image]

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>

/* Set these to your desired credentials. */
const char *ssid = "ssid";
const char *password = "";

ESP8266WebServer server(80);

/* Just a little test message.  Go to http://192.168.4.1 in a web browser
   connected to this access point to see it.
*/
void handleRoot() {
  server.send(200, "application/json", "{}");
}

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();
  Serial.print("Configuring access point...");
  /* You can remove the password parameter if you want the AP to be open. */
  WiFi.softAP(ssid, password);

  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  server.on("/overview.json", handleRoot);
  server.begin();
  Serial.println("HTTP server started");
}

void loop() {
  server.handleClient();
}

### Debug Messages

adrionics, 2019-01-04:
Is the WiFi client trying unsuccessfully to connect to an SSID? Turn on WiFi debugging and see what you get?

https://github.com/esp8266/Arduino/issues/1615

ladafajrak, 2019-01-07:
Hello,
I'm using ESP only in softAP mode, not in STA+SoftAP. I've tried (based on issue 1615) to add esp.eraseconfig(); command, but no change has occured. Here I attach WiFi debug report. Can you see something interesting there? Evts 7 and 9 after client connection are completely independent on lagging time (events are comming without lagging and lagging is coming without events).

Thank you

SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc

bcn 0
del if1
usl
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
AP IP address: 192.168.4.1
HTTP server started
wifi evt: 7
wifi evt: 7
wifi evt: 7
add 1
aid 1
station: ac:fd:ce:30:8c:a2 join, AID = 1
wifi evt: 5
wifi evt: 9
wifi evt: 9
wifi evt: 9
wifi evt: 7

d-a-v, 2019-01-19:
> Normaly, the response time is about 250 ms. But sometimes it rises to tens od seconds.

This may have been addressed by TCP-SACK-OUT from lwIP-2.1.2.
Please retry with an updated core, v2.5.0-beta2 or git head.

ladafajrak, 2019-01-22:
It doesn't helped. Problem is the same, plus in version 2.5.0 beta2 doesn't work SD file upload function server.streamFile(); ESP will reboot after calling it.

d-a-v, 2019-01-22:
Can you provide another MCVE with softAP, `server.streamFile()` involving SD ?
To solve, we need to reproduce.

ladafajrak, 2019-01-22:
`#define SD_CHIP_SELECT D8

#include <SPI.h>
#include <SD.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

ESP8266WebServer server(80); 

void setup() {
  WiFi.mode(WIFI_AP);  
  Serial.begin(19200);
  Serial.println();
  if (!SD.begin(SD_CHIP_SELECT)) Serial.println("Card failed, or not present");
  WiFi.mode(WIFI_AP);
  WiFi.softAP("ssid", "");
  server.onNotFound(handleNotFound);
  server.begin();                             // start the HTTP server
}

void loop() {
server.handleClient(); 
}

bool loadFromSdCard(String path) {
    Serial.print("loading");
    Serial.println(path);
    if (!SD.exists(path.c_str())) {
      return false;
    }
    File dataFile = SD.open(path.c_str());
    int fsizeDisk = dataFile.size();
    Serial.print("fsizeDisk: ");
    Serial.print(fsizeDisk/1024);
    Serial.println(" kb");

    size_t fsizeSent = server.streamFile(dataFile,"text/csv");

    Serial.print("fsizeSent: ");
    Serial.print(fsizeSent/1024);
    Serial.println(" kb");

    dataFile.close();
    return true;
}

void handleNotFound() {

  if (loadFromSdCard(server.uri())) {
    return;
  }
  Serial.print("not found ");
  Serial.println(server.uri());
  server.send(404, "text/html", "<html><body>Not found! Try <a href=\"/\">index.html</a></body></html>");

}`

This code fails with this error message:

loading/201901.csv
fsizeDisk: 1044 kb
Fatal exception 9(LoadStoreAlignmentCause):

Decoded stack:

`Decoding stack results
0x40202ea5: SPIClass::transferBytes(unsigned char const*, unsigned char*, unsigned int) at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\SPI\SPI.cpp line 524
0x40203473: spiRec() at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\SD\src\utility\Sd2Card.cpp line 52
0x40203898: Sd2Card::readData(unsigned int, unsigned short, unsigned short, unsigned char*) at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\SD\src\utility\Sd2Card.cpp line 478
0x40203c59: SdFile::read(void*, unsigned short) at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\SD\src\utility\SdFile.cpp line 708
[... 54 lines cut ...]
 line 395
0x40203030: File::read(void*, unsigned short) at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\SD\src\File.cpp line 96
0x40203044: File::readBytes(char*, unsigned int) at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\SD\src\File.cpp line 101
0x40100114: millis at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266\core_esp8266_wiring.c line 183
0x40205330: BufferedStreamDataSource ::get_buffer(unsigned int) at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WiFi\src/include/DataSource.h line 93
0x40205545: ClientContext::_write_some() at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WiFi\src/include/ClientContext.h line 485
0x4020a058: esp_yield() at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266\core_esp8266_main.cpp line 91
0x40205600: ClientContext::_write_from_source(DataSource*) at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WiFi\src/include/ClientContext.h line 445
0x40205b18: WiFiClient::write(Stream&) at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WiFi\src/include/ClientContext.h line 369
0x402024ea: loadFromSdCard(String) at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WebServer\src/ESP8266WebServer.h line 135
0x402095b7: String::reserve(unsigned int) at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266\WString.cpp line 146
0x40209600: String::copy(char const*, unsigned int) at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266\WString.cpp line 182
0x40209718: String::String(String const&) at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266\WString.cpp line 41
0x4020258e: handleNotFound() at C:\Users\Hejsek\Documents\Arduino\sketch_jan22a/sketch_jan22a.ino line 58
0x4020a64a: std::_Function_handler ::_M_invoke(std::_Any_data const&) at c:\users\hejsek\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x402068f6: std::function ::operator()() const at c:\users\hejsek\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 2465
0x40206a01: ESP8266WebServer::_handleRequest() at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp line 604
0x40100114: millis at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266\core_esp8266_wiring.c line 183
0x4020a608: WiFiServer::available(unsigned char*) at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WiFi\src\WiFiServer.cpp line 116
[... cut ...]
0x40205c52: ESP8266WebServer::begin() at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp line 100
0x402026e0: setup() at C:\Users\Hejsek\Documents\Arduino\sketch_jan22a/sketch_jan22a.ino line 23
0x40203030: File::read(void*, unsigned short) at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\SD\src\File.cpp line 96
0x4020a104: loop_wrapper() at C:\Users\Hejsek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266\core_esp8266_main.cpp line 125`

earlephilhower, 2019-01-22:
I found a bug in the sdfat lib very much like this crash, where SPI transfers were not aligned to 32-bits (it was actually in the comments that it didn't need to be aligned...which is wrong).

See https://github.com/earlephilhower/ESP8266SdFat/commit/ff64b2daebf4cd9cc54908635d1c6f63dc4a82d1

I think only the raw upstream lib has this issue, not the old one included w/the Arduino 8266 core.

d-a-v, 2019-02-01:
Linked to #4967

d-a-v, 2020-11-09:
Closing due to age and lack of feedback since the SPI fix.
Feel free to report back here or open a new request if the issue is still present.

closed 2020-11-09 by d-a-v