esp8266/Arduino #5583: Folder of SoftwareSerial are empty in Windows 10

lrmoreno007, 2019-01-04:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:					[NodeMCU v.1]
- Core Version:			[latest git]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows 10 1809]

### Settings in IDE

- Module:  			[Nodemcu]
- lwip Variant:   [v2 Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA]

### Problem Description

Folder esp8266com\esp8266\libraries\SoftwareSerial are empty in Windows 10.

I receive the following message in Arduino IDE:
Librería inválida encontrada en C:\Users\XXXXXX\Documents\Arduino\hardware\esp8266com\esp8266\libraries\SoftwareSerial: no headers files (.h) found in C:\Users\XXXXXX\Documents\Arduino\hardware\esp8266com\esp8266\libraries\SoftwareSerial

I inspect the github repository and found that the folder is called: "SoftwareSerial @ 23ae000" and isn't empty.

Pablo2048, 2019-01-04:
It's because the SoftwareSerial is git submodule see https://github.com/esp8266/Arduino/issues/3880 and https://github.com/esp8266/Arduino/issues/3596

devyte, 2019-01-04:
Note: need to update git install doc

lrmoreno007, 2019-01-04:
> 
> 
@Pablo2048  Thanks!! it's solved with: `git submodule update --init --recursive`

I update:
https://github.com/lrmoreno007/ESP-Easy-Updater-for-Windows

@devyte yes, update doc is needed.

commit 324eb56d (2019-01-18): Add fix for SoftwareSerial submodule errors to documentation (#5602)

* Add fix for SoftwareSerial submodule errors to documentation

* Update installing.rst

Replaces and based on #5602
Fixes #5583

* Update installing.rst

Remove forgotten section
Add update section for Other OS
closed 2019-01-18 by devyte