esp8266/Arduino #5588: Network Ports disappearing

sundeepgoel72, 2019-01-05:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

devyte, 2019-01-05:
@sundeepgoel72 you ignored the issue template. Which core version? What is yourr platform? What is  the MCVE sketch to reproduce? Etc.
Please edit your post and add the required fields.

CC @LaborEtArs .

sundeepgoel72, 2019-01-05:
> @sundeepgoel72 you ignored the issue template. Which core version? What is yourr platform? What is the MCVE sketch to reproduce? Etc.
> Please edit your post and add the required fields.
> 
> CC @LaborEtArs .

apologies for missing, its my first post here. Have updated the issue based on template

devyte, 2019-01-05:
The request is for an MCVE sketch, where M means *M*inimal. Could you please reduce? e.g.: remove thingspeak, sensors, etc and provide as small a sketch as possible that shows your probe issue.

What happens if you remove that delay(50) in the loop for all devices?

Side note: your timing calc millis() > nextTx won't work as expected on rollover. I suggest using the periodic polledTimeout instead. There are examples and device tests you can check out for usage.

sundeepgoel72, 2019-01-05:
> The request is for an MCVE sketch, where M means _M_inimal. Could you please reduce? e.g.: remove thingspeak, sensors, etc and provide as small a sketch as possible that shows your probe issue. [...]

Hi, updated issue with Minimal sketch, Same issue persists.

devyte, 2019-01-05:
@LaborEtArs the ArduinoOTA.handle() call internally calls MDNS.update().

devyte, 2019-01-05:
From the MCVE:
```cpp
ArduinoOTA.setHostname("myesp8266Roaming2");
```

Are all ESPs setting the same hostname?

sundeepgoel72, 2019-01-06:
> From the MCVE:
> 
> ```c++
> ArduinoOTA.setHostname("myesp8266Roaming2");
> ```
> 
> Are all ESPs setting the same hostname?

No, all have different hostnames.

d-a-v, 2019-01-12:
> This is very annoying, considering the time devoted to compilation.

While waiting for new mDNS to be debugged, you can start arduino OTA by hand:
```
# This script will push an OTA update to the ESP
# use it like: python espota.py -i <ESP_IP_address> -I <Host_IP_address> -p <ESP_port> -P <Host_port> [-a password] -f <sketch.bin>
```

Samlof, 2019-01-13:
Yeah same problem. I can take the esp8266-id.local address from Bonjour and use that with espota.py and I can upload thru net.
Doesn't show up in Ports. Using Arduino ide 1.8.8 with 2.5.0-beta2, with nodemcu v2. Using the BasicOTA example sketch.
I tried 2.4.0 core but that didn't work either. No idea is the problem here or with Arduino Ide

LaborEtArs, 2019-01-14:
If the problem exists with v2.4.x core as well as with the v2.5.x core, the reason can‘t be the new mDNS responder, as it was introduced with 2.5 only. At least it can’t be the only reason...
Has someone tried the old responder in the v2.5 core?
However the new mDNS responder had had a bug, that caused the ESP to get lost in ZeroConf in some configurations after the first host domain timeout (TTL, 120s) was reached. The PR for a fix is running right now...

pfeerick, 2019-01-19:
I may (or may not!) be be suffering the same issue. I'm using both the Arduino IDE (at times, when I'm lazy) and PlatformIO. The Arduino IDE is using 2.5.0 beta2, and PlatformIO is still using core 2_4_2. I've noticed of late only one or occasionally two/three mDNS discovery devices are showing up on the Arduino IDE, although that hasn't really bothered me, as it was just one more reason to be using PlatformIO as it is seeing them all fine! :-/ 

I found out at one point that disabling and re-enabling my ethernet connection seemed to wake the Arduino IDE up (sometimes, but doesn't seem to of late). This is when both PlatformIO and zeroconf browser (as I found bonjourBrowser a bit buggy at times) could see all the devices just fine. 

A side note - I noticed that the three devices not visible to the Arduino IDE have 'board' set to PLATFORMIO_OAK rather than ESP8266_OAK... related? Relevant?

[image]

[image]

[image]

LaborEtArs, 2019-01-20:
As the issue appeared when using core v2.4.x as well as when using core 2.5.0, it might not have been (completely) solved by the update of core 2.5.0 ...
I addition, there are a few minor issues left, for which the PR is running.
But: On my system, all the ESPs stay visible in the IDE (using Eclipse) and the Bonjour browser now :-)

sundeepgoel72, 2019-01-21:
> As the issue appeared when using core v2.4.x as well as when using core 2.5.0, it might not have been (completely) solved by the update of core 2.5.0 ...

Newbie question, how do you upgrade to core 2.5.0 from the IDE library manager?

pfeerick, 2019-01-22:
I take it you found the update option in the board manager then (as I was about to reply to your 'how to upgrade question' which you've since edited ;) ). The latest 'released' version is 2.5.0-beta2, but changes have been released since then, you need to consider installing the bleeding edge git version ([install instructions here](https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version)).

devyte, 2019-01-23:
2.5.0-beta1 had a complete MDNS responder rewrite. There have been recent fixes to mdns merged after beta2 that could be relevant. Please retest with latest git, installation instructions are on readthedocs. Or you can wait until beta3.

pfeerick, 2019-01-24:
I'm still not convinced recent changes to MDNS are the cuplrit here... after all, if other browsers such as PlatformIO and zeroconfbrowser can see the devices, then isn't MDNS working fine?

There are at least two relevant issues open on the Arduino github issue tracker - [6695](https://github.com/arduino/Arduino/issues/6695) - which suggests that if there are any issues on their end - it is coming from updating the `jmdns` library. There is also the recently opened [8408](https://github.com/arduino/Arduino/issues/8408) which builds on the first issue, and there is mention of use what appears to be a new mdns discovery system in the Arduino IDE betas... which I'm about to have a look at. 

But more relevantly to this issue... I'll just leave some screenshots to so some inconsistent behaviour based on IDE version and opening of the ports menu... I have one device that is running ArduinoOTA  - first built against 2.5.0beta2 whilst on windows, and then against current git version of the ESP8266 core when on linux.

Windows 10, Arduino IDE 1.8.8, ESP8266 v2.5.0 beta2, first open of ports menu
[image]

Windows 10, Arduino IDE 1.8.8, ESP8266 v2.5.0 beta2, second of ports menu (haven't closed the IDE, or changed ANYTHING else)
[image]

Ok.... so lets try on v1.8.1 as there was mention that v1.8.1 was pre some updates to the jmdns and discovery was working better. Windows 10, Arduino IDE 1.8.1, ESP8266 v2.5.0 beta2, first open of ports menu (no change on subsequent opens of menu). 

[image]

Whilst PlatformIO the whole time was showing... guess what... all four (actually five, but ignore Oak2 - it just work up from deepsleep in time to make an appearance for the screenshot). 
[image]

So what about against the latest git version, and throw linux into the mix to see if it's OS specific...

Ubuntu 18.10, Arduino IDE 1.8.8, ESP8266 git version, first open of ports menu
[image]

Ubuntu 18.10, Arduino IDE 1.8.8, ESP8266 git version, second open of ports menu
[image]

That's it... I give up on mDNS working properly with the Arduino IDE unless they've fixed it in the beta version!

pfeerick, 2019-01-24:
Woohoo! 1.9.0 beta of the Arduino IDE seems to have some... issues with duplication of entries, but at least mDNS discovery seems rock solid so far in my brief 5 minutes of attempting to break it... all devices are being discovered, consistently, and upload was working just fine. This is again against 2.5.0 beta2 as that's what my Windows 10 environment is currently on. 

[image]

pieman64, 2019-01-24:
@pfeerick I have Win and Ubuntu systems and my main Windows machine has always struggled with port discovery. After a reboot it works for a while but no for long. I switched to http OTA updates rather than local OTA updates.

Will check out 1.9.0 beta though before I discard local OTA altogether.

pfeerick, 2019-01-24:
@pieman64 Yeah, I've found over the years that linux is much more stable with networking stuff. on the OTA/mDNS front, the mDNS discovery that PlatformIO seems to be stable and working on Windows, hence why I point the finger at the Arduino IDE. Especially since it's mDNS discovery is just as bad on linux, whilst platformio is working just fine there also.

reaper7, 2019-01-25:
Latest changes looks very promising,
for a long time of work with arduino 1.8.8 all my esp82xx OTA devs are visible :)

reaper7, 2019-02-21:
looks like "disappearing network ports" problem back, when sdk was revert from pre-3.0 to 2.2.1 :(

I recompile and upload the same espurna project to 2 from my 5 devices
and I do not see these two devices in IDE and zeroConfServiceBrowser tool.
The other 3 devices compiled on pre-3.0 are still visible.

d-a-v, 2019-02-21:
sdk-pre-3 is still available in tool menu, available when using Generic esp8266 board.

VinceW31, 2019-04-26:
Ive just had the same problem as the OP. Im pretty sure its #include <ESP8266WiFiMulti.h> thats the problem.
In my case Id just upgraded my sketch to include WiFiMulti functionality (to choose between 2 sets of WiFi credentials, whichever was available) and uploaded it via OTA when all the problems with network ports in the Arduino IDE started and intermittent connections to all my ESP devices.
Rebooting the router got them back temporarily and I managed to revert all my devices back to the old sketch without WiFiMulti functionality and all is back to normal now.  The original post above doesn't need WiFiMulti functionality anyway because its only got one set of WiFi credentials to use.

reaper7, 2019-07-04:
For a long time OTA worked on my devices(espurna project) properly if I compiled on sdk pre-3.
All the time all devices were available in ide,
but in the last week, all of them have disappeared after cyclic update
and they only appeared for some time after their reset.

I decided to go back in this repository before commit https://github.com/esp8266/Arduino/commit/f9009b8a5e141235d3380cc6c6c06223008a1099
and re-compiled projects for two of my six devices. 
These two devices do not disappear.

I think that the disappearing port for OTA can cause something in mDNS

d-a-v, 2019-07-04:
@reaper7 
Can you try and replace [`a.ifUP()`](https://github.com/esp8266/Arduino/blob/f9009b8a5e141235d3380cc6c6c06223008a1099/libraries/ESP8266mDNS/src/LEAmDNS_Control.cpp#L128) by `true` and check if it solves your issue ?

reaper7, 2019-07-04:
I will check, of course...
but we will have to wait for the observation results :)

reaper7, 2019-07-06:
so, this is not this line, something else must cause this problem,
but I'm convinced that it's mDNS

I introduced your suggestion on Thursday (yesterday I did not have the opportunity to check the effects)
but this morning none of the devices were available in the ide...
...so I went back to commit https://github.com/esp8266/Arduino/commit/961b558a91caf9589cdf0daa76a2aafd4a6ec568, I've compiled images for all devices again,
and until now everything is visible. 
I will send the next report tomorrow.

reaper7, 2019-07-08:
all devices are still visible for OTA if the project is compiled **before** making mDNS changes (https://github.com/esp8266/Arduino/commit/f9009b8a5e141235d3380cc6c6c06223008a1099)

hreintke, 2019-07-08:
@reaper7 
I am trying to reproduce this issue in my environment but did not succeed. yet.
Can you describe your steps to reproduce ?

d-a-v, 2019-07-08:
Please check this [comment].

@hreintke it seems it doesn't work on android. We have at least two PRs about that not merged because they do not follow standards. It appears that android has an issue with `192.168.`.

reaper7, 2019-07-09:
@d-a-v - it looks like the udp fix has solved the problem

devyte, 2020-02-02:
Several comments say this is resolved, and a whole lot of critical fixes have been merged since this was opened. Closing.
If the problem is still valid, please open a new issue and fill out the required info, including a MCVE.

closed 2020-02-02 by devyte