esp8266/Arduino #5592: ESP8266 Older Code Not Being Replaced By New Code

iseegr8tfuldeadppl, 2019-01-05:
### Basic Infos

- [ DONE] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ DONE] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ NOT SURE] I have tested that the issue is present in current master branch (aka latest git).
- [ DONE] I have searched the issue tracker for a similar issue.
- [NOT SURE ] If there is a stack dump, I have decoded it.
- [ DONE] I have filled out all fields below.

#### Platform

- Hardware:				NodeMCU ESP8266
- Core Version:			1/6/2019
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 NodeMCU Module
- Flash Mode:			using arduino ide i guess
- Flash Size: 		[4MB/1MB] i'm not sure how it is flashing
- lwip Variant:   same response as above
- Reset Method:		also not sure how to, using arduino ide
- Flash Frequency:	4M
- CPU Frequency:		80Mhz
- Upload Using:		not sure
- Upload Speed:   115200

### Problem Description

  So i had flashed my esp8266 with older code from a .bin file but now using Arduino IDE and trying to upload the following code to launch a simple wifi isn't launching any wifi but showing the older wifi that was being used in the previous flash as if it's not even being flashed though i had tried LEDblink on it successfully.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```
#include <ESP8266WiFi.h>

void setup()
{
  Serial.begin(115200);
  Serial.println();

  WiFi.begin("ok", "123456789");

  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println();

  Serial.print("Connected, IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {}
```

### Debug Messages

`Uploading 229984 bytes from to flash at 0x00000000`

tablatronix, 2019-01-06:
Erase flash and try again

closed 2019-01-06 by iseegr8tfuldeadppl

tablatronix, 2019-01-06:
No it can become corrupted especially when updating lib versions