esp8266/Arduino #5593: WiFiMulti and open networks

adrionics, 2019-01-05:
WiFiMulti does not have the ability to connect to open networks due to using an "empty" string for the password instead of NULL.

I fixed it by adding one line in ESP8266WiFiMulti.cpp before the WiFi.begin:

`
//Set the password to NULL if it is empty
		if (strlen(bestNetwork.passphrase) == 0) bestNetwork.passphrase = NULL;
                WiFi.begin(bestNetwork.ssid, bestNetwork.passphrase, bestChannel, bestBSSID);
`
There may be better ways to do this, but this works.

devyte, 2019-01-06:
Which core version? There was a bug with empty strings fixed recently.

adrionics, 2019-01-06:
Sorry, I should have included that. 2.5.0-beta2.

devyte, 2019-01-06:
Yup, fix was after beta2. Please try latest git.

closed 2019-01-06 by adrionics