esp8266/Arduino #5596: Error compiling for board NodeMCU 1.0(ESP-12E Module)

marktech77, 2019-01-07:
----------------------------- Delete below -----------------------------

How do I when i cant compile code in esp8266 v2

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:		                esp8266
- Core Version:			version2
- Development Env:		Arduino IDE
- Operating System:		Windows 7

### Settings in IDE

- Module:  			Nodemcu 1.0(ESP-12E Module)
- Flash Mode:			[qio|dio|other]
- Flash Size: 		 4MB
- lwip Variant:          v2 prebuilt (mss=536) 
- Reset Method:		[nodemcu]
- Flash Frequency:	     40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		OTA
- Upload Speed:   115200

### Problem Description

D:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Program Files\Arduino\hardware -hardware C:\Users\darussyifa\AppData\Local\Arduino15\packages -tools D:\Program Files\Arduino\tools-builder -tools D:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\darussyifa\AppData\Local\Arduino15\packages -built-in-libraries D:\Program Files\Arduino\libraries -libraries C:\Users\darussyifa\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,UploadSpeed=115200 -ide-version=10805 -build-path C:\Users\DARUSS~1\AppData\Local\Temp\arduino_build_272396 -warnings=none -build-cache C:\Users\DARUSS~1\AppData\Local\Temp\arduino_cache_729322 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.esptool.path=C:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.12 -verbose C:\Users\darussyifa\Downloads\FBQ2P48IU5NBGGQ\FBQ2P48IU5NBGGQ.ino
D:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware D:\Program Files\Arduino\hardware -hardware C:\Users\darussyifa\AppData\Local\Arduino15\packages -tools D:\Program Files\Arduino\tools-builder -tools D:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\darussyifa\AppData\Local\Arduino15\packages -built-in-libraries D:\Program Files\Arduino\libraries -libraries C:\Users\darussyifa\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,UploadSpeed=115200 -ide-version=10805 -build-path C:\Users\DARUSS~1\AppData\Local\Temp\arduino_build_272396 -warnings=none -build-cache C:\Users\DARUSS~1\AppData\Local\Temp\arduino_cache_729322 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.esptool.path=C:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.12 -verbose C:\Users\darussyifa\Downloads\FBQ2P48IU5NBGGQ\FBQ2P48IU5NBGGQ.ino
Using board 'nodemcuv2' from platform in folder: C:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0
Using core 'esp8266' from platform in folder: C:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0
Detecting libraries used...
"C:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip2/include" "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\DARUSS~1\AppData\Local\Temp\arduino_build_272396/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266" "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\variants\nodemcu" "C:\Users\DARUSS~1\AppData\Local\Temp\arduino_build_272396\sketch\FBQ2P48IU5NBGGQ.ino.cpp" -o "nul"
"C:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip2/include" "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\DARUSS~1\AppData\Local\Temp\arduino_build_272396/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266" "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\variants\nodemcu" "-ID:\Program Files\Arduino\libraries\Ai_Ardulib_SSD1306-master" "C:\Users\DARUSS~1\AppData\Local\Temp\arduino_build_272396\sketch\FBQ2P48IU5NBGGQ.ino.cpp" -o "nul"
"C:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip2/include" "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\DARUSS~1\AppData\Local\Temp\arduino_build_272396/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266" "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\variants\nodemcu" "-ID:\Program Files\Arduino\libraries\Ai_Ardulib_SSD1306-master" "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\Wire" "C:\Users\DARUSS~1\AppData\Local\Temp\arduino_build_272396\sketch\FBQ2P48IU5NBGGQ.ino.cpp" -o "nul"
"C:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip2/include" "-IC:\Users\darussyifa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\DARUSS~1\AppData\Local\Temp\arduino_build_272396/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ff [... cut ...]
  oled.setTextXY(1,0); oled.putString("                ");
  oled.setTextXY(2,0); oled.putString("                ");
  oled.setTextXY(3,0); oled.putString("                ");
  oled.setTextXY(4,0); oled.putString("                ");
  oled.setTextXY(5,0); oled.putString("                ");
  oled.setTextXY(6,0); oled.putString("                ");
  oled.setTextXY(7,0); oled.putString("                ");
  oled.setTextXY(0,0); oled.putString("                ");              
}

### Debug Messages

```
Debug messages go here
```

5chufti, 2019-01-11:
as you obviously are beginner you should do a lot of BEFOREHAND reading, best starting with the above questionaire (and mentioned docs). Then please close the issue and look for help at more suitable places like forum of graphic library you used or https://www.esp8266.com as this is no issue with the esp8266 arduino core ...

devyte, 2019-01-11:
You seem to be trying to build an app that uses a 3rd party lib with core 2.4.0. 
Previous comment is correct: this is not the right place to request such help. This is an issue tracker, meant for tracking issues in the core.
Closing.

closed 2019-01-11 by devyte

marktech77, 2019-01-11:
Im so sorry