esp8266/Arduino #5608: SSL error through installing git version

IMAN4K, 2019-01-11:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			9515f466846f665fa68a7169465ee89aaa97b67e
- Development Env:		other
- Operating System:		Windows

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

There's an error in last installing step(running `get.py`) with the following logs:

### Debug Messages

> Platform: i686-mingw32
> Downloading i686-w64-mingw32.esptool-f80ae31.zip
> Traceback (most recent call last):
>   File "get.py", line 116, in <module>
>     get_tool(tool)
>   File "get.py", line 75, in get_tool
>     urlretrieve(url, local_path, report_progress)
>   File "C:\Users\Zahra\Documents\depot_tools\win_tools-2_7_6_bin\python\bin\lib\urllib.py", line 94, in urlretrieve
>     return _urlopener.retrieve(url, filename, reporthook, data)
>   File "C:\Users\Zahra\Documents\depot_tools\win_tools-2_7_6_bin\python\bin\lib\urllib.py", line 240, in retrieve
>     fp = self.open(url, data)
>   File "C:\Users\Zahra\Documents\depot_tools\win_tools-2_7_6_bin\python\bin\lib\urllib.py", line 208, in open
>     return getattr(self, name)(url)
>   File "C:\Users\Zahra\Documents\depot_tools\win_tools-2_7_6_bin\python\bin\lib\urllib.py", line 437, in open_https
>     h.endheaders(data)
>   File "C:\Users\Zahra\Documents\depot_tools\win_tools-2_7_6_bin\python\bin\lib\httplib.py", line 969, in endheaders
>     self._send_output(message_body)
>   File "C:\Users\Zahra\Documents\depot_tools\win_tools-2_7_6_bin\python\bin\lib\httplib.py", line 829, in _send_output
>     self.send(msg)
>   File "C:\Users\Zahra\Documents\depot_tools\win_tools-2_7_6_bin\python\bin\lib\httplib.py", line 791, in send
>     self.connect()
>   File "C:\Users\Zahra\Documents\depot_tools\win_tools-2_7_6_bin\python\bin\lib\httplib.py", line 1176, in connect
>     self.sock = ssl.wrap_socket(sock, self.key_file, self.cert_file)
>   File "C:\Users\Zahra\Documents\depot_tools\win_tools-2_7_6_bin\python\bin\lib\ssl.py", line 387, in wrap_socket
>     ciphers=ciphers)
>   File "C:\Users\Zahra\Documents\depot_tools\win_tools-2_7_6_bin\python\bin\lib\ssl.py", line 143, in __init__
>     self.do_handshake()
>   File "C:\Users\Zahra\Documents\depot_tools\win_tools-2_7_6_bin\python\bin\lib\ssl.py", line 305, in do_handshake
>     self._sslobj.do_handshake()
> IOError: [Errno socket error] [Errno 1] _ssl.c:507: error:1407742E:SSL routines:SSL23_GET_SERVER_HELLO:tlsv1 alert protocol version

earlephilhower, 2019-01-11:
Looks like a SSL downgrade attack on the link from your computer and GitHub. Closing as this isn't a core problem.

closed 2019-01-11 by earlephilhower

IMAN4K, 2019-01-11:
@earlephilhower Could you please help to resolve this issue?