esp8266/Arduino #5612: Eclipse 2018-12 local.eagle.app.v6.common.ld: No such file or directory

Superfred1, 2019-01-13:
Hello, 
the build with Eclipse fails with the Error "cannot open linker script file local.eagle.app.v6.common.ld: No such file or directory"

Esp8266 Arduino 2.5.0-beta2
Eclipse Version: 2018-12 (4.10.0) Build id: 20181214-0600 Windows 1809 64 Bit
Plugin: Eclipse C++ IDE for Arduino 3.0 (not the Sloeber Plugin), installed per default in Eclipse .

Exact Log message:
....
"C:/Users/xxxx/.arduinocdt/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-gcc" -Wl,-Map "-Wl,./T2.map" -g -w -Os -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static "-LC:\Users\xxxx\.arduinocdt\packages\esp8266\hardware\esp8266\2.5.0-beta2/tools/sdk/lib" "-LC:\Users\xxxx\.arduinocdt\packages\esp8266\hardware\esp8266\2.5.0-beta2/tools/sdk/ld" "-LC:\Users\xxxx\.arduinocdt\packages\esp8266\hardware\esp8266\2.5.0-beta2/tools/sdk/libc/xtensa-lx106-elf/lib" "-Teagle.flash.512k.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read  -o "./T2.elf" -Wl,--start-group project/T2.cpp.o   "./core.a" -lhal -lphy -lpp -lnet80211 -llwip2-536-feat -lwpa -lcrypto -lmain -lwps -lbearssl -laxtls -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group  "-L."

c:/users/xxxx/.arduinocdt/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot open linker script file local.eagle.app.v6.common.ld: No such file or directory

Board selected: Generic ESP 8266 Module

Even with this minimal Sketch:
#include <Arduino.h>
void setup() {
}
void loop() {
}

ESP32 and AVR Boards working fine.

I tried this but still not working: https://arduino-esp8266.readthedocs.io/en/latest/eclipse/eclipse.html

earlephilhower, 2019-01-14:
The `local.eagle.app.v6.common.ld` file is generated by a stage of the platform.txt using settings from the IDE or `#defines` to adjust where things like vtables are placed.

I think the document you're pointing to in our readthedocs is related to the Sloeber(sp?) plugin, but you mention another plugin.  Is there some steps you used to get the generic Arduino plugin going?

Superfred1, 2019-01-15:
I forgot:
You have to enable the libraries you need in the project properties:
Alt+Enter, go to Libraries and enable what you need (Wifi, etc).

earlephilhower, 2019-01-15:
I was able to repro this locally.  It's in the Eclipse IDE, so I'm not sure we can do anything on our end.

I've opened the following bug report there:
https://bugs.eclipse.org/bugs/show_bug.cgi?id=543476

d-a-v, 2019-01-15:
@superfred You probably can use [a generated version](https://gist.github.com/d-a-v/a359cf856f9c3dd76b49c9041698882a) of this file until the bug is solved.

Superfred1, 2019-01-16:
@d-a-v Now it compiles, but the .bin file is not created:
...
"C:/Users/xxxx/.arduinocdt/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\xxxx\.arduinocdt\packages\esp8266\hardware\esp8266\2.5.0-beta2/tools/sdk/include" "-IC:\Users\xxxx\.arduinocdt\packages\esp8266\hardware\esp8266\2.5.0-beta2/tools/sdk/lwip2/include" "-IC:\Users\xxxx\.arduinocdt\packages\esp8266\hardware\esp8266\2.5.0-beta2/tools/sdk/libc/xtensa-lx106-elf/include" "-I./core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -fexceptions -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10608 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 -I"C:/Users/xxxx/.arduinocdt/packages/esp8266/hardware/esp8266/2.5.0-beta2/cores/esp8266" -I"C:/Users/xxxx/.arduinocdt/packages/esp8266/hardware/esp8266/2.5.0-beta2/variants/generic" "C:/Users/xxxx/.arduinocdt/packages/esp8266/hardware/esp8266/2.5.0-beta2/cores/esp8266/WString.cpp" -o "core/WString.cpp.o"
"C:/Users/xxxx/.arduinocdt/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-ar" cru  "./core.a" "core/WString.cpp.o"
"C:/Users/xxxx/.arduinocdt/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-gcc" -Wl,-Map "-Wl,./T4.map" -g -w -Os -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static "-LC:\Users\xxxx\.arduinocdt\packages\esp8266\hardware\esp8266\2.5.0-beta2/tools/sdk/lib" "-LC:\Users\xxxx\.arduinocdt\packages\esp8266\hardware\esp8266\2.5.0-beta2/tools/sdk/ld" "-LC:\Users\xxxx\.arduinocdt\packages\esp8266\hardware\esp8266\2.5.0-beta2/tools/sdk/libc/xtensa-lx106-elf/lib" "-Teagle.flash.512k.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read  -o "./T4.elf" -Wl,--start-group project/T4.cpp.o   "./core.a" -lhal -lphy -lpp -lnet80211 -llwip2-536-feat -lwpa -lcrypto -lmain -lwps -lbearssl -laxtls -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group  "-L."
Program store usage: 283331 of maximum 499696 bytes
Initial RAM usage: 26875 of maximum 81920 bytes

esptool v0.4.13-1-gf80ae31 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM3
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
error: stat ./T3.bin failed: No such file or directory

Superfred1, 2019-01-16:
@earlephilhower @d-a-v 
Thank you for looking into this. 
I have an older version of Eclipse + Sloeber working, so no problem for me. 

I reported this because I see a great potential for the future in the native Arduino support in Eclipse.

Have a nice day.
Fred

earlephilhower, 2019-01-16:
http://git.eclipse.org/c/cdt/org.eclipse.cdt.git/tree/toolchains/arduino/org.eclipse.cdt.arduino.core/src/org/eclipse/cdt/arduino/core/internal/build/ArduinoBuildConfiguration.java?id=ffdb004e6711ddffc0174341f97dfc44d8619246#n329

The Arduino builder only looks for a very few, un-numbered properties.  It looks like it's not a big deal to have it iterate over any numbered ones as well as stick prelink/etc. in there somewhere (assuming there sis some way to combine multiple `buildModel.put()`s

````
buildModel.put("recipe_cpp_o_pattern", resolveProperty("recipe.cpp.o.pattern", properties)); //$NON-NLS-1$ //$NON-NLS-2$
buildModel.put("recipe_c_o_pattern", resolveProperty("recipe.c.o.pattern", properties)); //$NON-NLS-1$ //$NON-NLS-2$
buildModel.put("recipe_S_o_pattern", resolveProperty("recipe.S.o.pattern", properties)); //$NON-NLS-1$ //$NON-NLS-2$
buildModel.put("recipe_ar_pattern", resolveProperty("recipe.ar.pattern", properties)); //$NON-NLS-1$ //$NON-NLS-2$
buildModel.put("recipe_c_combine_pattern", resolveProperty("recipe.c.combine.pattern", properties)); //$NON-NLS-1$ //$NON-NLS-2$
buildModel.put("recipe_objcopy_eep_pattern", resolveProperty("recipe.objcopy.eep.pattern", properties)); //$NON-NLS-1$ //$NON-NLS-2$
buildModel.put("recipe_objcopy_hex_pattern", resolveProperty("recipe.objcopy.hex.pattern", properties)); //$NON-NLS-1$ //$NON-NLS-2$
buildModel.put("recipe_objcopy_bin_pattern", resolveProperty("recipe.objcopy.bin.pattern", properties)); //$NON-NLS-1$ //$NON-NLS-2$
buildModel.put("recipe_size_pattern", resolveProperty("recipe.size.pattern", properties)); //$NON-NLS-1$ //$NON-NLS-2$
````

d-a-v, 2019-05-20:
@grothag This is on eclipse, not us. Issues there are opened, see [above post](https://github.com/esp8266/Arduino/issues/5612#issuecomment-454761882)

You can however use the eclipse sloeber extension (which we do not support here but users like @Superfred1 (and @hreintke I think) use with no issue). What about trying it ?

grothag, 2019-05-20:
Ah. Sorry. Thanks for fast answer.

adonoghu, 2020-01-27:
I found a solution here. [https://www.eclipse.org/forums/index.php/t/1095090/](url). It fixed the problem for me. I also had to change the section of the platform.txt file that generates the .bin file from the .elf file.
Here is the corrected '##Create hex' section of the platform.txt file.

`recipe.objcopy.hex.pattern="{runtime.tools.python3.path}/python3" "{runtime.tools.elf2bin}" --eboot "{runtime.tools.eboot}" --app "{build.path}/{build.project_name}.elf" --flash_mode {build.flash_mode} --flash_freq {build.flash_freq} --flash_size {build.flash_size} --path "{runtime.tools.xtensa-lx106-elf-gcc.path}/bin" --out "{build.path}/{build.project_name}.bin"
recipe.hooks.postobjcopy.01.pattern="{runtime.tools.python3.path}/python3" "{runtime.tools.signing}" --mode sign --privatekey "{build.source.path}/private.key" --bin "{build.path}/{build.project_name}.bin" --out "{build.path}/{build.project_name}.bin.signed" --legacy "{build.path}/{build.project_name}.bin.legacy_sig"
recipe.hooks.postobjcopy.02.pattern="{runtime.tools.python3.path}/python3" "{runtime.tools.sizes}" --elf "{build.path}/{build.project_name}.elf" --path "{runtime.tools.xtensa-lx106-elf-gcc.path}/bin"`

neorob85, 2020-04-13:
Actually every solutions I found, don't work on my specific case. I resolved simple downgrading the esp8266 tool to the versions 2.4.1 which works without problems.
To do this you can use this json instead the last one: https://github.com/esp8266/Arduino/releases/download/2.4.1/package_esp8266com_index.json

earlephilhower, 2020-05-26:
Eclipse will be removing the Arduino plugin completely, so this bug's solution will have to be "Use Sloeber" as we currently have.

Closing.

````
https://bugs.eclipse.org/bugs/show_bug.cgi?id=543476
Product/Component: CDT / cdt-arduino

Jonah Graham <jonah@kichwacoders.com> changed:

           What    |Removed                     |Added
----------------------------------------------------------------------------
             Status|NEW                         |CLOSED
         Resolution|---                         |WONTFIX

--- Comment #7 from Jonah Graham <jonah@kichwacoders.com> --- Thank you for taking the time to file a bug report about the Arduino component of the CDT project. The Arduino component is no longer maintained and will be removed from CDT 10 (Bug 562498). The final Eclipse CDT version that will include Arduino support is the CDT 9.11 series. 

Please consider using The Arduino Eclipse IDE and plug-ins named Sloeber
(https://marketplace.eclipse.org/node/2637354) for your future Eclipse CDT powered Arduino development.
````

closed 2020-05-26 by earlephilhower