esp8266/Arduino #5618: Exception 28 when use HTTPClient::getString()

hamidelectronic, 2019-01-14:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.4.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		4MB/2MB
- lwip Variant:   v2 Higher Bandwidth
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [other] (serial upload only)

### Problem Description

hello
i use httpclint to http get method for my webservice which sometimes causes a reset

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

bool cloudhttpRequest(String &result, String arg)
{
  result = "";
  HTTPClient http;
  http.begin("http://domain.com/api/" + arg);
  int httpCode = http.GET();
  if (httpCode > 0) {
    if (httpCode == HTTP_CODE_OK) {
      result = http.getString();
      http.end();
      return true;
    } else {
      http.end();
      return false;
    }
  } else {
    http.end();
    return false;
  }
}

```

### Debug Messages

```
Decoding stack results
0x401006dc: free at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc\umm_malloc.c line 1755
0x4023bbec: String::invalidate() at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\WString.cpp line 141
0x4023bc08: String::~String() at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\WString.cpp line 125
0x40236b6c: HTTPClient::getString() at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 759
0x401006dc: free at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc\umm_malloc.c line 1755
0x4023bdf6: String::operator=(String&&) at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\WString.cpp line 232
0x40209c6f: cloudhttpRequest(String&, String) at E:\Hamid\electronic\BirdCare\atmega8 version - Alarm\mycode_nodemcu/mycode_nodemcu.ino line 643
0x4023bddc: String::String(String const&) at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\WString.cpp line 41
0x402150f8: cloudstringGet(String&) at E:\Hamid\electronic\BirdCare\atmega8 version - Alarm\mycode_nodemcu/mycode_nodemcu.ino line 592
0x402430f8: _strtol_r at ../../../.././newlib/libc/stdlib/strtol.c line 186
0x402431c0: strtol at ../../../.././newlib/libc/stdlib/strtol.c line 224
0x402331f8: SPIClass::setFrequency(unsigned int) at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\SPI\SPI.cpp line 236
0x40235919: TFT_eSPI::getTouch(unsigned short*, unsigned short*, unsigned short) at C:\Users\hamid\Documents\Arduino\libraries\TFT_eSPI\Extensions/Touch.cpp line 137
0x402358fe: TFT_eSPI::getTouch(unsigned short*, unsigned short*, unsigned short) at C:\Users\hamid\Documents\Arduino\libraries\TFT_eSPI\Extensions/Touch.cpp line 131
0x40100575: realloc at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc\umm_malloc.c line 1499
0x4023ca44: esp_yield() at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_main.cpp line 91
0x40202dde: delay at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_wiring.c line 54
0x4023bc33: String::changeBuffer(unsigned int) at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\WString.cpp line 156
0x4023bc7f: String::reserve(unsigned int) at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\WString.cpp line 146
0x4023591c: TFT_eSPI::getTouch(unsigned short*, unsigned short*, unsigned short) at C:\Users\hamid\Documents\Arduino\libraries\TFT_eSPI\Extensions/Touch.cpp line 137
0x4021a5f3: cloudserver() at E:\Hamid\electronic\BirdCare\atmega8 version - Alarm\mycode_nodemcu/mycode_nodemcu.ino line 655
0x40237bbc: TwoWire::endTransmission(unsigned char) at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\Wire\Wire.cpp line 143
0x40224a34: tuchcheeck() at E:\Hamid\electronic\BirdCare\atmega8 version - Alarm\mycode_nodemcu/draw_lcd.ino line 6341
0x401067d4: millis at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_wiring.c line 183
0x4023ca44: esp_yield() at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_main.cpp line 91
0x4022b42c: ESP8266WiFiSTAClass::status() at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 498
0x4022a3a1: loop() at E:\Hamid\electronic\BirdCare\atmega8 version - Alarm\mycode_nodemcu/mycode_nodemcu.ino line 3697
0x4023cad4: loop_wrapper() at C:\Users\hamid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_main.cpp line 125
```

earlephilhower, 2019-01-14:
This is not an MCVE sketch so can't be used to do anything by anyone but yourself.  It needs to be fully compilable and runnable by someone without your specific code or system.

And again, this is an out of memory problem or memory corruption in your code in another routine.  All RAM is unavailable so the String is not to be able to do it's allocations.

earlephilhower, 2019-01-16:
Sorry, but there is nothing for us to do.  Your entire application has a memory leak somewhere.  See https://en.wikipedia.org/wiki/Memory_leak for a description of what a memory leak is, and then you'll need to find it and fix it in your complete application.

mateiemil97, 2020-02-23:
I faced same problem and my problem gone with an downgrade from 2.6.3 to 2.5.2

closed 2020-02-23 by devyte