esp8266/Arduino #5620: BearSSL with Insecure HTTPS

coder137, 2019-01-16:
Hey, I found this thread when facing the same issue. #5277 

When using `BearSSL::WifiClientSecure`
I get code -1 with `https.get`

``` c
HTTPClient https;
BearSSL::WiFiClientSecure newSecure;
newSecure.setInsecure();
int checkBegin = https.begin(newSecure, "raw.githubusercontent.com", 443, "/coder137/TM4C123G/master/.gitattributes", false);
Serial.println(checkBegin);

int code = https.GET();
String payload = https.getString();
Serial.println(code);
Serial.println(payload);

// !
https.end();
newSecure.stop();
```

However when using the default axTLS::WifiSecure
This works perfectly

``` c
HTTPClient https;
WiFiClientSecure newSecure;
int checkBegin = https.begin(newSecure, "raw.githubusercontent.com", 443, "/coder137/TM4C123G/master/.gitattributes", false);
Serial.println(checkBegin);

int code = https.GET();
String payload = https.getString();
Serial.println(code);
Serial.println(payload);

// !
https.end();
newSecure.stop();
```

However in the docs it has been written that BearSSL is preferred over axTLS since the latter is deprecated. 
Am I doing something wrong or is this a bug?

Jeroen88, 2019-01-16:
What if you try _true_ at the https.begin() function? You _are_ using the secure WiFi client over port 443, so https should be true, but without setting security credentials (because of setInsecure())

coder137, 2019-01-16:
Amazing this works as intended.

For reference this is the code I used
``` c
HTTPClient https;
BearSSL::WiFiClientSecure secureClient;
secureClient.setInsecure();
int checkBegin = https.begin(secureClient, "raw.githubusercontent.com", 443, "/coder137/TM4C123G/master/.gitattributes", true);
Serial.println(checkBegin);

int code = https.GET();
String string = https.getString();
Serial.println(code);
Serial.println(string);

https.end();
secureClient.stop();
```

Thanks a lot for your help @Jeroen88 🥇

closed 2019-01-16 by coder137