esp8266/Arduino #5621: tcp client(or https client) will stuck when freeheap is low

i3water, 2019-01-16:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git](https://github.com/esp8266/Arduino/commit/7ee503d3539b107ca6796af77d11ce8a267ff23d)
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[WiFiduino]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [|v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

I rewrite the BasicHTTPSClient to descrip this bug, when the freeheap is low the tcp client or http client will stuck. But the WDT or softWDT still run for a minutes or a long time.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
/**
   BasicHTTPSClient.ino

    Created on: 20.08.2018

*/

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>

#include <WiFiClientSecureBearSSL.h>
// Fingerprint for demo URL, expires on June 2, 2019, needs to be updated well before this date
const uint8_t fingerprint[20] = {0x5A, 0xCF, 0xFE, 0xF0, 0xF1, 0xA6, 0xF4, 0x5F, 0xD2, 0x11, 0x11, 0xC6, 0x1D, 0x2F, 0x0E, 0xBC, 0x39, 0x8D, 0x50, 0xE0};

ESP8266WiFiMulti WiFiMulti;

uint32_t buff_num = 5000;
uint32_t test[5000];

void checkHeap() {
  Serial.print("heap: ");
  Serial.println(ESP.getFreeHeap());
}

void setup() {
  for(uint32_t num = 0; num < buff_num; num++)
  {
    test[num] = 0xFFFFFFFF;
  }

  Serial.begin(115200);
  // Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

  for (uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] WAIT %d...\n", t);
    Serial.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("mostfun", "18038083873");
}

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

    std::unique_ptr<BearSSL::WiFiClientSecure>client(new BearSSL::WiFiClientSecure);

    //client->setFingerprint(fingerprint);
    client->setInsecure();

    HTTPClient https;

    Serial.print("[HTTPS] begin...\n");

    checkHeap();
    
    if (https.begin(*client, "https://jigsaw.w3.org/HTTP/connection.html")) {  // HTTPS

      Serial.print("[HTTPS] GET...\n");
      // start connection and send HTTP header
      int httpCode = https.GET();

      // httpCode will be negative on error
      if (httpCode > 0) {
        // HTTP header has been send and Server response header has been handled
        Serial.printf("[HTTPS] GET... code: %d\n", httpCode);

        // file found at server
        if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
          String payload = https.getString();
          Serial.println(payload);
        }
      } else {
        Serial.printf("[HTTPS] GET... failed, error: %s\n", https.errorToString(httpCode).c_str());
      }

      https.end();
    } else {
      Serial.printf("[HTTPS] Unable to connect\n");
    }
  }

  Serial.println("Wait 10s before next round...");
  delay(10000);
}
```

### Debug Messages

```

SDK:3.0.0-dev(c0f7b44)/Core:win-2.5.0-dev=-195000/lwIP:STABLE-2_1_2_RELEASE/glue:1.0-11-g87c709d/BearSSL:79edbc8

[SETUP] WAIT 4...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 

connected with mostfun, channel 6
dhcp client start...
[SETUP] WAIT 3...
ip:192.168.1.156,mask:255.255.255.0,gw:192.168.1.1
[SETUP] WAIT 2...
[SETUP] WAIT 1...
[HTTPS] begin...
heap: 23352
[HTTP-Client][begin] url: https://jigsaw.w3.org/HTTP/connection.html
[HTTP-Client][begin] host: jigsaw.w3.org port: 443 url: /HTTP/connection.html
[HTTPS] GET...
pm_err,flash_tmp alloc fail
[HTTP-Client] failed connect to jigsaw.w3.org:443
[HTTP-Client][returnError] error(-1): connection refused
[HTTPS] GET... failed, error: connection refused
[HTTP-Client][end] tcp is closed
Wait 10s before next round...
[HTTPS] begin...
heap: 23120
[HTTP-Client][begin] url: https://jigsaw.w3.org/HTTP/connection.html
[HTTP-Client][begin] host: jigsaw.w3.org port: 443 url: /HTTP/connection.html
[HTTPS] GET...
[HTTP-Client] failed connect to jigsaw.w3.org:443
[HTTP-Client][returnError] error(-1): connection refused
[HTTPS] GET... failed, error: connection refused
[HTTP-Client][end] tcp is closed
Wait 10s before next round...
[HTTPS] begin...
heap: 22920
[HTTP-Client][begin] url: https://jigsaw.w3.org/HTTP/connection.html
[HTTP-Client][begin] host: jigsaw.w3.org port: 443 url: /HTTP/connection.html
[HTTPS] GET...

Abort called

>>>stack>>>

ctx: cont
sp: 3ffffc30 end: 3fffffc0 offset: 01b0
3ffffde0:  c012c008 c00dc003 3fff4d3c 40204789  
3ffffdf0:  3fffba0c 00000000 00000000 00000000  
3ffffe00:  3fffb77c 3fff66b4 3fff4d3c 40205021  
3ffffe10:  00000000 3ffffe70 402085fc 3fffefa0  
3ffffe20:  00000000 00000000 00000000 402014df  
3ffffe30:  00000d50 3fff4b04 3fff4d3c 40203f59  
3ffffe40:  000001bb 0000004f 3ffffe94 00000000  
3ffffe50:  000001bb 3fff4d3c 3fff66b4 00000000  
3ffffe60:  000001bb 3fff4d3c 3fff66b4 402051b6  
3ffffe70:  40209088 15341e80 3fffff10 40208c00  
3ffffe80:  3fff3bc0 3fff4d3c 3fffff10 4020623d  
3ffffe90:  3ffe88fe 3fffff88 3fff4b3c 40100ac8  
3ffffea0:  00000001 00000000 3fffff10 40206a08  
3ffffeb0:  00000030 3fffff88 00000001 3fff3ce4  
3ffffec0:  00000001 3fff4d3c 3fff3bc0 402071c0  
3ffffed0:  3ffe88e6 3fff4d3c 3fff4afc 4020749d  
3ffffee0:  3ffe85d8 00000000 3fffff88 3fff3ce4  
3ffffef0:  00000001 3fff4d3c 3fff3bc0 40206ab6  
3fffff00:  3fffdad0 3fff4d3c 3fff3bc0 4020278d  
3fffff10:  00000000 00000000 3fff4d3c 3fff66b4  
3fffff20:  0000000f 0000000d 000001bb 40001388  
3fffff30:  3fff4bc4 0000001f 00000015 3fff4e44  
3fffff40:  0000000f 00000005 3fff4e64 0000000f  
3fffff50:  00000000 3fff4ad4 0000001f 00000011  
3fffff60:  3fff4e84 0000000f 00000000 00000000  
3fffff70:  00000000 00000000 ffffffff 3fffef00  
3fffff80:  00000000 00000000 00000000 00000000  
3fffff90:  00000000 00000000 00000001 4020861d  
3fffffa0:  3fffdad0 00000000 3fff3cb4 402086a8  
3fffffb0:  feefeffe feefeffe 3ffe850c 40100b61  
<<<stack<<<

last failed alloc call: 40210E08(32)

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

d-a-v, 2019-01-16:
Thanks
You removed your logs that were useful, even more if you decode the (removed) stack trace and also enable OOM(OutOfMemory) debug option before running your sketch.

i3water, 2019-01-16:
> Thanks
> You removed your logs that were useful, even more if you decode the (removed) stack trace and also enable OOM(OutOfMemory) debug option before running your sketch.

Thanks,
I think the logs not same as I descrip, so I removed it. I will enable OOM debug and hope will get more useful logs.

i3water, 2019-01-20:
@d-a-v 
Hi,
dear d-a-v, I update the example codes, this codes can stuck and will not Abort called for a long time(may hous or days).
Excuse me, is there some way to avoid stuck when there is low heap?
I have set a function like below codes, but it will stuck some times. And with different version codes, the compare heaps is different.
```
if (ESP.getFreeHeap() < 4000) return;
```

jshankar494, 2019-05-24:
Board used:- ESP8266 - WeMos

Abort called

>>>stack>>>

ctx: cont
sp: 3ffffbb0 end: 3fffffc0 offset: 01b0
3ffffd60:  3fffdad0 00000001 3ffef4dc 3ffef60c  
3ffffd70:  3fffdad0 00000001 3ffef4dc 3ffef60c  
3ffffd80:  3ffef200 00000000 3ffef4dc 401004be  
3ffffd90:  feefeffe 00000000 3ffef4dc 40203fc4  
3ffffda0:  feefeffe feefeffe feefeffe feefeffe  
3ffffdb0:  feefeffe feefeffe feefeffe feefeffe  
3ffffdc0:  feefeffe feefeffe feefeffe feefeffe  
3ffffdd0:  feefeffe feefeffe feefeffe feefeffe  
3ffffde0:  feefeffe feefeffe feefeffe feefeffe  
3ffffdf0:  feefeffe feefeffe feefeffe feefeffe  
3ffffe00:  feefeffe feefeffe feefeffe feefeffe  
3ffffe10:  feefeffe feefeffe feefeffe feefeffe  
3ffffe20:  feefeffe feefeffe feefeffe feefeffe  
3ffffe30:  feefeffe feefeffe feefeffe feefeffe  
3ffffe40:  feefeffe feefeffe feefeffe feefeffe  
3ffffe50:  feefeffe feefeffe feefeffe feefeffe  
3ffffe60:  feefeffe feefeffe feefeffe feefeffe  
3ffffe70:  feefeffe feefeffe feefeffe feefeffe  
3ffffe80:  feefeffe feefeffe feefeffe feefeffe  
3ffffe90:  feefeffe feefeffe feefeffe feefeffe  
3ffffea0:  feefeffe feefeffe feefeffe feefeffe  
3ffffeb0:  feefeffe feefeffe feefeffe feefeffe  
3ffffec0:  feefeffe feefeffe feefeffe feefeffe  
3ffffed0:  feefeffe feefeffe feefeffe feefeffe  
3ffffee0:  feefeffe feefeffe feefeffe feefeffe  
3ffffef0:  feefeffe feefeffe feefeffe feefeffe  
3fffff00:  feefeffe feefeffe feefeffe feefeffe  
3fffff10:  feefeffe feefeffe feefeffe feefeffe  
3fffff20:  feefeffe feefeffe feefeffe feefeffe  
3fffff30:  feefeffe feefeffe feefeffe feefeffe  
3fffff40:  feefeffe feefeffe feefeffe feefeffe  
3fffff50:  feefeffe feefeffe feefeffe feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  feefeffe feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3ffef60c  
3fffffa0:  3fffdad0 00000000 3ffef5dc 4020c264  
3fffffb0:  feefeffe feefeffe 3ffe8570 401007a9  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vac02aff5
~ld
ISR not in IRAM!

Please help me to figure out the issue.

Thanks