esp8266/Arduino #5622: upgrade Ethernet library to pjrc Version 2.0.0

wolfgangr, 2019-01-16:
There is a new Ethernet library been developed by Paul Stoffregen 
https://www.pjrc.com/arduino-ethernet-library-2-0-0/
which has already found its way into arduino mainstream:
https://github.com/arduino-libraries/Ethernet/blob/master/library.properties

Here at esp8266/Arduino, current master is still at "version=1.0.4"
https://github.com/esp8266/Arduino/blob/master/libraries/Ethernet/library.properties

In his linked article, Paul reports succesful testing on a Adafruit Huzzah ESP8266 Feather , so I hoped, this lib would be copmatible or even included in esp8266/Arduino.

among other, I'd desperately use `Ethernet.hardwareStatus() ` and `Ethernet.linkStatus()` to debug my network test environment.

### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
Search for "Ethernet" does not macht any document
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
    no stack dump
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12] + WIZNET W5100
- Core Version:			[2.5.0-beta2] and [2.4.2]
- Development Env:		[Arduino IDE] ARDUINO 1.8.8 2018.12.07
- Operating System:		[debian 9.3]

### Settings in IDE

- Module:  			[Generic ESP8266] and |Nodemcu 0.9]
- Flash Mode:			 N/A
- Flash Size: 		[4MB] no SPIFFS
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		 N/A
- Flash Frequency:	 N/A
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] 

### Problem Description

When I override the Ethernet lib shipped with esp8266/Arduino with the new 2-0-0 Ethernet lib, I get loads of `error: invalid abstract return type for member function 'EthernetClient EthernetServer::available()'`. 
So obvioulsy there is more to change in the ESP8266 interface then just dropping this library.

Would be great to get it to work, however. Among else, it promises **extended hardware and link state debugging.** Im trying at the DNSAddressPrinter example. I see DHCPDISCOVER and DHCPOFFER in my router log, but communication seems to break down then.

I could not manage to test 2-0-0 Ethernet libs using the Arduino IDE.
So I did a forceful override:
```
~/.arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta2/libraries$ rm -rf Ethernet/
:~/.arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta2/libraries$ ln -s ~/Arduino/libraries/Ethernet-2.0.0/ ./Ethernet
```
### Debug Messages
Full error message:
```
Arduino: 1.8.8 (Linux), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Flash, Enabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from /home/wrosner/Arduino/libraries/Ethernet-2.0.0/src/EthernetServer.h:3:0,
                 from /home/wrosner/test/ESP8266/Ethernet-2.0.0/examples/DhcpAddressPrinter/DhcpAddressPrinter.ino:1:
/home/wrosner/Arduino/libraries/Ethernet-2.0.0/src/Ethernet.h:258:17: error: invalid abstract return type for member function 'EthernetClient EthernetServer::available()'
  EthernetClient available();
                 ^
/home/wrosner/Arduino/libraries/Ethernet-2.0.0/src/Ethernet.h:214:7: note:   because the following virtual functions are pure within 'EthernetClient':
 class EthernetClient : public Client {
       ^
In file included from /home/wrosner/Arduino/libraries/Ethernet-2.0.0/src/Ethernet.h:52:0,
                 from /home/wrosner/Arduino/libraries/Ethernet-2.0.0/src/EthernetServer.h:3,
[... 155 lines cut ...]
      //print your local IP address:
      Serial.print("My IP address: ");
      Serial.println(Ethernet.localIP());
      break;

    case 3:
      //rebind fail
      Serial.println("Error: rebind fail");
      break;

    case 4:
      //rebind success
      Serial.println("Rebind success");
      //print your local IP address:
      Serial.print("My IP address: ");
      Serial.println(Ethernet.localIP());
      break;

    default:
      //nothing happened
      break;
  }
}

```

d-a-v, 2019-01-16:
@wolfgangr 

You can add in `Ethernet.h` around L220
```
virtual bool flush(unsigned int maxWaitMs = 0) { (void)maxWaitMs; return flush(); }
virtual bool stop(unsigned int maxWaitMs = 0) { (void)maxWaitMs; return stop(); }
```
and
```
virtual int connect(const IPAddress& ip, uint16_t port) { return connect((IPAddress)ip, port) };
```
Compilation error or infinite loop may happen with this one, better change `connect()` signature to match this one instead (in .h and .cc)

```
virtual IPAddress remoteIP() const;
virtual IPAddress remotePort() const;
```
You may add `const` like above in  `Ethernet.h` L239,L240

@PaulStoffregen What version of this arduino core did you use for your testings ?

We could also replace the old version with Paul's updated one, but it's better if we could
use it as a submodule like we did with `SoftwareSerial`.

PaulStoffregen, 2019-01-16:
I am sad to report the testing I did last summer had DNS lookup fail on ESP8266.  Here's a blog article I wrote at the time.  Scroll to the end for test data.

https://www.pjrc.com/arduino-ethernet-library-2-0-0/

wolfgangr, 2019-01-16:
@d-a-v 
> edit: that was you :) d-a-v/W5500lwIP#4

yepp. small world, eh :)...
Just wanted to test basic functionality, before I dived into your testing instructions - and got stuck even at that basic level...

@PaulStoffregen 
> I am sad to report the testing I did last summer had DNS lookup fail on ESP8266. 

Ashes on my head: I have linked your article but not read it to the bottom?
So I'll try my test with a fixed IP?
But it does not even build... 
So, if you just could point me to some git to ESP8266-arduino where you built your tests aginst, I might be happy.

**Some remarks on the wider view angle**

In your blog, you quest for support from the ESP community.

So, I assume, there is no easy solution at hand yet.
This begs the question whether it is worth the effort to fix it.

Looks like just another case of ordered network configuration is conflicting with opaque ESP8266 core .
ESP8266 always has WiFi on it, and even if I look for ESP32, modules without WiFi sell at a multifold price than the ones with WiFi included.

While I think it is safe to say that WiFi sticks to ESP (both ...8266 and ...32) like s...orry..., 
not all use cases do so.

I see different constellations if it comes to ethernet on the ESP8266:
1. switch the onboard WiFi off, when Ethernet is active
2. independent but coexistent implementation of WiFI and Ethernet
3. integrate WiFi and Ethernet in a single network stack
4. rewrite the whole ESP core to overcome its quirks

**regarding 4.)**
While a real FOSS ESP core replacement would be the dream scenario, I don't feel elected to contribute, not even comment anything on that.

**regarding 1.)**
Disabling WiFi when I need ethernet would be OK for my test purpose currently at hand.

It also might fulfill the need e.g. of many tasmota and other IoT framework use cases:
Stay with the same platform and if possible with the same module, no matter whether a thingie (e.g. sensor interface, realy, button,....) is connected by wire or by thin WiFi air.

Of course it complicates code maintenance of such projects, that may end up with loads of #ifdefineds, and spoils the on-binary-for-all-purpose approach of e.g. tasmota.

Maybe, this might be implemented by a complete seperate fork of esp8266, which disables the onboard WiFi and just uses the MCU core. 
Maybe this even could be designed without any proprietary ESP code at all?
Is something like that flowing around?

**regarding 2.)**
Use case examples of coexisting but independent WiFi and Ethernet:
- Have tasmota running it's web GUI over WiFI and MQTT over Ethernet
- running debug, config or whatsoever over one and the app on the other
- small proxy-like thinges operating at higher OSI layers, where the app has two different stacks open
Are those cases really worth the effort?

**regarding 3.)**
WiFi and Ethernet in a single stack
The app does not care whether the core delivers its package by wire or by air.
I think this is what @d-a-v aims at with his lwip integration - correct?
Pls correct my when I'm wrong, but I think this is not possible with the Arduino ethernet Module, OK?
If it were possible to have ethernet in lwip, use cases 1.) and 2.) could be satisfied, too.

So, to make the point:
I opened this issue because I thought there was an easy solution around, that had just been overlooked. 
If there is much work to do, I'd prefer to invest this in the lwip based architecture.

d-a-v, 2019-01-16:
> *regarding 3.)*
> WiFi and Ethernet in a single stack
> The app does not care whether the core delivers its package by wire or by air.
> I think this is what @d-a-v aims at with his lwip integration - correct?

correct.

We could however propose a PR to make Paul's library compatible with esp8266 arduino core if he's opened with that. Small changes I guess, even for DNS not working. After that, his library could be submoduled here and replace the old Ethernet library.
For the sake of sketch portability without effort, the arduino way.

PaulStoffregen, 2019-01-16:
I'm currently very busy with a hardware project that's far behind schedule, so limited time I can spent right now.

I can tell you Ethernet 2.0.0 (on Arduino's repository) works on all other boards and it uses only the SPI library to access hardware.

PaulStoffregen, 2019-01-16:
> So, if you just could point me to some git to ESP8266-arduino where you built your tests aginst, I might be happy.

Sorry, I've deleted and reinstalled Arduino many times since the summer, even restarted new preferences.txt files several times (partly working on a bug added in Arduino 1.8.8).  It's no longer in the boards manager at all.

There is a lingering copy if the esp8266 package on my machine with this pathname: ~/.arduino15/packages/esp8266/hardware/esp8266/2.4.1

PaulStoffregen, 2019-01-16:
On this part:

> We could however propose a PR to make Paul's library compatible with esp8266 arduino core if he's opened with that.

I'm no longer maintaining a separate fork of the Ethernet library.  So what you're talking about is a patch to Ethernet 2.0.0 on Arduino's repository, meant for all Arduino compatible boards.

Adding more called to yield() in places that wait would probably be a very welcome patch.  Except in the optimized chip select pin stuff, very ESP specific stuff probably isn't appropriate in Arduino's repository.  The Wiznet chip is accessed using the SPI library, so in theory all you should need to do is make sure ESP8266 has a SPI library which follows Arduino's specs.

jandrassy, 2019-02-20:
esp8266 developers should not change the Arduino AIP base classes Client and Server

https://github.com/esp8266/Arduino/issues/5749

Jason2866, 2019-10-20:
Any news for integrating wired ethernet?

Smetanski, 2020-05-20:
Hi guys, what is going on with Ethernet lib integration ??

d-a-v, 2020-05-20:
@Smetanski you can either try #6812 or try the [0.0.2 alpha release](https://d-a-v.github.io).
Please report whether it's working with you.

devyte, 2020-05-20:
Or try #6680 which has a different Ethernet integrated with lwip and the wifi interfaces.

d-a-v, 2020-11-09:
#6812 is merged

closed 2020-11-09 by d-a-v