esp8266/Arduino #5628: bug in pgmspace.h for "float" in the if ... and else parts

Hoek67, 2019-01-18:
the lines

    #define pgm_read_float(addr)            (*reinterpret_cast<const float*>(addr))			 //dw fixed
    #define pgm_read_float(addr)            (*(const float*)(addr))				// dw fixed

was/is 

    #define pgm_read_float(addr)            (*reinterpret_cast<const float>(addr))	// error		 
    #define pgm_read_float(addr)            (*(const float)(addr))				// error

Notice both were missing the pointer part on the cast.

Hoek67, 2019-01-18:
Downloaded and installed "latest" esp8266 a few nights ago... but seems it wasn't the latest.

Hoek67, 2019-01-19:
checked and found the latest version and it still has the float issue. Another issue is uint32_t and float should use the aligned code still. Reason being I was accessing a float and uint32_t variables inside a blob and it kept crashing. I made uint32_t and float use the aligned read but it still crashed. Reason being 1st byte was in 1 4 byte block and the next 3 were in the next. This can also happen for uint16_t inside blobs. Ended up writing a quick and nasty replacement function that takes into account the bytes being on different 4 byte boundaries.

[code]
	static inline uint32_t pgm_read_with_offset(const void *addr)
	{
		uint32_t addr32 = (uint32_t)addr;
		uint32_t offset = addr32 & 3; // 0 = aligned ... 1 = 1 out etc

		if (!offset) // aligned
		{
			return *(uint32_t *)addr;
		}

		uint32_t dword1 = *(uint32_t *)(addr32 & 0xFFFFFFFC);  // align to 4 bytes
		uint32_t dword2 = *(uint32_t *)((addr32 + 4) & 0xFFFFFFFC); // align to 4 bytes then add 4 to go to next boundary

		offset *= 8; // bytes to bits

		return (dword1 >> (offset)) | (dword2 << (32 - offset));
	}		
[/code]

earlephilhower, 2019-01-23:
Updated title to better reflect the question.

The underlying reason is that misaligned word and half-word reads are not supported on the CPU in the 8266, and result in a hardware exception.  Many RISC CPUs have this same requirement (but ARMv7 supports it while earlier v5/etc. don't...so it "just works" on Arduino-brand ARM SAMD boards)

The current code is focused on maximum speed and minimum size.  Aligned 32-bit operands can be read directly from flash with a single instruction, like reading from any other part of memory.  Aligned 16-bit reads read the 32-bit word in 1 swoop and pick the upper or lower half with a couple instructions.

Adding in misaligned support will decrease read performance in the common, aligned case, by 2-3x and increase each instance of the inline pgm_read_* by probably 4-5x, but it needs to be written in assembly first.

The question up for debate is whether to:
1. Support full Arduino semantics (based off of an 8-bit CPU which couldn't even read a 16- or 32-bit value, so all word/halfwords are SW emulated) with respective code bloat and slowdown
2. Same as above, but add in a pgm_read_aligned macro which would be the existing ones.  All code in the core and libraries would need to be patched to use this to regain space/performance, but Arduino 8-bit stuff would just work
3. Inverse of above.  Support it with a pgm_read_*_unaligned macro that can be selectively used by people with the need (i.e. from packed structures coming from 8-bit-land)
4. Not support it at all because the HW can't do it natively

I'd be interested to see if others have any strong feelings one way or the other.  I'm inclined to (3) or (4) since the others mean a massive bloat or lots of work on SSL and core libs.

Makuna, 2019-01-24:
To me this is simple.  If the API is present in Arduino, it should be compatible with no changes to library or sketch. If performance is an issue, then enhancing the API to expose the support is what you should do.  So #2 above.     When I wrote the original this was the goal.  I guess the goal had gotten changed over time by other contributors.

earlephilhower, 2019-01-24:
So, actually, @Makuna, there's only a few spots in the core where pgm_read_word or pgm_read_dword are called.

I don't even bother w/pgm_read_dword in BearSSL.a (just read the 32-bit), but there are ~30 spots w/uint16_t's being read out that would need patching.

libc doesn't have it, it only does byte-wise pmem access.

LWIP, we'd need to @d-a-v to check.

devyte, 2019-01-24:
I'd like to point out here that slowing this down in the name of compatibility could be a breaking change, if the slowdown is enough to cause a side effect in some app or lib, e.g.: trigger the WDT, increase power consumption due to additional wake time, etc.

How about a mix of points 2 and 3?
- Implement pgm_*_aligned() with the current code (fast, smaller bin, but crashes with unaligned addresses)
- Implement pgm_*_unaligned() with the proposed code (compatible, but several X slower, and bigger bin)
- Implement pgm_*() as a simple forwarder to one of the two above. This would work as a selector.

Users who need one or the other can change the definition to the one they need. Or they can directly use the version they need.

Compatibility is good, but so is performance. The trade-off can be needed in either way depending on the user's need. Whatever else, I think the choice to go one way or the other should be available in some form, and having a kind of master switch for the behavior is easier to manage on our side, and easier to handle/migrate/whatever on the user side.

earlephilhower, 2019-01-24:
I was worried like you were, too, @devyte.  But after looking through things in the core and libs, I'm not so worried about performance.  The majority of PMEM accesses are bytewise (strings) which have no change or 32-bit direct w/o pgm_* macros (memcpy_P, strcpy_P, etc. which do byte until things are aligned, then go int32-wise).

3rd party libs, I would hazard, are going to have a similar profile.

earlephilhower, 2019-01-24:
Some sample code (ignore the ret.n, that'll go away with inline)
````
00000000 <pgm_read_dword>:
   0:	1258                	l32i.n	a5, a2, 4
   2:	0248                	l32i.n	a4, a2, 0
   4:	143020               	extui	a3, a2, 0, 2
   7:	602030               	neg	a2, a3
   a:	401200               	ssl	a2
   d:	a12500               	sll	a2, a5
  10:	400300               	ssr	a3
  13:	913040               	srl	a3, a4
  16:	202230               	or	a2, a2, a3
  19:	f00d                	ret.n
	...

0000001c <pgm_read_word>:
  1c:	1258                	l32i.n	a5, a2, 4
  1e:	0248                	l32i.n	a4, a2, 0
  20:	143020               	extui	a3, a2, 0, 2
  23:	602030               	neg	a2, a3
  26:	401200               	ssl	a2
  29:	a12500               	sll	a2, a5
  2c:	400300               	ssr	a3
  2f:	913040               	srl	a3, a4
  32:	202230               	or	a2, a2, a3
  35:	f42020               	extui	a2, a2, 0, 16
  38:	f00d                	ret.n
	...

0000003c <pgm_read_dword_aligned>:
  3c:	0228                	l32i.n	a2, a2, 0
  3e:	f00d                	ret.n

00000040 <pgm_read_word_aligned>:
  40:	001222               	l16ui	a2, a2, 0
  43:	f00d                	ret.n
````

Dword: 1 insn, 2 bytes => 9 insn, 31 bytes
Word: 1 insn, 3 bytes => 10 insn, 32 bytes

Makuna, 2019-01-24:
>           I hear you, compatibility is a good thing, but taken to extremes you end up with Windows 10.  Plus, we're running a little shy of the MSFT staffing levels. :(
Windows are issues around backward compatibility, what we are talking about here is compatibility across platforms.  The problem is more akin with the mess that happened with HTML 5 (way too many platform ifdefs).

earlephilhower, 2019-01-29:
#5692 will track the PR for implementing this.  Any assembly optimizations you can come up with for the misaligned versions of the macros would be appreciated, because there's pretty severe inflation even at -O3 which produces ASM which doesn't look bad at all to me...

PR #5692 (2019-04-11): Unaligned access support for pgm_read_word/dword
Two versions of the multibyte `pgm_read_*` macros are now available:  `pgm_read_*_aligned()` and `pgm_read_*_unaligned()`.  The `_aligned()` versions are the original ones that require machine-dependent alignment of 2- and 4-byte values and compile into 1-2 instructions, while the `_unaligned()` versions have no such restriction (but compile to 15x the instructions and take ~4x to ~15x the cycles)).

The standard `pgm_read_(dword/word)` macros are defined to the `unaligned()` versions to provide Arduino compatibility out of the box.

It looks to [... cut]
closed 2019-04-11 by earlephilhower