esp8266/Arduino #5630: Errors when used TomeLib in ESP8266 Arduino Core master branch from GitHub

mikekgr, 2019-01-18:
Dear Sirs,
recently I sync my ESP8266 Arduino Core from GitHub master branch.
I discovered that, in all of my sketches that I use the TimeLib library ( #include <TimeLib.h> ) I got errors at the compilation, even with the simpler sketch that use the TimeLib library  ex. : 
`
/*
  Blynk can provide your device with time data, like an RTC.
  Please note that the accuracy of this method is up to several seconds.

  App project setup:
    RTC widget (no pin required)
    Value Display widget on V1
    Value Display widget on V2

  WARNING :
  For this example you'll need Time keeping library:
    https://github.com/PaulStoffregen/Time

  This code is based on an example from the Time library:
    https://github.com/PaulStoffregen/Time/blob/master/examples/TimeSerial/TimeSerial.ino
 *************************************************************/

#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>
#include <TimeLib.h>
#include <WidgetRTC.h>

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "YourAuthToken";

// Your WiFi credentials.
// Set password to "" for open networks.
char ssid[] = "YourNetworkName";
char pass[] = "YourPassword";

BlynkTimer timer;

WidgetRTC rtc;

// Digital clock display of the time
void clockDisplay()
{
  // You can call hour(), minute(), ... at any time
  // Please see Time library examples for details

  String currentTime = String(hour()) + ":" + minute() + ":" + second();
  String currentDate = String(day()) + " " + month() + " " + year();
  Serial.print("Current time: ");
  Serial.print(currentTime);
  Serial.print(" ");
  Serial.print(currentDate);
  Serial.println();

  // Send time to the App
  Blynk.virtualWrite(V1, currentTime);
  // Send date to the App
  Blynk.virtualWrite(V2, currentDate);
}

void setup()
{
  // Debug console
  Serial.begin(9600);

  Blynk.begin(auth, ssid, pass);
  // You can also specify server:
  //Blynk.begin(auth, ssid, pass, "blynk-cloud.com", 80);
  //Blynk.begin(auth, ssid, pass, IPAddress(192,168,1,100), 8080);

  // Begin synchronizing time
  rtc.begin();

  // Other Time library functions can be used, like:
  //   timeStatus(), setSyncInterval(interval)...
  // Read more: http://www.pjrc.com/teensy/td_libs_Time.html

  // Display digital clock every 10 seconds
  timer.setInterval(10000L, clockDisplay);
}

void loop()
{
  Blynk.run();
  timer.run();
}
`

The errors that I got are:
In file included from E:\ARDUINO_ESP_DEV\arduino-IDE\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include/string.h:163:0,

                 from E:\ARDUINO_ESP_DEV\arduino-IDE\portable\sketchbook\libraries\Time\DateStrings.cpp:22:

E:\ARDUINO_ESP_DEV\arduino-IDE\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include/sys/string.h:32:0: warning: "strcpy_P" redefined [enabled by default]

 #define strcpy_P(dest, src)          strncpy_P((dest), (src), SIZE_IRRELEVANT)

 ^

E:\ARDUINO_ESP_DEV\arduino-IDE\portable\sketchbook\libraries\Time\DateStrings.cpp:20:0: note: this is the location of the previous definition

 #define strcpy_P(dest, src) strcpy((dest), (src))

Please help me, what is so wrong?

Thanks and Best Regards,
Mike Kranidis

devyte, 2019-01-18:
Your error message is telling you what's wrong:
> E:\ARDUINO_ESP_DEV\arduino-IDE\portable\sketchbook\libraries\Time\DateStrings.cpp:20:0: note: this is the location of the previous definition

Your  TimeLib is redefining strcpy_P.

Closing due to 3rd party lib.

closed 2019-01-18 by devyte

mikekgr, 2019-01-18:
> Your error message is telling you what's wrong:
> 
> > E:\ARDUINO_ESP_DEV\arduino-IDE\portable\sketchbook\libraries\Time\DateStrings.cpp:20:0: note: this is the location of the previous definition
> 
> Your TimeLib is redefining strcpy_P.
> 
> Closing due to 3rd party lib.

Dear @devyte yes but with all the same, when I switch back to 2.4.2 there is not problem...

devyte, 2019-01-18:
Yes, that's because newlib was updated and progmem support is now built-in. Just remove the #define in Timelib, or make it conditional, and make a pr for the author. Whatever else, that define doesn't belong in the lib, at least not in its current form.

mikekgr, 2019-01-18:
Dear @devyte 
Many thanks for the given explanation. I appreciate your time to explain me the problem. I will follow your suggestions.
Best Regards,
Mike Kranidis

mikekgr, 2019-01-22:
Dear @devyte 
following this "error" in lib, the lib maintainer, @PaulStoffregen in the bellow link says:
"I'm depending on people from the ESP community to submit pull requests for this sort of issue.
Please ask someone from the ESP community to create a correct pull request"

So please, can you do that? I am unable to work out this. Thanks in advance!

https://github.com/PaulStoffregen/Time/issues/110#issuecomment-456284922

Best Regards,
Mike Kranidis