esp8266/Arduino #5638: String is interfering with dereferenced int* causing system crash

euler2019, 2019-01-19:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			2.5.0-beta2 and 2.4.2
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			NodeMCU 1.0 (ESP-12E Module)
- Flash Mode:			dio
- Flash Size: 		4MB 1M SPIFFS
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

If a String is declared before a int pointer, dereferencing the pointer and setting a value will cause the system to crash. When the String declaration is removed everything works as expected. The problem also occurs in 2.4.2, but it must be nested in a function.

The stack dump is not helpful.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch for 2.5.0-beta2. This code will run on 2.4.2.

```cpp

#include <Arduino.h>

void setup()
{

  Serial.begin(115200);

  String a; //remove String declaration and it will work 
  int *c;
  *c = 4;

  Serial.println(*c);
  
}

void loop()
{
}

```

### Debug Messages

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v0fd86a07
~ld
Decoding stack results
0x40202960: loop_wrapper() at /Users/home/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta2/cores/esp8266/core_esp8266_main.cpp line 122
```
### [MCVE](https://stackoverflow.com/help/mcve) Sketch This code will fail on 2.4.2 and 2.5.0-beta2.

```cpp

#include <Arduino.h>

void c(int *b)
{

  String x;
  *b = 4;

}

void setup() 
{

  int *a;
  c(a);

}

void loop()
{
}

```

### Debug Messages

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

No stack dump
```

liebman, 2019-01-19:
This is not an issue with the core, you have errors in your code.  The int pointer is not initialized.

d-a-v, 2019-01-19:
You are using unititialized pointers.

closed 2019-01-19 by earlephilhower