esp8266/Arduino #5639: Is a beta3 possible at this point?

andrethomas, 2019-01-19:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:			[ESP-12]
- Core Version:			[bd11d02]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dout]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA]
- Upload Speed:   [] (serial upload only)

### Problem Description

Question - Is it possible to do a beta3 release at this junction?

I'm using the Sonoff-Tasmota sketch and there are two major problems which have been addressed since the release of beta2.

1. The negative size issues on the linker files which prevents making binaries for 4M devices which were fixed with PR https://github.com/esp8266/Arduino/pull/5522

2. PWM which appears to be fixed by PR https://github.com/esp8266/Arduino/pull/5578

We are actively encouraging the use of the 2.5.0 core for Sonoff-Tasmota builds but would like to use a released beta to be able to make it easier for users to easily test 2.5.0 in the hope that we can help in any way possible to iron out any potential bugs prior to the final release of 2.5.0. We also do pre-compiled binary releases for our development codebase and would like to use a (pre)released core to compile these as they are currently being automatically compiled using the beta2 release.

You guys do wonderful work and this is highly respected so I know it's not always possible to do what is being asked here due to time constraints but this is done in the spirit of trying to ensure that the 2.5.0 core is released with as minimal possible bugs (yes, we all want no bugs but it does not work like that in the real world :))

Thanks, much appreciated!

Jason2866, 2019-01-19:
+ 1

d-a-v, 2019-01-19:
>  in the spirit of trying to ensure that the 2.5.0 core is released with as minimal possible bugs

On behalf of maintainers, we agreed.
It will probably happen within the next few days.

domingosl, 2019-01-20:
Is there any public timeline from the maintainers?

andrethomas, 2019-01-20:
@domingosl "within the next few days"

andrethomas, 2019-01-24:
Thx @d-a-v, appreciated - Closing this one for now as to not clutter the Issues page :)

closed 2019-01-24 by andrethomas