esp8266/Arduino #5640: Mosquitto TLS

DaveCalaway, 2019-01-19:
#### Platform

- Hardware:				NodeMCU 1.0
- Core Version:			latest git hash
- Development Env:		Arduino IDE
- Operating System:		MacOS

Hi everyone,
i have a question about TLS and the MQTT.

I would like to send a message from my ESP8266 to the Mosquitto Broker using MQTT + TLS.

Using this [guide][1] i understood:  
 - how the TLS works  
 - how generate the keys needed  
 - how to set the Mosquitto for the TLS communication  

Now i search for a esp8266 example, i found the exercise with "BearSSL WiFi Classes".  

This is my code:
```
#include <ESP8266WiFi.h>
#include <PubSubClient.h> // MQTT Client

WiFiClient espClient;
PubSubClient client(espClient);

const char ca_cert[] PROGMEM = R"EOF(
-----BEGIN CERTIFICATE-----
XXXXXXX.......XXXX
-----END CERTIFICATE-----
)EOF";

void setup() {
        Serial.begin(9600);

        // WIFI connection
        setup_wifi();

        // MQTT TLS
        BearSSL::WiFiClientSecure client;
        BearSSL::X509List cert(ca_cert);
        client.setTrustAnchors(&cert);

        // MQTT
        client.setServer(mqtt_rasp, 8883);
        client.setCallback(callback);
}

void loop() {
        // Wait for MQTT client
        if (!client.connected()) {
                reconnect();
        }
        client.loop();

        if ( millis() - oldtime > refrash ) {
                oldtime = millis();
                bool msg = 1;
                // MQTT Public
                Serial.print("Publish message: ");
                client.publish("outTopic", msg); //publish (topic, payload) 
        }
}
```
The callback, setup_wifi, and the variables are setted correctly.

The error:
```
/var/folders/91/_rb11v6d67gb2dd7q5k14td80000gn/T/arduino_modified_sketch_785115/sketch_jan19d.ino: In function 'void setup()':
sketch_jan19d:175:16: error: 'class BearSSL::WiFiClientSecure' has no member named 'setServer'
         client.setServer(mqtt_rasp, 1883);
                ^
sketch_jan19d:176:16: error: 'class BearSSL::WiFiClientSecure' has no member named 'setCallback'
         client.setCallback(callback);
                ^
exit status 1
'class BearSSL::WiFiClientSecure' has no member named 'setServer'
```

Someone can help me?

  [1]: http://www.steves-internet-guide.com/mosquitto-tls/

devyte, 2019-01-22:
@DaveCalaway this is not the right place to request generic help. This is an issue tracker, meant to track issues in the core hosted in this repo.
Off the top of my head, I suggest setting up a generic TLS server  with your certificates and getting the ESP to connect to it. Then, once you fully understand how that works, continue on to MQTT. At a glance, you seem to be on the right track (you're setting up the wificlientsecure individually), but I can't tell if there's anything wrong.
As a second suggestion, you might seek help from the MQTT lib author.

Closing due to 3rd party lib.

closed 2019-01-22 by devyte

apanasara, 2019-06-17:
@DaveCalaway : setclock so that certificate can be validated