esp8266/Arduino #5644: error: 'time_t' has not been declared in HardwareSerial.h

pat1, 2019-01-20:
in the last git version using hardware serial I get at compile time:
packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/HardwareSerial.h:210:34: error: 'time_t' has not been declared
     unsigned long detectBaudrate(time_t timeoutMillis);

this little patch solve the problem:
diff --git a/cores/esp8266/HardwareSerial.h b/cores/esp8266/HardwareSerial.h
index ce9a1c5..5f083f7 100644
--- a/cores/esp8266/HardwareSerial.h
+++ b/cores/esp8266/HardwareSerial.h
@@ -30,6 +30,7 @@
 #include <inttypes.h>
 #include "Stream.h"
 #include "uart.h"
+#include "time.h"
 
 enum SerialConfig {
     SERIAL_5N1 = UART_5N1,

Jason2866, 2019-01-20:
You are using core 2.4.0. Latest git version is 2.5.0?!
`packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/HardwareSerial.h`

andrethomas, 2019-01-21:
@pat1 Core 2.4.0 issues are no longer supported. Please upgrade to 2.5.0-beta2 and if the issue persists please re-open the issue and complete all the required information from the template.

Please also close this issue as it is no longer relevant.

pat1, 2019-01-21:
if I say that it is the last git version it is.
You look at the name of directory that is not the same (I start to work with 2.4.0 version a long time ago and the dir was not changed).
[pat1@localhost 2.4.0]$ git pull
Already up-to-date.
[pat1@localhost 2.4.0]$ git branch
* master

pat1, 2019-01-21:
[pat1@localhost bin]$ ./xtensa-lx106-elf-c++ --version
xtensa-lx106-elf-c++ (crosstool-NG 1.20.0) 4.8.2
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

I get the same after cd tools; python ./get.py

The problem is not the toolchain; the use of time_t is possible only after the appropriate #include
In the most of the use cases you use the serial with other library that include time.h
To reproduce you have to link a library  that use only one include
#include <HardwareSerial.h>
that library cannot compile.

d-a-v, 2019-01-21:
> In the most of the use cases you use the serial with other library that include time.h
To reproduce you have to link a library that use only one include
#include <HardwareSerial.h>
that library cannot compile.

Ah, *this* is useful info to help us understand your issue,
thanks.

PR #5648 (2019-01-21): hardwareSerial: add missing include
fix #5644
closed 2019-01-21 by earlephilhower