esp8266/Arduino #5661: SPIFFS corrupting files

gitandlucsil, 2019-01-24:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP-01
- Core Version:			2.4.2
- Development Env:		Arduino IDE
- Operating System:		Windows 10

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio
- Flash Size: 		        1MB
- lwip Variant:                  v2 Lower Memory
- Reset Method:		ck
- Flash Frequency:	        80Mhz
- CPU Frequency:		160MHz
- Upload Using:		SERIAL
- Upload Speed:              115200

### Problem Description

I have a trouble when a try to read a .csv file from SPIFFS filesystem. The reads works in most times, but after any minutes, when I try to read, the file is corrupted and the MESSAGE  that show in serial is:

Panic C:\Users\Andre\Documents\arduino-1.8.6\portable\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src\include/DataSource.h:93 const uint8_t* BufferedStreamDataSource<TStream>::get_buffer(size_t) [with TStream = Stream; uint8_t = unsigned char; size_t = unsigned int]: Assertion 'cb == stream_rem' failed.
11:10:29.214 -> 
11:10:29.214 -> ctx: cont 
11:10:29.214 -> sp: 3ffffbe0 end: 3fffffd0 offset: 01b0
11:10:29.214 -> 
11:10:29.214 -> >>>stack>>>
11:10:29.214 -> 3ffffd90:  3fff10c4 00000020 3ffefd60 40202acd  
11:10:29.214 -> 3ffffda0:  3fff0254 00000000 3fff108c 4020a081  
11:10:29.214 -> 3ffffdb0:  3ffef1a4 00000356 00000356 00000020  
11:10:29.214 -> 3ffffdc0:  00000020 00000000 3fff0c2c 4020e489  
11:10:29.214 -> 3ffffdd0:  00000020 00000000 00000020 4020f3f8  
11:10:29.214 -> 3ffffde0:  00000000 4bc6a7f0 00000020 4020cde8  
11:10:29.260 -> 3ffffdf0:  00000000 00000000 3fff0c2c 40209d06  
11:10:29.260 -> 3ffffe00:  3fff0254 3ffeef20 40106644 00000000  
11:10:29.260 -> 3ffffe10:  00000020 00000020 3fff108c 40209f8d  
11:10:29.260 -> 3ffffe20:  0446b616 3ffeef20 3ffffea4 00000020  
11:10:29.260 -> 3ffffe30:  3fff108c 3ffffe80 0000001c 00000020  
11:10:29.260 -> 3ffffe40:  3fff108c 3ffffe80 3fff108c 4020a048  
11:10:29.260 -> 3ffffe50:  00000020 3ffffe80 00000000 4020a3d0  
11:10:29.307 -> 3ffffe60:  00000000 3ffeef20 3ffe8924 4020b3f8  
11:10:29.307 -> 3ffffe70:  00000020 3ffeef20 3ffe8924 40207a14  
11:10:29.307 -> 3ffffe80:  4020f7c0 00000000 000003e8 3ffef1a4  
11:10:29.307 -> 3ffffe90:  3fff0f2c 3fff1054 00000000 00000000  
11:10:29.307 -> 3ffffea0:  00000000 3fff106c 0000000f 0000000a  
11:10:29.307 -> 3ffffeb0:  3fff1014 0000000f 00000008 4010020c  
11:10:29.307 -> 3ffffec0:  00000001 00000000 3ffeef19 40207a9d  
11:10:29.354 -> 3ffffed0:  00000001 00000001 3fff0a64 4020f022  
11:10:29.354 -> 3ffffee0:  00000000 00000000 3fff0a64 4020b3ee  
11:10:29.354 -> 3ffffef0:  3fff0a64 3ffeef64 3fffff20 4020b42a  
11:10:29.354 -> 3fffff00:  00000000 00000000 00000000 4020d620  
11:10:29.354 -> 3fffff10:  3fff0a64 3ffeef64 3ffeef20 4020b4b1  
11:10:29.354 -> 3fffff20:  3fff0dec 0000000f 00000008 00000000  
11:10:29.354 -> 3fffff30:  00000000 3ffe8a30 40106644 0003b2bc  
11:10:29.401 -> 3fffff40:  3ffeef64 00000001 3ffe856c 3ffeef48  
11:10:29.401 -> 3fffff50:  00000001 00000000 4020a878 0000000f  
11:10:29.401 -> 3fffff60:  00000000 3fff108c 3ffeef20 3ffef124  
11:10:29.401 -> 3fffff70:  00000001 3ffeef48 3ffeef20 4020b700  
11:10:29.401 -> 3fffff80:  401066a8 3ffef064 00000000 4020cf91  
11:10:29.401 -> 3fffff90:  00000000 3fff108c 3ffef064 4020cfd0  
11:10:29.401 -> 3fffffa0:  3fffdad0 00000000 3ffef11c 402091b0  
11:10:29.448 -> 3fffffb0:  3fffdad0 00000000 3ffef11c 4020e048  
11:10:29.448 -> 3fffffc0:  feefeffe feefeffe 3ffe856c 40100739  
11:10:29.448 -> <<<stack<<<
11:10:29.448 -> 
11:10:29.448 ->  ets Jan  8 2013,rst cause:1, boot mode:(3,2)
11:10:29.448 -> 
11:10:29.448 -> load 0x4010f000, len 1384, room 16 
11:10:29.495 -> tail 8
11:10:29.495 -> chksum 0x2d
11:10:29.495 -> csum 0x2d
11:10:29.495 -> vbb28d4a3
11:10:29.495 -> ~ld

My read file code:

void handleLeituraH2O()
{
  if(writing_values == false){
      reading_values = true;  //Marca a flag que esta lendo um arquivo, assim impedindo uma nova escrita
      handleCSV();            //Chama a funcao que trata os arquivos csv armazenados na SPIFFS do ESP
      reading_values = false; //Desmarca a flag que esta lendo um arquivo, assim possibilitando uma nova escrita
  }
}
```
In main file, I have:

void setup()
{
  Serial.begin(115200);                                             //Inicia comunicacao Serial
  Serial.print("Iniciou!");
  EEPROM.begin(SIZE_EEPROM);                                        //Inicializa memoria EEPROM
  readEEPROM();                                                     //Le as informacoes salvas na memoria EEPROM
  wdt_enable(WDTO_8S);                                              //Seta o WDT para 8 segundos
  SPIFFS.begin();                                                   //Inicializa o sistema de arquivos do SPIFFS
  File fileh2o;                                                     //Variavel temporaria para manipular o arquivo da H2O
  File filetemp;                                                    //Variavel temporaria para manipular o arquivo da Temperatura
  File filesilo;                                                    //Variavel temporaria para manipular o arquivo do Silo
  //Verifica se os arquivos existem. Caso nao existam, cria
  if (SPIFFS.exists(LOG_H2O_FILE)) {
    fileh2o = SPIFFS.open(LOG_H2O_FILE, "r");
  } else {
    fileh2o = SPIFFS.open(LOG_H2O_FILE, "w+");
  }
  fileh2o.close();
  if (SPIFFS.exists(LOG_TEMP_FILE)) {
    filetemp = SPIFFS.open(LOG_TEMP_FILE, "r");
  } else {
    filetemp = SPIFFS.open(LOG_TEMP_FILE, "w+");
  }
  filetemp.close();
  if (SPIFFS.exists(LOG_SILO_FILE)) {
    filesilo = SPIFFS.open(LOG_SILO_FILE, "r");
  } else {
    filesilo = SPIFFS.open(LOG_SILO_FILE, "w+");
  }
  filesilo.close();
  WiFi.mode(WIFI_AP);                                              //Inicializa o modulo WiFi no modo de Access Point (ou seja, outros dispositivos se conectam a ele para usar o servidor web)
  WiFi.softAP(esp_values.ssid);                                    //Seta o ssid do modulo
  Serial.print("Ponto de acesso: ");
  Serial.print(esp_values.ssid);
  Serial.println(" iniciado");
  Serial.print("endereco IP: ");
  Serial.println(WiFi.softAPIP());
  //Declara os metodos HTTP
  server.on("/ssid", HTTP_POST, handleConfigSsid);
  server.on("/version_unium", HTTP_GET, handleLeituraVersaoUnium);
  server.on("/version_wificom", HTTP_GET, handleLeituraVersaoWificom);
  server.on("/readbalanca", HTTP_GET, handleLeituraBalanca);
  server.on("/h2o.csv", HTTP_GET, handleLeituraH2O);
  server.on("/temp.csv", HTTP_GET, handleLeituraTemperatura);
  server.on("/silo.csv", HTTP_GET, handleLeituraSilo);
  server.on("/", HTTP_GET, handleRoot);
  server.onNotFound(handleNotFound);
  server.begin();                                               //Inicializa o servidor
}
/*-------------------------------------Funcao loop--------------------------------------------------------------*/
void loop()
{
  server.handleClient();                                       //Trata requisicoes e conexoes dos clientes
  if (reading_values == false) {
    listen_serial();                                             //Fica em um loop ouvindo a serial, para receber comandos de leitura nos arquivos
  }
  Serial.flush();
}

devyte, 2019-01-26:
The requirement is a MCVE sketch: Minimial Complete Verifiable Example. Your code is neither minimal nor complete.
Please edit your post and provide a minimal and complete sketch that can be used to reproduce your issue.. Please do not post your entire project.

5chufti, 2019-01-26:
you are using a ESP-01, there are problems with ESP-01 with puya flash.
As you are using 2.4.2 AFAIK a/the needed patch may not be applied.
see these issues https://github.com/esp8266/Arduino/issues?utf8=%E2%9C%93&q=is%3Aissue+puya+spiffs+

gyromike, 2019-02-15:
I am having a similar issue. Here is the very simple code.  I write a zero and read a two.
`
#define DATASTART   0x0           // start address of data buffer

bool openspiffs() {
  rowIndex = 0;
  datafile = SPIFFS.open(carddatafile, "r+b");        // exsisting file opened
  if (!datafile) {
    datafile = SPIFFS.open(carddatafile, "w+b");  // create new file
  }
    write16data(0, rowIndex);       //save the card count at offset 0
  return datafile;
}

void closespiffs() {

  if (datafile) {
    datafile.close();
  }
}

uint16_t read16data(uint16_t whatbyte) {
  uint16_t out;
  uint8_t high;
  uint8_t low;

  datafile.seek(DATASTART + whatbyte, SeekSet);
  datafile.read(&high, 1);
  datafile.read(&low, 1);

  out = high << 8;
  out += low;
  return out;
}

void write16data(uint16_t whatbyte, uint16_t databuffer) {
  uint8_t high;
  uint8_t low;
  high = highByte(databuffer);
  low = lowByte(databuffer);
  datafile.seek(DATASTART + whatbyte, SeekSet);
  datafile.write(&high, 1);
  datafile.write(&low, 1);
  datafile.flush();
}`

earlephilhower, 2019-07-24:
Closing this due to lack of MCVE or feedback.  The comment about PUYA flash and needing a) a newer version (2.5.2 or later I think) of the core and b) a define to enable the PUYA broken-flash-workaround for anyone hitting something like this in the future.

closed 2019-07-24 by earlephilhower