esp8266/Arduino #5678: 2.5.0-beta3's libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp doesn't compile

crankyoldgit, 2019-01-27:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[other] / N/A
- Core Version:			2.5.0-beta3 / 21db8fc9285d2f548c0123206d53cab059cf3b3d
- Development Env:		Arduino IDE 1,8.8
- Operating System:		Unix / Travis

### Settings in IDE

- Module:  			Wemos D1 mini r2 & Nodemcu
- Flash Mode:			N/A
- Flash Size: 		4M
- lwip Variant:   N/A
- Reset Method:		N/A
- Flash Frequency:	N/A
- CPU Frequency:		80Mhz
- Upload Using:		N/A
- Upload Speed:   N/A

### Problem Description
2.5.0-beta3's libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp doesn't compile, has errors.

See https://travis-ci.org/markszabo/IRremoteESP8266/jobs/484962845#L1515
```
$ arduino --verify --board esp8266:esp8266:nodemcuv2:xtal=80,eesz=4M3M $PWD/examples/IRGCTCPServer/IRGCTCPServer.ino
/home/travis/.arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta3/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp: In member function 'bool ESP8266WiFiSTAClass::hostname(const char*)':
/home/travis/.arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta3/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp:537:52: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
         intf->hostname = wifi_station_get_hostname();
                                                    ^
/home/travis/.arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta3/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp:539:33: error: 'netif_dhcp_data' was not declared in this scope
         if (netif_dhcp_data(intf) != nullptr) {
                                 ^
exit status 1
```

This looks like it was introduced in #5652 (https://github.com/esp8266/Arduino/commit/7c5be91430a2fe47334e2d928a5d8a35f2683cd7)

The code has compiled fine with all previous versions of the ESP8266 core libraries for years. e.g. 2.5.0-beta2 was fine. Beta3 broke it.

### Sketch

https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRGCTCPServer/IRGCTCPServer.ino
https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino
https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRServer/IRServer.ino

### Debug Messages

N/A

d-a-v, 2019-01-27:
> * lwip Variant: N/A

Are you using lwIP-v1.4 ?

crankyoldgit, 2019-01-27:
No idea tbh. It's using what ever the default would be.
Our travis environment is built using:

  - wget http://downloads.arduino.cc/arduino-1.8.8-linux64.tar.xz
  - tar xf arduino-1.8.8-linux64.tar.xz
  - sudo mv arduino-1.8.8 /usr/local/share/arduino
  - sudo ln -s /usr/local/share/arduino/arduino /usr/local/bin/arduino
  - wget https://raw.githubusercontent.com/google/styleguide/gh-pages/cpplint/cpplint.py
  - ln -s $PWD /usr/local/share/arduino/libraries/
  - git clone https://github.com/tzapu/WiFiManager.git /usr/local/share/arduino/libraries/WiFiManager
  - git clone https://github.com/knolleary/pubsubclient.git /usr/local/share/arduino/libraries/PubSubClient
  - arduino --pref "boardsmanager.additional.urls=http://arduino.esp8266.com/stable/package_esp8266com_index.json" --save-prefs
  - arduino --install-boards esp8266:esp8266
  - arduino --board D=esp8266:esp8266:nodemcuv2:xtal=80,eesz=4M3M --save-prefs
  - arduino --pref "compiler.warning_level=all" --save-prefs

(https://github.com/markszabo/IRremoteESP8266/blob/master/.travis.yml)

d-a-v, 2019-01-27:
This bug will be fixed asap.

> `arduino --board D=esp8266:esp8266:nodemcuv2:xtal=80,eesz=4M3M --save-prefs`

You may update your board FQBN with `ip=lm2f:exception=disabled`:

`arduino --board D=esp8266:esp8266:nodemcuv2:xtal=80,eesz=4M3M:ip=lm2f:exception=disabled --save-prefs`

crankyoldgit, 2019-01-27:
Thanks. I'll try it out now.

crankyoldgit, 2019-01-27:
> You may update your board FQBN with ip=lm2f:exception=disabled:
>
> `arduino --board D=esp8266:esp8266:nodemcuv2:xtal=80,eesz=4M3M:ip=lm2f:exception=disabled --save-prefs`

I think you meant: `ip=lm2f,exception=disabled,`

`arduino --board D=esp8266:esp8266:nodemcuv2:xtal=80,eesz=4M3M,ip=lm2f,exception=disabled --save-prefs`

Which seems to be working/compiling so far.

What does that addition actually do?

d-a-v, 2019-01-27:
> What does that addition actually do?

I am curious to know wether #5657 would answer to your question.

crankyoldgit, 2019-01-28:
Not really. e.g. It doesn't explain what disabling `exceptions` does and has nothing about the `ip=` clause.

d-a-v, 2019-01-28:
I'm trying to make it clear to read. There is an update in the above PR.

To answer your question, 
There are two available versions of the IP stack (lwIP), you are using the former which is broken in 2.5.0-beta3 (soon to be fixed). The second one is the default, and that's what `ip=lm2f` is: "use the default"

Because you were not specifying them in your scripts and default options in IDE are not used by default with the arduino builder (at least in your case), `exception=disabled` is another case for issues about flash occupation. It's about enabling C++ exceptions.

Anyway if it is fixed for you, you can close the issue.

ChrSchultz, 2019-02-01:
I have a similar roblem with compilattion 
arduino 1.8.8  windows-store 
win10-64bit TinyGSM, esp8622-2.5.0-beta3,
Wemos D1 mini
[test.ino.txt](https://github.com/esp8266/Arduino/files/2821433/test.ino.txt)
if I try to compile the file above( rename to ,ino) I got the following errors...

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware -hardware D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\libraries -libraries D:\Benutzer\ic-sc\Dokumente\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -ide-version=10808 -build-path C:\Users\ic-sc\AppData\Local\Temp\arduino_build_470391 -warnings=none -build-cache C:\Users\ic-sc\AppData\Local\Temp\arduino_cache_828420 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.esptool.path=D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9 -prefs=runtime.tools.esptool-2.5.0-3-20ed2b9.path=D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -verbose C:\Users\ic-sc\AppData\Local\Temp\arduino_modified_sketch_875821\HttpClient.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware -hardware D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\libraries -libraries D:\Benutzer\ic-sc\Dokumente\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -ide-version=10808 -build-path C:\Users\ic-sc\AppData\Local\Temp\arduino_build_470391 -warnings=none -build-cache C:\Users\ic-sc\AppData\Local\Temp\arduino_cache_828420 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.esptool.path=D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9 -prefs=runtime.tools.esptool-2.5.0-3-20ed2b9.path=D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -verbose C:\Users\ic-sc\AppData\Local\Temp\arduino_modified_sketch_875821\HttpClient.ino
Using board 'd1_mini' from platform in folder: D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta3
Using core 'esp8266' from platform in folder: D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta3
Detecting libraries used...
"D:\\Benutzer\\ic-sc\\Dokumente\\ArduinoData\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-ID:\\Benutzer\\ic-sc\\Dokumente\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.0-beta3/tools/sdk/include" "-ID:\\Benutzer\\ic-sc\\Dokumente\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.0-beta3/tools/sdk/lwip2/include" "-ID:\\Benutzer\\ic-sc\\Dokumente\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.0-beta3/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\ic-sc\\AppData\\Local\\Temp\\arduino_build_470391/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10808 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO  [... cut ...]
Mehrere Bibliotheken wurden für "TinyGsmClient.h" gefunden
 Benutzt: D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\TinyGSM
 Bibliothek TinyGSM in Version 0.3.5 im Ordner: D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\TinyGSM  wird verwendet
Bibliothek ArduinoHttpClient in Version 0.3.1 im Ordner: D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\ArduinoHttpClient  wird verwendet
exit status 1
cannot declare variable 'client' to be of abstract type 'TinyGsmSim800::GsmClient'

If I compile that with the latest stable version the compilation will work fine.

d-a-v, 2019-02-01:
@ChrSchultz 
* it is unrelated
* it looks like an installation problem.
  For example this type `TimeElements tm;` does not exist in this core,
  your environment at least doesn't find the external libraries you are using.

The OP is fixed, we are waiting for feedback.
Please try with a clean installation and if the problem persists, open a **new** issue and fill the issue template.

crankyoldgit, 2019-02-02:
@d-a-v I'll close this for now as the work-around fixes it. I'm assuming #5682 is the fix. I'll re-open/comment etc here if the next beta/final release doesn't resolve it. I won't be backing out the recommended work-around till then.

closed 2019-02-02 by crankyoldgit