esp8266/Arduino #5680: MQTT TLS Verify Fingerprint Fail with Core 2.5.0

majherek, 2019-01-27:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-01]
- Core Version:			[7b36071]
- Development Env:		[MVSC with Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Using the same configuration in my_user_config.h Tasmota 6.4.1.9 on Core 2.4.2 can verify MQTT TLS Cert and work well. The same tasmota compiled with Core 2.5.0 (Stage) can't verify cert.

It is weird, because in core 2.5.0 much more RAM is available compared to 2.4.2 (taken from Tasmota WIKI) - so I think IT IS NOT RAM ISSUE!!!

I create an issue [#4952](https://github.com/arendst/Sonoff-Tasmota/issues/4952) on Tastmota GitHUB.  The TLS verification works since Core 2.3.0. Now I can't use it. 

**To Reproduce**
Compilte Tasmota with MQTT TLS with commented out almost everything in config (I use only BMP and DS18b20). I can provide my_user_config.h.
Core 2.4.2:
```
0:00:00 Project sonoff Sonoff 12 Version 6.4.1.9(sonoff)-2_4_2
00:00:00 WIF: Connecting to AP1 atomix in mode 11N as sonoff-0476...
00:00:04 WIF: Connected
20:32:07 MQT: Attempting connection...
20:32:07 MQT: Verify TLS fingerprint...
20:32:07 MQT: Verified using FingerprintCA
20:32:07 MQT: Connected
20:32:07 MQT: tele/sonoff/LWT = Online (retained)
20:32:08 MQT: cmnd/sonoff/POWER =
20:32:08 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"6.4.1.9(sonoff)","FallbackTopic":"cmnd/sonoff12_fb/","GroupTopic":"sonoffs"}
20:32:08 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
20:32:08 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
20:32:08 MQT: stat/sonoff/POWER = OFF
20:32:16 MQT: tele/sonoff/STATE = {"Time":"2019-01-16T20:32:16","Uptime":"0T00:00:15","Vcc":3.538,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"atomix","BSSId":"06:8D:DB:DB:97:47","Channel":11,"RSSI":92}}
20:32:24 CMD: mqttfingerprint1
20:32:24 MQT: stat/sonoff/RESULT = {"MqttFingerprint1":"69 93 F5 A6 EE A5 AD DA 2D DC 3C E7 DE 72 AB 3B DC CA 81 4D"}

```

Core 2.5.0
```
00:00:00 Project sonoff Sonoff 12 Version 6.4.1.9(sonoff)-STAGE
00:00:00 WIF: Connecting to AP1 atomix in mode 11N as sonoff-0476...
00:00:04 WIF: Connected
21:09:31 MQT: Attempting connection...
21:09:31 MQT: Verify TLS fingerprint...
21:09:31 MQT: Failed
21:09:42 MQT: Attempting connection...
21:09:42 MQT: Verify TLS fingerprint...
21:09:42 MQT: Failed
[... 111 lines cut ...]
; VTABLES in Flash (default)
                            -DVTABLES_IN_FLASH
; VTABLES in Heap
;                            -DVTABLES_IN_DRAM
; VTABLES in IRAM
;                            -DVTABLES_IN_IRAM
; enable one option set -> No exception recommended
; No exception code in firmware
                            -fno-exceptions
                            -lstdc++
; Exception code in firmware /needs much space! 90k
;                           -fexceptions
;                           -lstdc++-exc

[core_active]
; Select one core set for platform and build_flags
;platform                  = ${core_2_3_0.platform}
;build_flags               = ${core_2_3_0.build_flags}
;platform                  = ${core_2_4_2.platform}
;build_flags               = ${core_2_4_2.build_flags}
;platform                  = ${core_2_5_0.platform}
;build_flags               = ${core_2_5_0.build_flags}
platform                  = ${core_stage.platform}
build_flags               = ${core_stage.build_flags}
```

majherek, 2019-01-27:
On newest Core I have:
```
--- Miniterm on COM7  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
11:50:42 MQT: Attempting connection...
11:50:42 MQT: Verify TLS fingerprint...

Exception (2):
epc1=0x3fffefb0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fffefb0 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffed9c end: 3fffffb0 offset: 01a0
3fffef3c:  3ffe8674 00000000 3fffdad0 3fff1d30
3fffef4c:  4021afe3 3ffe9e60 40000f49 3fffdab0
3fffef5c:  40000f49 40000e19 40001878 00000001
3fffef6c:  00000000 3fffff10 aa55aa55 000000ed
3fffef7c:  401054c0 401054c6 00000001 23002200
3fffef8c:  4000050c 3fffc278 40103074 3fffc200
3fffef9c:  00000022 3fffefb0 3fffef3c 401017a9
3fffefac:  3ffff808 4022c9c6 00000030 00000016
3fffefbc:  ffffffff 4022c5a4 24e278ea 20a0682a
3fffefcc:  0e627621 7281f549 5a6e8078 30ace92e
3fffefdc:  0cf21ae0 3ffff174 40268648 40268688
3fffefec:  1efd6c4a ccf075c1 3bf30b52 e7c374db
3fffeffc:  00000030 a2bcb357 1cf1a8a7 9ad500bd
3ffff00c:  ffca459a ecd83c41 b4cc4d2d c74f755a
3ffff01c:  38897d89 5c5c5c5c 5c5c5c5c 5c5c5c5c
3ffff02c:  5c5c5c5c 5c5c5c5c 5c5c5c5c 5c5c5c5c
3ffff03c:  5c5c5c5c cc59d3de 47412458 86396ab6
3ffff04c:  a43069bd 401c322a bb1ad59f 0ceeb41f
3ffff05c:  d95a61e6 d451011f 03b3cc54 4549ef1f
3ffff06c:  e03d2006 86186e95 41aad7f0 1494d83a
3ffff07c:  32a7da9a b89dc533 ca050875 36291bd0
3ffff08c:  9f58e378 cdd9238f d0280128 43fc4681
3ffff09c:  f7e3165f a53a9f22 719b655b 5ac58183
3ffff0ac:  ac9b5045 0c709892 059000e5 9b847819
3ffff0bc:  0e1ecd20 bf4efe99 b3c85c59 6c4742f1
3ffff0cc:  4e1d2bb3 80000000 00000000 00000000
3ffff0dc:  00000000 00000000 00000000 00000000
3ffff0ec:  00000300 d6f0fa0b 45b1e1f7 f08d8d6a
3ffff0fc:  76242133 e87bfb8e 8856f701 14874a63
3ffff10c:  db51265a 122aaaf8 a85e6ff5 764ba32d
3ffff11c:  00000080 03030000 010000db 82519f72
3ffff12c:  0000377b 8ee65349 0de9f8e2 5c843fb7
3ffff13c:  48f1b679 70bd0000 53d2e7c9 a8c02bc0
3ffff14c:  5acca9cc 30c0acc0 2fc02cc0 afc023c0
3ffff15c:  adc0aec0 28c009c0 27c024c0 14c02dc0
3ffff16c:  13c00ac0 32c025c0 31c02ec0 2ac004c0
3ffff17c:  29c026c0 0f009c00 0ec005c0 9dc0a0c0
3ffff18c:  9dc09cc0 3d002f00 a1003c00 12c003c0
3ffff19c:  35c008c0 54e6e9f1 96261c51 ad44e4d9
3ffff1ac:  b0526d4f 013cea5b b9a231c8 688fe3ad
3ffff1bc:  9d97d6c8 7ff197af 31d8f759 dcf5a248
3ffff1cc:  e61a968c de2956dd b98b8b40 decfb1fb
3ffff1dc:  9e74e01e a3bc4da2 86489aa9 8039d56b
3ffff1ec:  adf63453 98a8f42f 5ddc0082 7b872822
3ffff1fc:  bab770fb 4a71a188 75334bbf 7a794253
3ffff20c:  db045394 2cfa466a b6b2ff0b 185c29fc
3ffff21c:  eab6fe03 eff73e70 44967627 afcb6566
3ffff22c:  676d233e 188546f4 9936d658 27fe3203
3ffff23c:  a0e196d1 740dcb73 aa11aaab 021b0571
3ffff24c:  60ac33a4 7b750b39 c526dd26 00a69a4f
3ffff25c:  2806ed3d 2205965e 35080941 914e876e
3ffff26c:  5c140648 8682c483 97b1a5cd 7d734901
3ffff27c:  480db589 04f471c6 7d123eaf f9d5260b
3ffff28c:  f91b8cdb 930d90a7 ce7c026c ab0262d1
3ffff29c:  e23f009d 3fc8d617 b896be0d e96600fe
3ffff2ac:  1679be0b 9407df68 906bd338 c7e27a8c
[... 536 lines cut ...]
3fffff7c:  3fff1d30 3fffdad0 3fff0b34 0001a3cf
3fffff8c:  40213fbe 00000000 00000000 00000001
3fffff9c:  3fff1d30 3fffdad0 00000000 3fff1d00
3fffffac:  4021b0d4 feefeffe feefeffe 3ffe8674
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

00:00:00 Project sonoff Sonoff 12 Version 6.4.1.9(sonoff)-STAGE
00:00:00 WIF: Connecting to AP1 atomix in mode 11N as sonoff-0476...
00:00:04 WIF: Connected
11:52:52 MQT: Attempting connection...
11:52:52 MQT: Verify TLS fingerprint...
11:52:52 MQT: Failed
11:53:03 MQT: Attempting connection...
11:53:03 MQT: Verify TLS fingerprint...
```

earlephilhower, 2019-01-31:
The maintainers here don't have any control over the Tasmota or Sonoff, you've not provided a MCVE, and you did not provide the requested decoded stack dumps, so there's nothing we can do with this.  I suggest following up with the Tasmota team (who also seem to have closed the issue).

Closing.

closed 2019-01-31 by earlephilhower