esp8266/Arduino #5681: DHCP NTP Server configurable

Meinsda, 2019-01-27:
Hello

Is it possible that you can make an option at the init of the ntp client that you can say your NTP server you get from the DHCP, manualy or both? At the moment, i have troubles with mirkotik router as DHCP that i can't change the NTP Server in Tasmota because the client alwys use the NTP from the DHCP Server.
If i don't set a ntp Server at the dhcp he also use a adress of the dhcp.

PEter

whyameye, 2019-04-04:
Would this be solved by sending `DHCP_OPTION_NTP` during the DHCP handshake w/ the server then parsing the result? Would that all be done in `https://github.com/esp8266/Arduino/blob/master/tools/sdk/lwip/src/core/dhcp.c`?

In terms of implementation, it looks like lines 540-548 of dhcp.c check whether an option was returned and get the value that was returned. 918-930 appear to show how to set what options we are going to request. Lines 1441-1493 parse the returned options. Does this all sound like the way to implement this?

d-a-v, 2019-04-04:
Not in `tools/sdk/lwip` that is lwIP-1.4RC2. We are now using lwIP-2.1.2 (`tools/sdk/lwip2`).

This functionality is a [harcoded option](https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/a501b5722bd050f41a9351e69ca75937150ec138/glue-lwip/arduino/lwipopts.h#L943) in our configuration.

Do you need to disable the feature or being able to add other servers ?

`dhcp_set_ntp_servers()` is used as follow: [link](http://git.savannah.nongnu.org/cgit/lwip.git/tree/src/core/ipv4/dhcp.c?h=STABLE-2_1_2_RELEASE#n663)

Since `LWIP_DHCP_MAX_NTP_SERVERS` is 1 and `SNTP_MAX_SERVERS` is 3, there is room for two others.

whyameye, 2019-04-04:
So you are saying there is already a way to get local ntp servers from the dhcp server using `dhcp_set_ntp_servers()`? From the ntp examples I had seen I thought it was only possible to use a server like `pool.ntp.org` and not a local ntp server.

d-a-v, 2019-04-04:
The first NTP server given by a properly configured DHCP servers is automatically used in the ESP with lwIP-v2.
OP wants to disable that if I understand it and provide an own NTP server.

whyameye, 2019-04-04:
I don't mean to hijack the thread here but where is there an example of that? The only ntp example I am familiar with is `BearSSL_Validation` which hardcodes `pool.ntp.org` as its ntp server. Also, I've always pulled from github so how does `git.savannah.nongnu.org` relate to that?

d-a-v, 2019-04-04:
> so how does git.savannah.nongnu.org relate to that?

lwip2 is a submodule of our repo. It is an adaptation layer for upstream lwIP.
upstream lwIP is on git.savannah.nongnu.org.

The source code of `tools/sdk/liblwip2*.a` is coming right from that tree.

If you run with git and linux, just try
```
cd tools/sdk/lwip2
make
```
Source code will be downloaded from there and compiled.

> The only ntp example I am familiar 

If your DHCP server has NTP server settings properly configured, and if your sketch uses DHCP, and if you select lwIP-v2 (default selection anyway), then your ESP is synchronized with NTP.

whyameye, 2019-05-14:
> The first NTP server given by a properly configured DHCP server is automatically used in the ESP with lwIP-v2.

> If your DHCP server has NTP server settings properly configured, and if your sketch uses DHCP, and if you select lwIP-v2 (default selection anyway), then your ESP is synchronized with NTP.

Can you show me an example of this? The only example I see is `libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino` and in that example `pool.ntp.org` is used to set the time, not the NTP server specified by the DHCP server.

whyameye, 2019-05-14:
> The first NTP server given by a properly configured DHCP server is automatically used in the ESP with lwIP-v2.

> If your DHCP server has NTP server settings properly configured, and if your sketch uses DHCP, and if you select lwIP-v2 (default selection anyway), then your ESP is synchronized with NTP.

I don't see how to do this and I don't see an example of this. The only example I can find is `libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino` which uses `pool.ntp.org` to set the time, not the NTP server configured by the DHCP server.

whyameye, 2019-05-15:
Sorry for the noise, since I now figured it out. It takes a second or 2 after the IP address is assigned by DHCP for the time to get updated. It just becomes unnecessary for a line like `configTime(0 * 3600, 0, "pool.ntp.org", "time.nist.gov");` when the DHCP server is configured correctly to return the NTP server with option 42 when using LwIP-v2

mcspr, 2019-09-04:
> If your DHCP server has NTP server settings properly configured, and if your sketch uses DHCP, and if you select lwIP-v2 (default selection anyway), then your ESP is synchronized with NTP.

Perhaps, example should take note of that? We can access ntp server list via sntp_getserver(idx) on setup()
https://www.nongnu.org/lwip/2_1_x/group__sntp.html#ga2a28523cb9f2b5b025a4818bc2c1afc1
Or it would make sense to check that via configTime internally / make configTime() without NTP server settings?

Secondly, I do see that this behaviour can be disabled:
https://github.com/d-a-v/lwIP/blob/2ff0db9a9b047d1c94ddbeea010561d1b9032101/src/apps/sntp/sntp.c#L759
Documentation does not list this method, but it is surely public.

d-a-v, 2019-09-04:
> example should take note of that?

The NTP-TZ-DST example explains NTP server is coming from DHCP.
#6373 has an [update](https://github.com/esp8266/Arduino/blob/f0ebbe6b8024ba7a756362e62e5f8a1aa741f26e/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino) for this example.

> Would make sense to check that via configTime internally / make configTime() without NTP server settings?

I'm not sure changing default behavior is safe.
What can be done is another example with
```
#include <lwip/apps/sntp.h>
    ...
    // in this order
    sntp_servermode_dhcp(0);
    WiFi.begin();
    configTime(TZ_SEC, DST_SEC, "pool.ntp.org");
    ...
```
(to be tested and proposed as a PR)

mcspr, 2019-09-05:
configTime can be before WiFi.begin(), though, but any order makes sntp respect configTime setting. With dnsmasq configuration containing `dhcp_option=42,<IP>` it uses the server given by the method instead of DHCP.

But I was wondering if it is ok to give out lwip api instead of something Core-specific
Maybe `configTime(TZ, DST, "server")` server1 argument should also be optional, like the server2 and 3, but it would set servermode to 0 if at least one is set?

> The NTP-TZ-DST example explains NTP server is coming from DHCP.

Yes, and lwip initializes sntp automatically, even without configTime present. But that's another issue altogether with how sntp app works

PR #6373 (2019-09-29): time: import IANA timezone definitions, expose SNTP API
- `configTime("timezone", "ntp servers...")` added
- timezone definitions by country/cities (TZ.h)
- script to update timezone definitions
- updated example
- sntp additions (see below)

Closes #4637
Closes #5681
Fixes #6562
closed 2019-09-29 by devyte