esp8266/Arduino #5685: Connecting to AWS IoT using certificate via Arduino-MQTT/PubSubClient

debsahu, 2019-01-28:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[2.5.0-beta3/https://github.com/platformio/platform-espressif8266.git#feature/stage]
- Development Env:		[Arduino IDE/Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Goal: Trying to connect to AWS on ESP8266

It doesn't matter if I use [PubSubClient](https://github.com/debsahu/ESP-MQTT-AWS-IoT-Core/tree/master/Arduino/PubSubClient) or [Arduino MQTT](https://github.com/debsahu/ESP-MQTT-AWS-IoT-Core/tree/master/Arduino/MQTT) **reconnecting** to MQTT crashes ESP8266. Just to be clear, initial connect works flawlessly. Same code is written for ESP32 as well, and no issues on AWS side or the code. To circumvent that, I reboot the ESP8266 and it re-connects flawlessly. I have tried using pointers and reinitializing before a reconnect, does not help. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

Code: https://github.com/debsahu/ESP-MQTT-AWS-IoT-Core/tree/master/Arduino

### Debug Messages

```

SDK:3.0.0-dev(c0f7b44)/Core:2.5.0-beta3=20499903/lwIP:STABLE-2_1_2_RELEASE/glue:1.0-11-g87c709d/BearSSL:6778687

bcn 0
del if1
usl
mode : sta(2c:3a:e8:4c:1d:33)
add if0
Attempting to connect to SSID: XXXXXXX...scandone
no XXXXXX found, reconnect after 1s
reconnect
...scandone
no XXXXXX found, reconnect after 1s
reconnect
...scandone
no XXXXXX found, reconnect after 1s
reconnect
...scandone
no XXXXXX found, reconnect after 1s
reconnect
...scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with XXXXXX, channel 1
[... 154 lines cut ...]
0x4020830d: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at C:\Users\YYYYY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 497
0x40223658: br_ssl_hs_client_run at src/ssl/ssl_hs_client.c line 958
0x40223600: br_ssl_hs_client_init_main at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/sys/pgmspace.h line 67
0x4022310e: br_ssl_engine_hs_reset at src/ssl/ssl_engine.c line 1305
0x4020851c: BearSSL::WiFiClientSecure::_wait_for_handshake() at C:\Users\YYYYY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 521
0x402086af: BearSSL::WiFiClientSecure::_connectSSL(char const*) at C:\Users\YYYYY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 930
0x40209604: esp_yield() at C:\Users\YYYYY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266\core_esp8266_main.cpp line 91
0x402023d7: delay at C:\Users\YYYYY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266\core_esp8266_wiring.c line 51
0x4020731a: WiFiClient::connect(IPAddress const&, unsigned short) at C:\Users\YYYYY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\libraries\ESP8266WiFi\src/include/ClientContext.h line 136
0x40208464: BearSSL::WiFiClientSecure::available() at C:\Users\YYYYY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 358
0x40208772: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\YYYYY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 218
0x40209a4d: MQTTClient::connected() at C:\Users\YYYYY\Documents\Arduino\libraries\MQTT\src/MQTTClient.h line 456
0x40203979: MQTTClient::connect(char const*, bool) at C:\Users\YYYYY\Documents\Arduino\libraries\MQTT\src/MQTTClient.h line 289
0x40209083: String::changeBuffer(unsigned int) at C:\Users\YYYYY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266\WString.cpp line 156
0x402089dc: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\YYYYY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266/HardwareSerial.h line 174
0x40208ce5: Print::write(char const*) at C:\Users\YYYYY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266/Print.h line 60
0x4020794b: BearSSL::WiFiClientSecure::connected() at C:\Users\YYYYY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 246
0x40209a4d: MQTTClient::connected() at C:\Users\YYYYY\Documents\Arduino\libraries\MQTT\src/MQTTClient.h line 456
0x40203ceb: connectToMqtt(bool) at C:\Users\YYYYY\Documents\GitHub\ESP-MQTT-AWS-IoT-Core\Arduino\MQTT/MQTT.ino line 120
0x40209040: String::invalidate() at C:\Users\YYYYY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266\WString.cpp line 141
0x40203e7a: checkWiFiThenMQTT() at C:\Users\YYYYY\Documents\GitHub\ESP-MQTT-AWS-IoT-Core\Arduino\MQTT/MQTT.ino line 162
0x40209a4d: MQTTClient::connected() at C:\Users\YYYYY\Documents\Arduino\libraries\MQTT\src/MQTTClient.h line 456
0x40205288: loop() at C:\Users\YYYYY\Documents\GitHub\ESP-MQTT-AWS-IoT-Core\Arduino\MQTT/MQTT.ino line 241
0x402096b0: loop_wrapper() at C:\Users\YYYYY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266\core_esp8266_main.cpp line 125
```

earlephilhower, 2019-01-28:
This is memory corruption because you are passing in a stack allocated, function-local X509 cert.  As soon as setup() ends, the cert is gone and its memory will be reused by other functions, and very bad things will happen.

https://github.com/debsahu/ESP-MQTT-AWS-IoT-Core/blob/78ff5998a8ec5b72f319382e5fbde9770a32650c/Arduino/MQTT/MQTT.ino#L217

See the examples for info on how to allocate on heap/globally.

closed 2019-01-28 by earlephilhower

debsahu, 2019-01-28:
@earlephilhower Thank you! I am embarrassed with that amateur oversight. Declaring 
```
BearSSL::X509List cert(cacert);
BearSSL::X509List client_crt(client_cert);
BearSSL::PrivateKey key(privkey);
```
globally solved all issues.

prassaad, 2019-02-24:
Any one help me solve this...
error: 'X509List' in namespace 'BearSSL' does not name a type on compiling

Alejandrocha, 2020-03-15:
This may be late. It happened to me - I'm guessing - after I uploaded the code while choosing the wrong board on the IDE.

I solved it by re-flahing the ESP8266.