esp8266/Arduino #5686: MDNS.setInstanceName("XXXXXXX") causing crashes

debsahu, 2019-01-28:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[core v2.5.0-beta3]
- Development Env:		[Arduino IDE|Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

`MDNS.setInstanceName("XXXXXXXXXXXXXXX");` when paired with `MDNS.update();` in loop() causes crashes.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

Below is the **mDNS_Web_Server** example with `MDNS.setInstanceName("XXXXXXXXXXXXXXX");` line added before MDNS.begin();

```cpp
/*
  ESP8266 mDNS responder sample

  This is an example of an HTTP server that is accessible
  via http://esp8266.local URL thanks to mDNS responder.

  Instructions:
  - Update WiFi SSID and password as necessary.
  - Flash the sketch to the ESP8266 board
  - Install host software:
    - For Linux, install Avahi (http://avahi.org/).
    - For Windows, install Bonjour (http://www.apple.com/support/bonjour/).
    - For Mac OSX and iOS support is built in through Bonjour already.
  - Point your browser to http://esp8266.local, you should see a response.

*/

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiClient.h>

#ifndef STASSID
#define STASSID "iHacKedUrWiFi"
#define STAPSK  "aishu1357"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

// TCP server at port 80 will respond to HTTP requests
WiFiServer server(80);

void setup(void) {
[... 145 lines cut ...]

```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x402090d2: esp8266::MDNSImplementation::MDNSResponder::_sendMDNSMessage_Multicast(esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&, int) at C:\Users\IIIIIIIIIIII\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 121
EXCVADDR: 0x00000000

Decoding stack results
0x40206d3c: esp8266::MDNSImplementation::MDNSResponder::stcMDNS_RRDomain::addLabel(char const*, bool) at C:\Users\IIIIIIIIIIII\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\libraries\ESP8266mDNS\src\LEAmDNS_Structs.cpp line 676
0x402091d2: esp8266::MDNSImplementation::MDNSResponder::_sendMDNSMessage(esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at C:\Users\IIIIIIIIIIII\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 94
0x402091e6: esp8266::MDNSImplementation::MDNSResponder::_sendMDNSMessage(esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at C:\Users\IIIIIIIIIIII\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 95
0x40206e88: esp8266::MDNSImplementation::MDNSResponder::stcMDNS_RRQuestion::stcMDNS_RRQuestion() at C:\Users\IIIIIIIIIIII\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\libraries\ESP8266mDNS\src\LEAmDNS_Structs.cpp line 869
0x40204e14: esp8266::MDNSImplementation::MDNSResponder::_sendHostProbe() at C:\Users\IIIIIIIIIIII\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp line 1235
0x4020527a: esp8266::MDNSImplementation::MDNSResponder::_updateProbeStatus() at C:\Users\IIIIIIIIIIII\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp line 1053
0x402034fc: ESP8266WiFiSTAClass::isConnected() at C:\Users\IIIIIIIIIIII\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 339
0x402060fb: esp8266::MDNSImplementation::MDNSResponder::_process(bool) at C:\Users\IIIIIIIIIIII\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp line 82
0x402045fe: esp8266::MDNSImplementation::MDNSResponder::update() at C:\Users\IIIIIIIIIIII\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\libraries\ESP8266mDNS\src\LEAmDNS.cpp line 1095
0x402026af: loop() at C:\Users\IIIIIIIIIIII\AppData\Local\Temp\1\arduino_modified_sketch_795762/mDNS_Web_Server.ino line 80
0x40100500: _umm_free at C:\Users\IIIIIIIIIIII\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266\umm_malloc\umm_malloc.c line 1300
0x40100948: free at C:\Users\IIIIIIIIIIII\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266\umm_malloc\umm_malloc.c line 1760
0x402098f0: String::invalidate() at C:\Users\IIIIIIIIIIII\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266\WString.cpp line 141
0x4020990c: String::~String() at C:\Users\IIIIIIIIIIII\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266\WString.cpp line 125
0x40202612: setup() at C:\Users\IIIIIIIIIIII\AppData\Local\Temp\1\arduino_modified_sketch_795762/mDNS_Web_Server.ino line 72
0x40209ecd: esp_schedule() at C:\Users\IIIIIIIIIIII\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266\core_esp8266_main.cpp line 95
0x40209f58: loop_wrapper() at C:\Users\IIIIIIIIIIII\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266\core_esp8266_main.cpp line 125
```

LaborEtArs, 2019-01-28:
In fact this is a bug in the mDNSResponder. The call to ‘setInstanceName’ before ‘begin()’ causes a skipping of the initialization code in ‘begin()’.
As a quick workaround you can place ‘setInstanceName’ after ‘begin’.

devyte, 2019-01-30:
Closing via #5689 .

closed 2019-01-30 by devyte