esp8266/Arduino #5688: ESP8266-12F failed after flash and `Erase all the flash`

kingofnull, 2019-01-28:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP8266-12F]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[other]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

The firmware flash but don't run after I choose Erase: All Flash Contents.
The module (ESP8266-12F) don't works any more and only flash-able with esptool but the don't load any firmware (I try some other firmware-s but none of them worked).

the out but log after boot is this:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 25020, room 16 
tail 12
chksum 0x9a
ho 0 tail 12 room 4
load 0xffffffff, len -1, room 12 

```

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

#Flash Log
```
Sketch uses 258024 bytes (51%) of program storage space. Maximum is 499696 bytes.
Global variables use 26656 bytes (32%) of dynamic memory, leaving 55264 bytes for local variables. Maximum is 81920 bytes.
Uploading 262176 bytes from C:\Users\...\AppData\Local\Temp\arduino_build_552560/Blink.ino.bin to flash at 0x00000000
................................................................................ [ 31% ]
................................................................................ [ 62% ]
................................................................................ [ 93% ]
.................                                                                [ 100% ]
```

#Boot Log (Baud Rate:74880)
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 25020, room 16 
tail 12
chksum 0x9a
ho 0 tail 12 room 4
load 0xffffffff, len -1, room 12 

```

devyte, 2019-01-29:
@kingofnull said:
> Maximum is 499696 bytes.

You have set 512KB of flash size, which disagrees with what you said in the template:
> Flash Size: [4MB]

I'm pretty sure there are no ESP12Fs out there with flash that size. The only boards with that flash size are older ESP-01s, which are no longer supported even by Espressif.
That means you either have custom hw, or more likely, you have wrong settings/wrong board selected in the IDE.

In addition, all of my boards are ESP12Fs, and I can't reproduce.

Closing due to user error.

closed 2019-01-29 by devyte

bertrandlo, 2019-01-29:
Both ESP-01s and ESP-12F i owned. These two modules worked well with the newer Arudino Core and IDE.

ESPtool could detect hardware configuration. Maybe you could follow that detection message to modify the board configuration. then recompiled for download ?

kingofnull, 2019-01-29:
> @kingofnull said:
>
>
> You have set 512KB of flash size, which disagrees with what you said in the template:
>
>
> I'm pretty sure there are no ESP12Fs out there with flash that size. The only boards with [...]

I tried it before with 4MB config but nothing changed.
And about setting I don't what is the proper setting for 12F module if I doing something wrong please notice me.

kingofnull, 2019-01-29:
> Both ESP-01s and ESP-12F i owned. These two modules worked well with the newer Arudino Core and IDE.
> 
> ESPtool could detect hardware configuration. Maybe you could follow that detection message to modify the board configuration. then recompiled for download ?

I didn't get what do you mean but I set pins this way:
- GPIO 15 -> GND
- GPIO 2 -> VCC
- GPIO 0 -> GND
- En -> VCC

I will appreciate  if you describe more about how I can decode and use that detection message to configure setting.