esp8266/Arduino #5694: Binary signing should be available under Windows

Misiu, 2019-01-29:
[This PR](https://github.com/esp8266/Arduino/pull/5213) adds support for signing binary file with certificate.
Unfortunately it is available only on Linux and Mac.
As suggested by @earlephilhower I'm oppening this issue to track support for Windows.

OpenSSL it is easily installable on Windows - https://wiki.openssl.org/index.php/Binaries.
I don't mind manual installation of it, especially is it done only once.
If installing OpenSSL manually and adding it to the path (so that OpenSSL executable is available from the command line) is the only step then this should be definitively added to docs.

earlephilhower, 2019-01-30:
This would be predicated on #5635.  If we don't include Python by default w/the Windows install, this isn't (sanely) possible.

Misiu, 2019-01-31:
@earlephilhower if Python will be installed on Windows (manual step is done one by the user) then this is doable?
I think that everybody who needs binary signing will manage to install Python by themselves.
I'm using PlatformIO with Visual Studio Code, so Python is one of the requirements.

earlephilhower, 2019-01-31:
@Misiu, the problem is if python is not installed the IDE will abort when you try and start the signing.py script and it won't ever successfully build a BIN.

The referenced PR installs a complete copy of python as part of the core subdirectory, so we can safely call that tool on all Windows boxes and signing.py can then gracefully exit if OpenSSL isn't present.

earlephilhower, 2019-02-25:
Signing should now work if OpenSSL command line utilities are installed since #5635 was merged.

closed 2019-02-25 by earlephilhower

fa1rid, 2020-05-02:
I have both Python and openSSL installed and added to the path but I get this error:
`[WinError 2] The system cannot find the file specifiedNot signing the generated binary
`