esp8266/Arduino #5697: 2.5.0-beta3 doesn't compile with Generic ESP8266 Modules

4Source, 2019-01-30:
### Basic Infos

- [ X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X ] I have tested that the issue is present in current master branch (aka latest git).
- [ X ] I have searched the issue tracker for a similar issue.
- [ X ] If there is a stack dump, I have decoded it.
- [ X ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[30.01.19]
- Development Env:		[Arduino]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

Didnt compile with latest version 2.5.0-beta3. Previously I used Version 2.4.2 every thing works fine.
Error shown: 
exec: "cmd.exe": executable file not found in% PATH%
Compile Error for the Board Generic ESP8266 Modules.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

Exampel/ESP8266/Blink.ino
or all other tested programms

lrmoreno007, 2019-01-30:
It does not seem the same problem, but check this please:

https://github.com/esp8266/Arduino/blob/master/doc/faq/a04-board-generic-is-unknown.rst

lrmoreno007, 2019-01-30:
Ok, forget my last comment.

The problem is that the user you use in Windows does not have enough privileges. Run Arduino IDE as an administrator and it should work, or elevate your user's privileges.

devyte, 2019-01-31:
This is either a problem in your Windows, or a problem with your installation. I suggest a clean install from scratch of the Arduino IDE and then of this core. If that doesn't work, you'll need help from someone who knows a bit about how Windows works to figure out why cmd.exe isn't found. I suggest seeking help at esp8266.com or stackoverflow.

Either way, it's not a problem in the core hosted in this repo. Closing.

closed 2019-01-31 by devyte

zoomx, 2019-02-19:
I got the same error. Core 2.5.0, Windows Seven
The solution, for me, was to delete all folders in Temp (C:\Users\<YourUsername>\AppData\Local\Temp\) because enabling verbose output during compilation I noticed that the compiler used a folder with compiled binaries of another sketch.

It's not a PATH problem, I have not C:\Windows\system32 in my path and it works.

zoomx, 2019-02-20:
Further investigations.

In 2.4.2 core platform.txt has this line

`## windows-compatible version without git`

in 2.5.0 core platform.txt there are these lines

```
## windows-compatible version without git
recipe.hooks.core.prebuild.1.pattern.windows=cmd.exe /c rem cannot sign on windows
```

in 2.5.6-Dev core platform.txt these lines where removed again

So I was wrong, I need C:\Windows\system32 only for 2.5.0 but it is not a Windows problem. It was changed something between cores <=2.4.2 and 2.5.0 that need cmd.exe.
If I put a comment on recipe.hooks.core...... I get an error that python is not in my PATH.

4Source, 2019-02-23:
> You installed with Board Manager or from Git?

I installed with Board Manager.

Thanks for all your help.