esp8266/Arduino #5700: Board manager error downloading issue

bandaru5, 2019-01-31:
can you help me with the following error. I'm using Arduino IDE 1.8.8 and using Olimexino 32U4 board.
I'm getting the following error 
   Error downloading https://raw.githubusercontent.com/OLIMEX/Arduino_configurations/master/AVR/package_olimex _avr_index.json

I tried opening the page with the link but it says 404 : not found.

please help me, I'm stuck here.
[image]

earlephilhower, 2019-01-31:
This is not the OLIMEX repo nor the AVR arduino (as shown in your URL), so please direct your question there.  Closing.

closed 2019-01-31 by earlephilhower

bandaru5, 2019-01-31:
I don't understand the error. where should I direct this. can you help me with this.