esp8266/Arduino #5701: ESP resets when a client is connection using BearSSL in STATION-mode

Chrizey91, 2019-01-31:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					ESP-8266
- Core Version:			2.5.0beta-1|2.5.0beta-2|2.5.0beta-3|2.5.0dev(git)
- Development Env:		ArduinoIDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio|dio
- Flash Size: 		4MB/1MB|4MB/0MB|2MB/0MB
- lwip Variant:   v2 Lower Memory|Higher Bandwidth
- Reset Method:		ck
- Flash Frequency:	40Mhz|80Mhz
- CPU Frequency:		80Mhz|160MHz
- Upload Using:		SERIAL
- Upload Speed:   115200

I have the exact same problem with the exact same arduino sketch that was posted as an [issue](https://github.com/esp8266/Arduino/issues/4898#issue-339063339) here before.

I upload the provided sketch to my ESP8266, connect with Chrome and after 3 times of "Incomming connection" and "Connection closed" it freezes for a few seconds after "Incomming connection" and then resets.

I used different clock/flash speeds and Iwip versions and Core versions (see above) but it ends alyways the same way...

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <time.h>

#ifndef STASSID
#define STASSID "SSID"
#define STAPSK  "PW"
#endif

const char *ssid = STASSID;
const char *pass = STAPSK;

// The HTTPS server
BearSSL::WiFiServerSecure server(443);

// The server's private key which must be kept secret
const char server_private_key[] PROGMEM = R"EOF(
-----BEGIN PRIVATE KEY-----
MIIEvAIBADANBgkqhkiG9w0BAQEFAASCBKYwggSiAgEAAoIBAQDJblrg47vF3qlE
NMRM7uG8QwE6v/AKpxOL+CLb/32s+dW9Psgf+oZKJgzGkYUoJdWpLitTmTZeykAs
Sq7Iax5Rq/mGqyAc7oJAUUAupfNRU0KwkD1XqtpQWEFoiqoIqZbOZ4CRX5q8z/MN
BH1aPVBMKaL33uwknkgJBzxwZJ2+uGKxRJt8+koj1CXgUCk5lEAEEG5kqE326MjN
O/c4gBqulBV8AIoq6/trY3apTS7FEOiN47qh1PVzoBm/oGVwXvoZAZOj7+gGGo91
sBC5oHJy5Y2BOcNB3opTNXQTiK3Z80b5wc3iQS+h83qAfHwhs6tfAW22WkAf+jtt
x8KdRWFNAgMBAAECggEAPd+jFL9/d1lc/zGCNuuN9YlTgFti/bKyo2UWOCOz1AVu
LVJyoLgQtggYFoqur1Vn2y7uaiB+/gD8U16hb7jPuGCuJjq8g4aUBfOvVmTtZ8a+
joPQA/TcWJ+zf8xQTJbjVwWeDYmje2oZC5+cbbK1zp9fiuoz+U+RawyI+TE+700i
ESCmsKFIHy2Ifruva8HgcPYIPpZ9zLxJj0Dii+WDs7zM9h2dzO4HfImSG/DPmgoV
ydo9IcrUE7KoMLa8Uo7u1b2h6BnTn7GfYiMSUsYcYR3CnpDBknBWjZMwrV0uqv9q
TbVc4QXt+c1q89HDg7BIJaOAzbCvJfgAfXUqZyqwQQKBgQD5ENFjicUzCqPw7fOy
Q5Z8GeUbIJ5urT1MheAq7SPd2kK8TsO3hUjNC0LLNSyKPs6gsYaIiObO3wDGeZZk
xeHBhrUVaz2nIjI7TrnCUpMDOrdxcPr4bc+ifV5YT4W3OFBWQ9chQEx3Nm3DbiX4
fpno34AiFrJF791JkTPFj9OIUQKBgQDPCgcae1pQr77q+GL5Q2tku3RrE4cWtExf
m8DzAb4Vxe3EhPz8bVr+71rqr/KqNfG1uKE3sT0fhB6VMTkHTOQU13jDrvpPUS3W
Vg8cVr5/+iiyF0xb+W8LQ+GVdR5xnMPSZHUtXyURvtzT4nnTAlAtN7lEytX9BzbX
xhltOOwGPQKBgA/Y/BnDSGLpCGlqGpl7J3YaB7PkLXCJYV8fHZZdpGyXWKu2r0lc
F7fEQanAZmcde/RJl2/UlisPkXMPhXxAAw9XTOph+nhJ+rw/VB6DNot8DvQO5kks
Y4vJQlmIJc/0q1fx1RxuhO8I7Y8D0TKwi4Z/wh1pKEq+6mul649kiWchAoGAWn8B
l9uvIHGRO9eSO23ytTcSrfL9Kzln4KqN7iom0hGP2kRe6F9MVP5+ePKrWSb3Hf0z
ysoX83ymeYPob352e32rda04EA9lv7giJrrrzbikrSNt5w3iMcRcCB4HTpW9Kmtq
pIhgBZ+tmpf1s/vg28LtoloeqtjKagpW9tzYnekCgYAZFZ84EGqS9SHw5LELgGY4
mQLMwbYZ6wBMA2PlqYi/17hoAVWz37mLDjtWDB4ir78QMoGbesQVtK9W/4vzmez4
ZLKlffdL5tCtA08Gq9aond1z83Xdnh1UjtwHIJvJPc/AoCFW1r5skv/G6acAk6I2
Zs0aiirNGTEymRX4rw26Qg==
-----END PRIVATE KEY-----
)EOF";

// The server's public certificate which must be shared
const char server_cert[] PROGMEM = R"EOF(
-----BEGIN CERTIFICATE-----
MIIDUTCCAjmgAwIBAgIJAOcfK7c3JQtnMA0GCSqGSIb3DQEBCwUAMD8xCzAJBgNV
BAYTAkFVMQ0wCwYDVQQIDAROb25lMQ0wCwYDVQQKDAROb25lMRIwEAYDVQQDDAlF
U1BTZXJ2ZXIwHhcNMTgwMzE0MTg1NTQ1WhcNMjkwNTMxMTg1NTQ1WjA/MQswCQYD
VQQGEwJBVTENMAsGA1UECAwETm9uZTENMAsGA1UECgwETm9uZTESMBAGA1UEAwwJ
RVNQU2VydmVyMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAyW5a4OO7
xd6pRDTETO7hvEMBOr/wCqcTi/gi2/99rPnVvT7IH/qGSiYMxpGFKCXVqS4rU5k2
XspALEquyGseUav5hqsgHO6CQFFALqXzUVNCsJA9V6raUFhBaIqqCKmWzmeAkV+a
vM/zDQR9Wj1QTCmi997sJJ5ICQc8cGSdvrhisUSbfPpKI9Ql4FApOZRABBBuZKhN
9ujIzTv3OIAarpQVfACKKuv7a2N2qU0uxRDojeO6odT1c6AZv6BlcF76GQGTo+/o
BhqPdbAQuaBycuWNgTnDQd6KUzV0E4it2fNG+cHN4kEvofN6gHx8IbOrXwFttlpA
H/o7bcfCnUVhTQIDAQABo1AwTjAdBgNVHQ4EFgQUBEk8LqgV+sMjdl/gpP1OlcNW
14EwHwYDVR0jBBgwFoAUBEk8LqgV+sMjdl/gpP1OlcNW14EwDAYDVR0TBAUwAwEB
/zANBgkqhkiG9w0BAQsFAAOCAQEAO1IrqW21KfzrxKmtuDSHdH5YrC3iOhiF/kaK
xXbigdtw6KHW/pIhGiA3BY5u+d5eVuHTR5YSwIbbRvOjuoNBATAw/8f5mt5Wa+C3
PDpLNxDys561VbCW45RMQ0x5kybvDYi0D1R/grqZ18veuFSfE6QMJ/mzvr575fje
8r5Ou0IZOYYF8cyqG5rA4U7BYXEnH44VgwlpkF8pitPsnyUWaAYqE0KnZ0qw0Py4
HCkfGJNlNOOamnr6KakVlocwKY0SdxcLoXSs5ogTQvTSrAOjwcm1RA0hOCXr8f/f
UsQIIGpPVh1plR1vYNndDeBpRJSFkoJTkgAIrlFzSMwNebU0pg==
-----END CERTIFICATE-----
)EOF";

void setup() {
  system_update_cpu_freq(160);
  Serial.begin(115200);
  Serial.println();
  Serial.println();

  // We start by connecting to a WiFi network
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");

  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  // Attach the server private cert/key combo
  BearSSL::X509List *serverCertList = new BearSSL::X509List(server_cert);
  BearSSL::PrivateKey *serverPrivKey = new BearSSL::PrivateKey(server_private_key);
  server.setRSACert(serverCertList, serverPrivKey);

  // Actually start accepting connections
  server.begin();
}

static const char *HTTP_RES =
        "HTTP/1.0 200 OK\r\n"
        "Connection: close\r\n"
        "Content-Length: 62\r\n"
        "Content-Type: text/html; charset=iso-8859-1\r\n"
        "\r\n"
        "<html>\r\n"
        "<body>\r\n"
        "<p>Hello from ESP8266!</p>\r\n"
        "</body>\r\n"
        "</html>\r\n";

void loop() {
  BearSSL::WiFiClientSecure incoming = server.available();
  if (!incoming) {
    return;
  }
  Serial.println("Incoming connection...\n");
  
  // Ugly way to wait for \r\n (i.e. end of HTTP request which we don't actually parse here)
  uint32_t timeout=millis() + 1000;
  int lcwn = 0;
  for (;;) {
    unsigned char x=0;
    if ((millis() > timeout) || (incoming.available() && incoming.read(&x, 1) < 0)) {
      incoming.stop();
      Serial.printf("Connection error, closed\n");
      return;
    } else if (!x) {
      yield();
      continue;
    } else if (x == 0x0D) {
      continue;
    } else if (x == 0x0A) {
      if (lcwn) {
        break;
      }
      lcwn = 1;
    } else
      lcwn = 0;
  }
  Serial.println("Writing to client");
  incoming.write((uint8_t*)HTTP_RES, strlen(HTTP_RES));
  incoming.flush();
  incoming.stop();
  Serial.printf("Connection closed.\n");
}

```

### Debug Messages

```
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 11
cnt 

connected with ChrizFr1z, channel 11
dhcp client start...
wifi evt: 0
ip:192.168.178.37,mask:255.255.255.0,gw:192.168.178.1
wifi evt: 3
.
WiFi connected
IP address: 
192.168.178.37
pm open,type:2 0
WS:ac
:rn 517
:ref 1
:rd 5, 517, 0
:rdi 517, 5
:rd 512, 517, 5
:rdi 512, 512
:c0 512, 517
:wr 517 0
:wrc 517 517 0
:wr 517 0
:wrc 517 517 0
:wr 229 0
:wrc 38 229 0
:wr 191 38
:wr 191 38
WS:ac
:rn 517
WS:ac
:wr 191 38
:wr 191 38
:wr 191 38
:ack 1072
:wr 191 38
:wrc 191 191 0
:ack 191
:rn 93
:rd 5, 93, 0
:rdi 93, 5
:rd 37, 93, 5
:rdi 88, 37
:rd 5, 93, 42
:rdi 51, 5
:rd 1, 93, 47
:rdi 46, 1
:rd 5, 93, 48
:rdi 45, 5
:rd 40, 93, 53
:rdi 40, 40
:c0 40, 93
:wr 6 0
:wrc 6 6 0
:wr 45 0
:wrc 45 45 0
WS:av
:ref 2
:ur 2
Incoming connection...

:ack 6
:ack 45
:rcl
:abort
Connection error, closed
:ur 1
WS:dis
:del
:ref 1
:rd 5, 517, 0
:rdi 517, 5
:rd 512, 517, 5
:rdi 512, 512
:c0 512, 517
:wr 517 0
:wrc 517 517 0
:wr 517 0
:wrc 517 517 0
:wr 229 0
:wrc 38 229 0
:wr 191 38
:wr 191 38
:wr 191 38
:ack 1072
:wr 191 38
:wrc 191 191 0
:ack 191
:rn 93
:rd 5, 93, 0
:rdi 93, 5
:rd 37, 93, 5
:rdi 88, 37
:rd 5, 93, 42
:rdi 51, 5
:rd 1, 93, 47
:rdi 46, 1
:rd 5, 93, 48
:rdi 45, 5
:rd 40, 93, 53
:rdi 40, 40
:c0 40, 93
:wr 6 0
:wrc 6 6 0
:wr 45 0
:wrc 45 45 0
WS:av
:ref 2
:ur 2
Incoming connection...

:ack 6
:ack 45
:rcl
:abort
:rn 517
Connection error, closed
:ur 1
WS:dis
:del
:ref 1
:rd 5, 517, 0
:rdi 517, 5
:rd 512, 517, 5
:rdi 512, 512
:c0 512, 517
:wr 517 0
:wrc 517 517 0
:wr 517 0
:wrc 517 517 0
:wr 229 0
:wrc 38 229 0
:wr 191 38
:wr 191 38
:wr 191 38
:ack 1072
:wr 191 38
:wrc 191 191 0
:ack 191
:rn 93
:rd 5, 93, 0
:rdi 93, 5
:rd 37, 93, 5
:rdi 88, 37
:rd 5, 93, 42
:rdi 51, 5
:rd 1, 93, 47
:rdi 46, 1
:rd 5, 93, 48
:rdi 45, 5
:rd 40, 93, 53
:rdi 40, 40
:c0 40, 93
:wr 6 0
:wrc 6 6 0
:wr 45 0
:wrc 45 45 0
WS:av
:ref 2
:ur 2
Incoming connection...

:ack 6
:ack 45
:rn 436

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

earlephilhower, 2019-01-31:
Did you set the CPU speed to 160MHz?  It's in the Tools menu IIRC, and it's not listed in your form above.

80MHz is marginal for enabling any form of SSL communications with the very slow core and flash interface we've got on the 8266.

Chrizey91, 2019-01-31:
I did list it in the form. It's she seventh entry under "Setting in IDE". :)
Yes, I tried that, also a lot of different Flash sizes and Flash frequencies, if that would make any differences.

earlephilhower, 2019-01-31:
Ah, I saw "80|160" and didn't realize you meant both and hadn't just left the field chosen.

closed 2019-02-01 by earlephilhower

Chrizey91, 2019-02-01:
Thanks a lot! Can confirm that this commit solves the issue.
It even works with a CPU frequency of only 80Mhz!

PR #5702 (2019-02-01): Fix BearSSL Server WDT
Fixes #5701 WDTs and other issues with BearSSL::WiFiServerSecure

The BSSL server was creating the client it returns on a connection in a
way that caused the counter for the stack_thunk to get out of sync and
cause it to be freed improperly by having the destructor be called on
more time than the constructor.  Looks like RVO.

Rewrite the ::available() function in order to avoid this issue with
help from @devyte.
commit 3f8cd46d (2019-02-01): Fix BearSSL Server WDT (#5702)

Fixes #5701 WDTs and other issues with BearSSL::WiFiServerSecure

The BSSL server was creating the client it returns on a connection in a
way that caused the counter for the stack_thunk to get out of sync and
cause it to be freed improperly by having the destructor be called one
more time than the constructor.  Looks like RVO.

Rewrite the ::available() function in order to avoid this issue with
help from @devyte.