esp8266/Arduino #5705: TinyGSM example didn't compile with 2.5.0-beta3.

ChrSchultz, 2019-02-01:
If I try to compile the TinyGSM-example forHttpClient (see file)
[test.ino.txt](https://github.com/esp8266/Arduino/files/2821530/test.ino.txt) rename to .ino.
I got the following errors:

In file included from D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\ArduinoHttpClient\src/ArduinoHttpClient.h:8:0,

                 from D:\Benutzer\ic-sc\Dokumente\Arduino\test\test.ino:32:

D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\ArduinoHttpClient\src/HttpClient.h: In member function 'virtual void HttpClient::flush()':

D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\ArduinoHttpClient\src/HttpClient.h:310:50: error: return-statement with a value, in function returning 'void' [-fpermissive]

     virtual void flush() { return iClient->flush(); };

                                                  ^

D:\Benutzer\ic-sc\Dokumente\Arduino\test\test.ino: At global scope:

test:67:15: error: cannot declare variable 'client' to be of abstract type 'TinyGsmSim800::GsmClient'

 TinyGsmClient client(modem);

               ^

In file included from D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\arduino_267956\src/TinyGsmClientSIM808.h:12:0,

                 from D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\arduino_267956\src/TinyGsmClient.h:29,

                 from D:\Benutzer\ic-sc\Dokumente\Arduino\test\test.ino:31:

D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\arduino_267956\src/TinyGsmClientSIM800.h:53:7: note:   because the following virtual functions are pure within 'TinyGsmSim800::GsmClient':

 class GsmClient : public Client

       ^

In file included from D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\arduino_267956\src/TinyGsmCommon.h:25:0,

                 from D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\arduino_267956\src/TinyGsmClientSIM800.h:21,

                 from D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\arduino_267956\src/TinyGsmClientSIM808.h:12,

                 from D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\arduino_267956\src/TinyGsmClient.h:29,

                 from D:\Benutzer\ic-sc\Dokumente\Arduino\test\test.ino:31:

D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266/Client.h:29:21: note: 	virtual int Client::connect(const IPAddress&, uint16_t)

         virtual int connect(CONST IPAddress& ip, uint16_t port) =0;

                     ^

D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266/Client.h:37:22: note: 	virtual bool Client::flush(unsigned int)

         virtual bool flush(unsigned int maxWaitMs = 0) = 0;

                      ^

D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266/Client.h:38:22: note: 	virtual bool Client::stop(unsigned int)

         virtual bool stop(unsigned int maxWaitMs = 0) = 0;

                      ^

test:68:12: error: cannot declare variable 'http' to be of abstract type 'HttpClient'

 HttpClient http(client, server, port);

            ^

In file included from D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\ArduinoHttpClient\src/ArduinoHttpClient.h:8:0,

                 from D:\Benutzer\ic-sc\Dokumente\Arduino\test\test.ino:32:

D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\ArduinoHttpClient\src/HttpClient.h:41:7: note:   because the following virtual functions are pure within 'HttpClient':

 class HttpClient : public Client

       ^

In file included from D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\arduino_267956\src/TinyGsmCommon.h:25:0,

                 from D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\arduino_267956\src/TinyGsmClientSIM800.h:21,

                 from D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\arduino_267956\src/TinyGsmClientSIM808.h:12,

                 from D:\Benutzer\ic-sc\Dokumente\Arduino\libraries\arduino_267956\src/TinyGsmClient.h:29,

                 from D:\Benutzer\ic-sc\Dokumente\Arduino\test\test.ino:31:

D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266/Client.h:29:21: note: 	virtual int Client::connect(const IPAddress&, uint16_t)

         virtual int connect(CONST IPAddress& ip, uint16_t port) =0;

                     ^

D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266/Client.h:37:22: note: 	virtual bool Client::flush(unsigned int)

         virtual bool flush(unsigned int maxWaitMs = 0) = 0;

                      ^

D:\Benutzer\ic-sc\Dokumente\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta3\cores\esp8266/Client.h:38:22: note: 	virtual bool Client::stop(unsigned int)

         virtual bool stop(unsigned int maxWaitMs = 0) = 0;

                      ^
If i use esp8266 library 4.2.4 (stable) compilation will work fine. so I think, the TinyGSM-library is working.

Arduino-IDE 1.8.8 windows-store Win10-64bit, TinyGSM, ESP8266-2.5.0-beta3 Lolin D1 mini.

earlephilhower, 2019-02-01:
@ChrSchultz, this looks like a 3rd-party library issue.  Your best best for help is to get in touch with the TinyGSM author for assistance, or try https://www.esp8266.com forums.

ChrSchultz, 2019-02-01:
no I don't think that this is relatet to 3rd party lib. with esp8266-4.2.4 it works fine and compilation runs.

earlephilhower, 2019-02-01:
`ArduinoHttpClient` isn't the same as our `WiFiClient` and it looks like it has a function definition problem from what I see.

I'll reopen as maybe this was due to a change in our code.  @d-a-v, did we do any signature changes in the `Client` class?

devyte, 2019-02-02:
The signature for flush() did change in its return value.
Having said that, this line:
`virtual void flush() { return iClient->flush(); };`
makes no sense. While it is legal to return a void value in a function declared with return void, I believe there is only one case where implementing in that way makes sense: template functions/methods, where the return type is a template parameter, and the return type of the called method must be the same as the return type of the caller method, and that parameter could potentially be void type. This is not the case here.

A method with return void like in the case shown should be implemented like this:
`virtual void flush() { iClient->flush(); };`
or this:
`virtual void flush() { (void)iClient->flush(); };`
or at worst this:
`virtual void flush() { return (void)iClient->flush(); };`

The ArduinoHttpClient library should be fixed to decouple the return types as above. 

In addition, the ArduinoHttpClient library used is the one from arduino-libraries, and its implementation does not state explicitly that it supports the ESP. There could be hidden issues, such as timing, because standard implementations from arduino are not under the same timing restrictions as code for the ESP. I suggest either looking for an implementation that explicitly states ESP support, or using our ESP8266HttpClient.

pfabri, 2019-04-09:
@ChrSchultz 
Dear Christoph,
Could you please let me know how to roll back to version 2.4.2? I use PlatformIO but for the life of my I can't figure out how to do this. I've perused the docs but to no avail.
Many thanks!

d-a-v, 2019-04-09:
It wIll be fixed with #5749

pfabri, 2019-04-09:
@d-a-v Regardless of #5749, could you please point me into the right direction on how to downgrade to a custom esp8266/Arduino firmware? I'd greatly appreciate it.
Thanks!

ChrSchultz, 2019-04-10:
@pfabri regardles the coming bugfix you can roll back platformio platform espressif8266 to 2.0.0  with:
`` 
pio platform install espressif8266@2.0.0 
`` 
what maches esp8266 version 4.2.4  see 
https://github.com/platformio/platform-espressif8266

pfabri, 2019-04-12:
@ChrSchultz Many thanks!

closed 2019-04-26 by d-a-v

roysG, 2022-01-03:
Can you send please working example of ssl with tinyGSM?

PR #5969 (2019-04-26): restore proper arduino Client:: & Wire:: API
fixes #5749
fixes #5901 
fixes #5705 
~includes #5997~