esp8266/Arduino #5707: 2.5.3 IRAM usage core_esp8266_si2c.c

freeck, 2019-02-01:
Hi there, I was rather surprised by the increase of ram-usage of my application when using release v2.5.3 for the first time. The map shows that the i2c-library is the greatest consumer of ram i.e. Aprox. 2500 bytes! As on the other hand the developers did their utmost to decrease ram-usage with aprox 700 bytes....
This increase is caused by the extension of the library with slave-functionality.
Wouldn't it be a good idea to have two versions of this library , with and without slave-functions. Or some kind of conditional compilation and/or linking two separate modules.

earlephilhower, 2019-02-02:
Could you give a specific example or sketch that shows a difference?

freeck, 2019-02-02:
Mind you: I use Sloeber as IDE. 
I just included "wire.h" in a basic blink.ino example, made one referende to the original library , and there you are. Next I deleted all the IRAMCACHE definitions in the headers of the i2c-functions. Result: 2500 IRAM extra.

earlephilhower, 2019-07-24:
Check out #6326.  It's untested, but reduced IRAM usage by >500 bytes in the TWI code.  Any testing you'd be able to provide would be welcome!

devyte, 2019-10-31:
The referenced PRs are merged, IRAM has improved quite a bit. Closing this.

closed 2019-10-31 by devyte