esp8266/Arduino #5711: mdns is not working STA and AP mode

sonukushwah221, 2019-02-02:
Hello 

I want to use mDns service with AP and STA mode but it's working fine in the AP mode but it's not working in the STA mode. please share the suggestions and solutions how we can access this mdns in both mode.

codebeat-nl, 2019-02-06:
1. Wrong question title, at least one mode works as stated in your question;
2. Are you sure you not confusing both? 

You need a bonjour (mDNS) service on the network (STA=Station mode, part of the network) and with AP (Access Point, own network) there is no bonjour service available so that's why it is not working. This is not an error.

d-a-v, 2019-02-06:
There's a possibility where mDNS can be used in AP mode, where a computer running bonjour or avahi is connected. ESP broadcasts/multicasts its mDNS name and the computer's daemon should be able to resolve this name.
We need to check or reproduce that issue.
- is mDNS broadcasting to 255.255.255.255 or multicasting ?
- is lwIP sending such broadcast/multicast packets on every up/enabled interfaces (ap & sta) ?

devyte, 2019-02-08:
@d-a-v I could be wrong, but I suspect that @sonukushwah221 request is to be able to use mdns in STA_AP mode on both interfaces at the same time. I haven't tested with the @LaborEtArs implementation, but at least the legacy class would work ether in STA mode, or in AP mode, or only on the STA interface when in STA_AP mode.

LaborEtArs, 2019-02-08:
Currently LEAmDNS will work in STATION OR AP mode, not in both. If both are activated, the AP netif is used as the multicast interface.

flav1972, 2019-03-09:
For me is not working in AP mode. When STA or both are enabled it is working.

d-a-v, 2019-06-25:
Is it possible for you to try PR #6224 ?

arantius, 2019-12-07:
I'm commenting on this open bug after finding at least #2602 and #6290 (already resolved)

1. Open Examples > ESP8266mDNS > mDNS_Web_Server
2. Type in SSID/PSK values.
3. Flash/upload.
4. mDNS works, device is discovered correctly (from LAN client, via STA address).
5. Change to `WiFi.mode(WIFI_AP_STA);` (and change the name so it's clear which version is running, by the mDNS answer).
6. Flash/upload.
7. mDNS fails, name never resolves.

I've tested various versions -- 2.3.0 and 2.4.0 definitely work as expected both ways.  This lines up with 2.5.0 including an mDNS rewrite.  As a workaround for now, I've copied the library from 2.4.2 and I'm depending on that instead of the one bundled with the esp8266 core.

TD-er, 2019-12-07:
Do you call this function after changing the wifi mode?
https://github.com/esp8266/Arduino/blob/8a6a7f7480e95bee6b9cf0784d10440c82a0bfda/libraries/ESP8266mDNS/src/ESP8266mDNS_Legacy.h#L86-L87

arantius, 2019-12-07:
I'm not "changing" the wifi mode (at run time).  I'm changing the source code, recompiling, and re-flashing.  Like I wrote above.

ramiws, 2020-05-23:
So is the one bundled with the esp8266 core still broken? How do I check what version is included now in the core I am running?

papadenis, 2020-08-07:
[Try this](https://github.com/esp8266/Arduino/issues/2602#issuecomment-670414843l)
#2602

d-a-v, 2020-10-15:
A major update has landed in current git version. mDNS is now interface agnostic. Can you try it ?

arantius, 2020-10-17:
2.7.4 seems to work as expected but I'm not prepared to build from source to test that.