esp8266/Arduino #5712: Hadware SPI Slave Select (SS) setup and hold bit states in SPI1U not maintained

Bodmer, 2019-02-03:
I had some issues with lost SPI data in overlap mode. Tracked this down to the Slave Select (SS) setup and hold bits getting cleared by different SPI calls.

The bits are set [here](https://github.com/esp8266/Arduino/blob/master/libraries/SPI/SPI.cpp#L127) by the call [here](https://github.com/esp8266/Arduino/blob/master/libraries/SPI/SPI.cpp#L72).

But then immediately afterwards cleared [here](https://github.com/esp8266/Arduino/blob/master/libraries/SPI/SPI.cpp#L84). These bits are also cleared elsewhere in the SPI functions, [here ](https://github.com/esp8266/Arduino/blob/master/libraries/SPI/SPI.cpp#L472)and [here](https://github.com/esp8266/Arduino/blob/master/libraries/SPI/SPI.cpp#L496) and [here](https://github.com/esp8266/Arduino/blob/master/libraries/SPI/SPI.cpp#L508).

devyte, 2019-02-08:
@Bodmer I'm confused. Are you proposing a fix for your issue? If so, please propose your changes in a PR.
If not, then please provide more details, because pointers to the set/clear of those bits isn't enough to understand the problem.
Please be aware the none of the currently active maintainers are involved with low level SPI, let alone overlap mode, so any investigation and testing would need to be done by a user. I encourage you to continue seeking an answer yourself, or you can wait to see if someone else picks this up.

Bodmer, 2019-02-08:
There is a simple bug in the code that may not affect users who do not need longer setup and hold times on the hardware controlled SS line. The setHwCs() function sets the control bits required for extra setup and hold time, but other SPI calls clear them. The links how where this happens.

I can provide a work around for the users code but the esp8266 team need to decide on the best approach to fix the bug.

devyte, 2019-02-19:
@Bodmer I suggest you propose your solution in a PR, that would allow a better understanding of what you mean, and someone familiar with SPI could eventually take a look.

devyte, 2019-04-29:
@JiriBilek do you know if the changes proposed here have merit?

NdK73, 2019-04-29:
I think it's related to the problem I'm having to drive a SPI display + SPI touch controller using the "extra" pins on the ESP12E module, the ones usually reserved for the EEPROM chip.
"Degrading" flash mode to DIO / DOUT frees GPIO9 and GPIO10, that could be used as CS pins for the display and the touch even if they're still connected to the flash chip (/hold and /wp lines: no ill effects if driven low outside of flash access as long as they're brought back high before accessing the flash again).
That would mean having two extra GPIO pins (the ones that would end up used by the two CS lines, usually GPIO0 dor the display and another for the touch).
I'm still "a bit" lost in the ESP datasheet, but it seems doable...

JiriBilek, 2019-04-30:
@devyte sorry, I haven't tried the SPI Master mode on ESP8266. My knowledge ends with the slave mode. I don't understand the refences, too.