esp8266/Arduino #5717: NODEMCU esp8266 error (espcomm_open - espcomm_upload_mem failed)

chongboy59, 2019-02-04:
I am running the Ardiuno IDE with the NODEMCU esp8266 and have no had any problems until now and I can not figure out exactly what is wrong. 

I tried using new and old esp8266, one with the CP2102 chip, and one with the CH34OG chip. Neither are working now and I do not know why. 

Both connect to my computer and both are visible in my device manager (windows 10). 

I have tried 
switching cables, purchasing new esp8266 microcontrollers, and reinstalling the Arduino IDE. 
Maybe it has something to do with the tools setting I have set in my Arduino IDE. I will leave a picture of what I have as my tools settings. 
[image]
 

Please let me know! Thank you!

Zeron, 2019-02-04:
You have the wrong reset method selected for a NomeMCU board.

chongboy59, 2019-02-04:
> You have the wrong reset method selected for a NomeMCU board.

I changed my reset method to "none" and "nodemcu". I tried uploading both settings and It still gave me the same error message on both boards. (one with the CH34OG chip, and one with the CP2102 chip)

Zeron, 2019-02-04:
Have you tried manually resetting the device and forcing bootloader mode with the buttons before pressing the upload button in Arduino?

Press on hold the FLASH button on the NodeMCU board, then press the RST button,  Then press the Upload button in Arduino.  Release the FLASH button once the upload starts.