esp8266/Arduino #5721: FSBrowser doesn't work

GeorgeFlorian, 2019-02-05:
Hello there ! I've been tinkering with the Arduino IDE + ESP32 for the past week.

I tried the FSBrowser + AutoConnect from [here](https://github.com/Hieromon/AutoConnect/blob/master/examples/FSBrowser/FSBrowser.ino) but with no luck.

> Hardware:
> 
> Board: ESP32 DEVKIT1
> 
> Core Installation version: 1.0.1-git this one
> 
> IDE name: Arduino IDE
> 
> Flash Frequency: 80Mhz
> 
> PSRAM enabled: no
> 
> Upload Speed: 115200
> 
> Computer OS: Linux Mint 19.1 Mate
> 

This is my sketch:

```
/*
  FSWebServer - Example WebServer with SPIFFS backend for esp8266
  Copyright (c) 2015 Hristo Gochkov. All rights reserved.
  This file is part of the ESP8266WebServer library for Arduino environment.
*/

#include <FS.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>
#include <SPIFFS.h>
#include <AutoConnect.h>

#define DBG_OUTPUT_PORT Serial
#define LED 2

const char* ssid = "wifi-ssid";
const char* password = "wifi-password";
const char* host = "esp32fs";

WebServer server(80);

AutoConnect portal(server);
//holds the current upload
File fsUploadFile;

bool startCP(IPAddress ip) {
    digitalWrite(LED, HIGH);
    DBG_OUTPUT_PORT.println("Captive Portal started ! IP: " + WiFi.localIP().toString());
    return true;
  }

//format bytes
String formatBytes(size_t bytes){
  if (bytes < 1024){
    return String(bytes)+"B";
  } else if(bytes < (1024 * 1024)){
    return String(bytes/1024.0)+"KB";
  } else if(bytes < (1024 * 1024 * 1024)){
    return String(bytes/1024.0/1024.0)+"MB";
  } else {
    return String(bytes/1024.0/1024.0/1024.0)+"GB";
  }
}

[... 268 lines cut ...]
> 10:27:08.444 -> [AC] 192.168.1.103/_ac
> 10:27:08.477 -> [PB] at leaving build: 259664 free
> 10:27:08.477 -> [PB] Res:200, Chunked:0
> 10:27:08.477 -> [PB] Free heap:262440, content len.:6643
> 10:27:08.477 -> [PB] Transfer-Encoding:chunked
> 10:27:08.477 -> [E][WiFiClient.cpp:346] write(): 104
> 10:27:13.455 -> [D][WiFiClient.cpp:463] connected(): Disconnected: RES: 0, ERR: 128
> 10:27:13.854 -> [AC] esp32fs.local/favicon.ico
> 10:27:13.854 -> [AC] Page[] allocated
> 10:27:13.854 -> [E][WebServer.cpp:602] _handleRequest(): request handler not found
> 10:27:13.854 -> [D][WiFiClient.cpp:463] connected(): Disconnected: RES: 0, ERR: 128
> 10:27:18.866 -> [AC] esp32fs.local/_ac
> 10:27:18.866 -> [AC] Page[/_ac] allocated
> 10:27:18.866 -> [AC] 192.168.1.103/_ac
> 10:27:18.899 -> [PB] at leaving build: 259372 free
> 10:27:18.899 -> [PB] Res:200, Chunked:0
> 10:27:18.899 -> [PB] Free heap:262132, content len.:6643
> 10:27:18.899 -> [PB] Transfer-Encoding:chunked
> 10:27:18.899 -> [E][WiFiClient.cpp:346] write(): 104
> 

As you can see I've uploaded 4 files into the ESP32's FlashMemory. I would like to "edit" them. Maybe delete one of them, or create a new file. I can't do that since /edit does nothing.

Do you have any idea ?
Thank you !

Pablo2048, 2019-02-05:
?? this is ESP8266 repo, not https://github.com/espressif/arduino-esp32 one...

GeorgeFlorian, 2019-02-05:
> ?? this is ESP8266 repo, not https://github.com/espressif/arduino-esp32 one...

Since FSBrowser was made for the ESP8266 I thought here was a great place for starting. I am sorry if I assumed wrong.

closed 2019-02-05 by devyte

d-a-v, 2019-02-05:
From this repository's `FSBrowser.ino` comments (that you appear to not have)
```
upload the contents of the data folder with MkSPIFFS Tool ("ESP8266 Sketch Data Upload" in Tools menu in Arduino IDE)
or you can upload the contents of a folder if you CD in that folder and run the following command:
for file in `\ls -A1`; do curl -F "file=@$PWD/$file" esp8266fs.local/edit; done
access the sample web page at http://esp8266fs.local
edit the page by going to http://esp8266fs.local/edit
```
`edit.htm.gz` needs to be uploaded.

https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer/examples/FSBrowser/data