esp8266/Arduino #5722: Crash when using delay in both loop and callback

jeroenst, 2019-02-05:
When doing a delay in both the main loop and in the onwificonnect callback the esp will crash.

I hit this error because I wanted to send an udp packet upon onwificonnect to a hostname, which does start an DNS resolve. In this dns resolve there is a delay used to wait for the dns request to complete. That's why my code was crashing over and over.

Versions:
Arduino 1.8.8
esp8266 2.4.2

#include <ESP8266WiFi.h>

WiFiEventHandler wifiConnectHandler;
WiFiEventHandler wifiDisconnectHandler;

void onWifiConnect(const WiFiEventStationModeGotIP& event)
{
  Serial.println("Wifi connected...");
  delay(100);
  Serial.println("onWifiConnect Done...");
} 

void onWifiDisconnect(const WiFiEventStationModeDisconnected& event)
{
}

void connectToWifi()
{
  WiFi.begin();
}

void setup() {
  Serial.begin(115200);
  Serial.println("");
  Serial.println("ESP8266 Started...");

  wifiConnectHandler = WiFi.onStationModeGotIP(onWifiConnect);
  wifiDisconnectHandler = WiFi.onStationModeDisconnected(onWifiDisconnect);
  WiFi.setAutoReconnect(false); // We handle reconnect our self
  //WiFi.setSleepMode(WIFI_NONE_SLEEP); // When sleep is on regular disconnects occur https://github.com/esp8266/Arduino/issues/5083
  WiFi.mode(WIFI_STA);
  WiFi.hostname("test");

  connectToWifi();
}

void loop() {
  delay(100);
}

Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x40104b14
EXCVADDR: 0x00000003

Decoding stack results
0x4020efad: dhcp_recv at core/ipv4/dhcp.c line 1791
0x40202dd0: HardwareSerial::write(unsigned char const*, unsigned int) at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/HardwareSerial.h line 159
0x40202f99: Print::write(char const*) at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/Print.h line 60
0x40201550: delay at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/core_esp8266_wiring.c line 46
0x40202fd8: Print::println(char const*) at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/Print.cpp line 190
0x402024c5: onWifiConnect(WiFiEventStationModeGotIP const&) at /home/jeroen/Arduino/gethostbynamebug/gethostbynamebug.ino line 10
0x4020369c: std::_Function_handler ::_M_invoke(std::_Any_data const&, WiFiEventStationModeGotIP const&) at /home/jeroen/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2073
0x402026f0: std::_Function_handler  )::__lambda4>::_M_invoke(const std::_Any_data &, _esp_event *) at /home/jeroen/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2073
0x402028af: ESP8266WiFiGenericClass::_eventCallback(void*) at /home/jeroen/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_list.h line 155
0x40202c78: ESP8266WiFiSTAClass::status() at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp line 498
0x40203420: esp_yield() at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/core_esp8266_main.cpp line 91
0x4020155b: delay at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/core_esp8266_wiring.c line 51
0x40202500: loop() at /home/jeroen/Arduino/gethostbynamebug/gethostbynamebug.ino line 42

devyte, 2019-02-05:
The wifi callbacks execute in the SYS context, and you can't yield/delay in there. If you need to do that, I suggest to use the wifi callback to schedule another callback with your code that requires yield/delay.

Same thing as the Ticker callback, etc.

In general, you should not do heavy operations in any callback. There are timing restrictions to observe.

Closing due to user error.

closed 2019-02-05 by devyte

jeroenst, 2019-02-05:
Thanx for info

mbscyclone, 2020-10-24:
Thanx for info